ucb-bar/chipyard #952: Why might the binary failed to be found? (checked misspellings / directory issues)

michael-etzkorn, 2021-08-22:
Attempting to install the riscv toolchain directly within chipyard using the shell script would fail, but I managed to install the toolchain from `ariane-sdk` and directly installed riscv-tests. However when I set the binary as  `BINARY=$RISCV/riscv64-unknown-elf/share/riscv-tests/isa/rv64ui-p-simple` it says it is unable to open.  `$RISCV` here is `/home/ye/ariane-sdk/install/riscv/toolchain`. I installed riscv-tests within a folder named share. The path does exist and the binary is present in isa. I guess I shouldn't be surprised that my workaround for installing the riscv toolchain failed, but I was hoping someone could tell me where to look to start debugging this issue.

jerryz123, 2021-08-23:
Is there sufficient permissions on the binary?

You should try to get the included riscv-toolchain included. There are no guarantees that the ariane-sdk will work here, especially since it looks like that toolchain hasn't been updated in some time.

closed 2021-08-23 by michael-etzkorn

michael-etzkorn, 2021-08-31:
You pass binary without `BINARY=`