ucb-bar/chipyard #954: How to setup VCU118Config for CVA6?

michael-etzkorn, 2021-08-22:
I thought a config for cva6 would be as simple as 
```
class cva6VCU118Config extends Config(
  new WithVCU118Tweaks ++ 
  new chipyard.CVA6Config )
```
but when compiling ExtMem isn't defined as a parameter. 

```
make[1]: 离开目录“/mnt/hgfs/SageProjects/chiselToys/chipyard/fpga/src/main/resources/vcu118/sdboot”
[error] java.util.NoSuchElementException: None.get
[error] 	...
[error] 	at cva6.CVA6TileModuleImp.<init>(CVA6Tile.scala:207)
[error] 	at cva6.CVA6Tile.$anonfun$module$1(CVA6Tile.scala:159)
[error] 	at chisel3.internal.plugin.package$.autoNameRecursively(package.scala:52)
[error] 	at cva6.CVA6Tile.module$lzycompute(CVA6Tile.scala:159)
[error] 	at cva6.CVA6Tile.module(CVA6Tile.scala:159)
[error] 	at cva6.CVA6Tile.module(CVA6Tile.scala:106)
[error] 	at freechips.rocketchip.diplomacy.LazyModuleImpLike.$anonfun$instantiate$3(LazyModule.scala:278)
[error] 	at chisel3.Module$.do_apply(Module.scala:54)
[error] 	at freechips.rocketchip.diplomacy.LazyModuleImpLike.$anonfun$instantiate$2(LazyModule.scala:278)
[error] 	at chisel3.internal.plugin.package$.autoNameRecursively(package.scala:52)
[error] 	at freechips.rocketchip.diplomacy.LazyModuleImpLike.$anonfun$instantiate$1(LazyModule.scala:278)
[error] 	at scala.collection.immutable.List.flatMap(List.scala:338)
```
At least I think that's what's happening. I thought WithSystemModifications would define ExtMem for us, so I'm confused on what I need to add to the CVA6 config beyond what I have here.

jerryz123, 2021-08-23:
The CVA6 shim assumes AXI4-basex backing memory. This should be fixed in the shim.

For now, can you replace `p(ExtMem).get.master.base` and `p(ExtMem).get.master.size` with `0x80000000` and whatever the size of DRAM in the system is?

michael-etzkorn, 2021-08-23:
`new chipyard.config.WithTLBackingMemory` sets `ExtMem` to `None` so I removed it from `WithVCU118Tweaks`. But perhaps the backing memory should be TL and the fix should be as above. 

After running without `WithTLBackingMemory` I'm facing a similar issue with the `SPIOverlayKey` in `TestHarness`. My guess is `sdio` is `None` because `pmod_is_sdio` is false. Need to look into what sets `pmod_is_sdio` if we end up needing SPI (don't think we will), but I'm commenting it out for now.
```
[error] java.util.NoSuchElementException: head of empty list
[error] 	...
[error] 	at chipyard.fpga.vcu118.VCU118FPGATestHarness.<init>(TestHarness.scala:77)
```

michael-etzkorn, 2021-08-23:
As I think about it... If SPI is for the SD card, it'll be necessary to get anything running on the core.

michael-etzkorn, 2021-08-23:
Thanks @jerryz123 that swap works. Trying to remove `WithTLBackingMemory` causes `dp(DDROverlayKey)` to return `None`.

closed 2021-08-23 by michael-etzkorn

Yuxin-Yu, 2023-02-08:
Hello，I chang the same code as @jerryz123 said,but when I boot Linux，it stuck.Just like:
[image]
May I ask what could be the reason? Thanks!

Yuxin-Yu, 2023-02-08:
I'm wondering if it's the problem of my boot image, and do I  need to make appropriate modifications to the image for CVA6

michael-etzkorn, 2023-02-08:
This is usually because the SD card's partition is incorrect or the fw
image is loaded incorrectly.
Make sure your 1st partition starts at sector 34.

Yuxin-Yu, 2023-02-09:
> This is usually because the SD card's partition is incorrect or the fw image is loaded incorrectly. Make sure your 1st partition starts at sector 34. Yuxin ***@***.***> 于2023年2月8日周三 01:49写道：
> […](#)

I'm sure my 1st partition start at sector 34,because I have boot Linux in rocket core in many times.I don't think that's the reason for the current mistake

michael-etzkorn, 2023-05-05:
> I'm sure my 1st partition start at sector 34,because I have boot Linux in rocket core in many times.I don't think that's the reason for the current mistake

Curious if you ever resolved this. I have not looked at CVA6 since back in 2021 the shim still had some issues with booting linux https://github.com/ucb-bar/chipyard/issues/957#issuecomment-906607725. 

Looking back over this issue though, what I think is most likely is that you have set the size or the address of DDR wrong when replacing `p(ExtMem)`. For the VCU118, the base is `0x8000_0000` and I would allocate at least a gig for size (`0x4000_0000`)

37eex9, 2025-11-07:
I want to provide a summary of my config, hopefully helping others ending up in this thread. (Just like me ;~)

`fpga/src/main/scala/vcu118/Configs.scala`
```
class CVA6VCU118Config extends Config(
  new WithFPGAFrequency(50) ++
  new WithVCU118Tweaks ++
  new chipyard.CVA6Config
)
```

In `generators/cva6/src/main/scala/cva6/CVA6Tile.scala` I replaced both `p(ExtMem).get.master.base` and `p(ExtMem).get.master.size` with `0x80000000`. According to the [Board User Guide](https://docs.amd.com/v/u/en-US/ug1224-vcu118-eval-bd)(page 22), the board provides 5GB = 2 x 2.5GB = 2 x (5 x 256Mb x 16) DDR4 RAM. Never the less, changing the size to anything more than 2GiB = 0x80000000 (I tried GiB steps and `0x9502f900` and `0x98968000`) did not work and the board hung at the `BOOT` step.

I tried other frequencies, too, but the fastest I found working was 50MHz.

A little off topic but it took me long to figure out the following: to get any of my SD cards working I had to change the `SPI_CLK` in `fpga/src/main/resources/vcu118/sdboot/sd.c`. There is some discussion [here](https://groups.google.com/g/chipyard/c/i1kTOzeXa6M/m/ChMgi1kcAgAJ).