ucb-bar/chipyard #957: VCU118: Is it possible to use a pre-built Linux Image?

michael-etzkorn, 2021-08-24:
Trying to use FireMarshal but it looks like it crashes the entire computer when running (I didn't get a good look at the error messages, but I'll follow up there if it happens a third time). Will prebuilt images like at [ariane-sdk](https://github.com/pulp-platform/ariane-sdk/releases) work with bitstreams targeted for the VCU118 generated from Chipyard?

abejgonzalez, 2021-08-25:
I think other binaries should work assuming they follow this criteria:
1. Initramfs... It is a diskless version of the binary (everything bundled together) (provided by FireMarshal by `--nodisk` flag)
2. It is flattened so that the SDCard can load it properly: https://github.com/firesim/FireMarshal/blob/daf1040c0f0ec449734cd183be876880357644fc/boards/default/installers/prototype/prototype.py#L16
With that being said, I think its best to use the FireMarshal tool set. Happy to help on bugs.

michael-etzkorn, 2021-08-25:
FireMarshal now gives me an image. I'm worried that the SD card were using has a protective MBR region. I assume we should try to find an SD card without that?
on running gdisk (after non-expert mode -o command)

```
Partition table scan:
  MBR: protective
  BSD: not present
  APM: not present
  GPT: present
```

michael-etzkorn, 2021-08-25:
On second thought, seems like o command creates the protective mbr in the first place, so maybe it'll work with that.

michael-etzkorn, 2021-08-25:
Just to be clear, `apfs` is Apple File System right? I'm really hoping to not have this be the point of bug introduction.

michael-etzkorn, 2021-08-25:
https://puu.sh/I6uBh/8f5a40c93e.png I'm at least getting output, but gonna try another sd card since I've been stuck at BOOT with no further output for a bit.

Yuxin-Yu, 2022-07-26:
> Just to be clear, `apfs` is Apple File System right? I'm really hoping to not have this be the point of bug introduction. And `HFS` is `Apple HFS`? I have `af0a` and `af00` for these.

Hello, what system are you using? When I use gdisk format SD card to convert to apfs partition, there is no option for af0a, and the system uses centos-7

michael-etzkorn, 2022-07-26:
My guess is it probably has more so to do with the gdisk version. 

- CentOS Linux release 7.8.2003 (Core)
- GPT fdisk (gdisk) version 0.8.10

Yuxin-Yu, 2022-07-26:
> Hi @Yuxin-Yu My guess is it probably has more so to do with the gdisk version.
> 
> * CentOS Linux release 7.8.2003 (Core)
> * GPT fdisk (gdisk) version 0.8.10

I found that my version of gdisk is the same as yours, but I don't know why there is no option for af0a. Is it because the format of the SD card is wrong? My SD card is also entering gdisk and the following phenomenon is also observed:
```
GPT fdisk (gdisk) version 0.8.10

The protective MBR's 0xEE partition is oversized! Auto-repairing.

Partition table scan:
   MBR: protective
   BSD: not present
   APM: not present
   GPT: present

Found valid GPT with protective MBR; using GPT.
```
Do you use other commands besides the command "o", and I found that after I use the command "o", the display results are as follows:
```
Command (? for help): o
This option deletes all partitions and creates a new protective MBR
```
I feel that the display here is not as expected. It should display the GPT partition, right? can you tell me what your process is like

michael-etzkorn, 2022-07-27:
Ah, I apologize. I haven't used `gdisk` in a while. I forgot I didn't actually use the CentOS machine to format the SD card. I used Mint 20.2 in a virtual machine and whatever gdisk comes installed with that. It's probably a later version. 

You should be able to use ext4 file system instead. I will confirm later when I have an SD card on hand. The "new partition with protective MBR" looks correct as it is saying it will create a GPT partition with MBR.

Yuxin-Yu, 2022-07-27:
> 
I tried the mint system you mentioned, it works, thanks

leduchuybk, 2022-09-19:
i met same problem as @michael-etzkorn had. I got stuck at BOOT with no futher output. 
I do not have non nodisk version of binary after running
`./marshal -v -d build br-base.json`
`./marshal -v -d install -t prototype br-base.json`
I only had following files under `images` folder:
`br-base-bin-dwarf-nodisk
br-base-bin-nodisk
br-base-bin-nodisk-flat
br-base.img
br.07bb.img`

I ran `./marshal launch` to check my binary on qemu. but it said there is no `br-base-bin` under images folder. So i changed qemu command using br-base-bin-nodisk instead. it can boot and enter linux to type down user name and password.

[image]
[image]

leduchuybk, 2022-09-19:
> I've removed the mbr with `dd if=/dev/zero of=/dev/[sd card here] bs=512 count=1`

@michael-etzkorn can you explain again the order of formating sd card? how to remove mbr partitions. why do you need to remove it?

michael-etzkorn, 2022-09-19:
> > I've removed the mbr with `dd if=/dev/zero of=/dev/[sd card here] bs=512 count=1`
> 
> @michael-etzkorn can you explain again the order of formating sd card? how to remove mbr partitions. why do you need to remove it?

Don't remove the mbr region, it is part of the GPT formatting. Perhaps the documentation should be clearer on this.

leduchuybk, 2022-09-20:
> i met same problem as @michael-etzkorn had. I got stuck at BOOT with no futher output.
> I do not have non nodisk version of binary after running
> `./marshal -v -d build br-base.json`
> `./marshal -v -d [...]

@michael-etzkorn can you confirm my sdcard format is correct?
even better,
can you upload a prototype binary of linux kernel you had built so that i can try on my board?

michael-etzkorn, 2022-09-20:
I think it's probably an SD card issue. If I remember correctly, I don't think your MBR is meant to be corrupted. 
Are you following the steps outlined in the doc to format and partition the SD card? 

Are you using dd to load br-base-bin-nodisk-flat into the first partition?

leduchuybk, 2022-09-20:
> I think it's probably an SD card issue. If I remember correctly, I don't think your MBR is meant to be corrupted. 
> 
> Are you following the steps outlined in the doc to format and partition the SD card? 
> 
> 
> 
> Are you using dd to load br-base-bin-nodisk-flat into the first partition? 

you meant it should look like this right? [image]
i used `gdisk` (like in the doc) to partition sdcard and `dd` to load binary file

michael-etzkorn, 2022-09-20:
So you're passing the -no-disk-flat image with dd to `/dev/sdb1` yeah?

leduchuybk, 2022-09-20:
> So you're passing the -no-disk-flat image with dd to `/dev/sdb1` yeah?

yes. the whole command is:
`sudo dd if=br-base-bin-nodisk-flat of=/dev/sdb1`
is there anyway to read back data after this command? to check whether i write correctly. 
my idea is to build jtag connect. using openocd `mdw` to read `0x80000000` address after completing of sd.c program. 
or maybe, can you upload your image binary so that i can try myself?

michael-etzkorn, 2022-09-21:
> Is there anyway to read back data after this command

There likely is, but I am not aware of one. You can always try using a different SD card (sometimes they just don't work). 

I've attached my br-base-bin-nodisk-flat. 

[br-base-bin-nodisk-flat.tar.gz](https://github.com/ucb-bar/chipyard/files/9612672/br-base-bin-nodisk-flat.tar.gz)

leduchuybk, 2022-09-21:
@michael-etzkorn thank you for uploading binary file. i tried but got same result. 
[image]
these are sd cards i am using. they are both sdhc cards.

michael-etzkorn, 2022-09-22:
If you wrote your changes in gdisk, it should exit after that. I always skip the last step of the documentation to make the second partitiod a file system though that shouldn't influence this. You can use `fdisk -l` or `lsdisk` to inspect the partitions of the SD card.  If you're confident in your following of the SD card partition instructions in the documentation, and it's still not being recognized, the next step would be trying a few different brands of SD cards.

michael-etzkorn, 2022-09-22:
The picture finally loaded. I don't think it's likely for two different SD cards to pose this problem, I did once have a brand of SD card that didn't  work (though I don't remember what company's SD card), but Sandisk should be reliable.

I did suddenly remember something. Sometimes with `dd`, if the SD card wasn't detected, you'll create a file handle that'll take the same spot the device partition handle is. Linux won't know the difference since Linux treats devices like files. You will end up writing to the file instead of the SD card. If you take the SD card out and the `/dev/sdb1` is still there, you've created a file handle (this is usually a different color when displayed with `ls` than the device handles).  Remove this with `rm` and plug the sd card back in. Assure it shows up under the `/dev/` folder before running the `dd` command again.

leduchuybk, 2022-09-24:
@michael-etzkorn thank you for your support
From your note, I detected something might be the cause. After using `dd` to copy binary file into SDcard, i re-check it with `gparted`. The 1st partition may be corrupted and cannot be detected as in the picture. Did you experience the same thing?
[image]

leduchuybk, 2022-09-30:
@michael-etzkorn i detected my problem. it is because of wrong tcl script for my MiG IP. I copied the config of vcu118 to use for zcu104