ucb-bar/chipyard #964: Vcu118 has added ddr2, but does not create the node of ddr2

zhuxuanlong, 2021-08-28:
Hi,
By reading the code, I found a problem, that is DDR2 is added to vcu118 in FPGA, but no DDR2 node is created. Why?
`class VCU118FPGATestHarness(override implicit val p: Parameters) extends VCU118ShellBasicOverlays {

  def dp = designParameters

  val pmod_is_sdio  = p(VCU118ShellPMOD) == "SDIO"
  val jtag_location = Some(if (pmod_is_sdio) "FMC_J2" else "PMOD_J52")

  // Order matters; ddr depends on sys_clock
  val uart      = Overlay(UARTOverlayKey, new UARTVCU118ShellPlacer(this, UARTShellInput()))
  val sdio      = if (pmod_is_sdio) Some(Overlay(SPIOverlayKey, new SDIOVCU118ShellPlacer(this, SPIShellInput()))) else None
  val jtag      = Overlay(JTAGDebugOverlayKey, new JTAGDebugVCU118ShellPlacer(this, JTAGDebugShellInput(location = jtag_location)))
  val cjtag     = Overlay(cJTAGDebugOverlayKey, new cJTAGDebugVCU118ShellPlacer(this, cJTAGDebugShellInput()))
  val jtagBScan = Overlay(JTAGDebugBScanOverlayKey, new JTAGDebugBScanVCU118ShellPlacer(this, JTAGDebugBScanShellInput()))
  val fmc       = Overlay(PCIeOverlayKey, new PCIeVCU118FMCShellPlacer(this, PCIeShellInput()))
  val edge      = Overlay(PCIeOverlayKey, new PCIeVCU118EdgeShellPlacer(this, PCIeShellInput()))
  val sys_clock2 = Overlay(ClockInputOverlayKey, new SysClock2VCU118ShellPlacer(this, ClockInputShellInput()))
  val ddr2       = Overlay(DDROverlayKey, new DDR2VCU118ShellPlacer(this, DDRShellInput()))`

When creating DDR node below, only one is created.

`  val ddrNode = dp(DDROverlayKey).head.place(DDRDesignInput(dp(ExtTLMem).get.master.base, dutWrangler.node, harnessSysPLL)).overlayOutput.ddr`

michael-etzkorn, 2021-08-28:
Yeah, this is also true of `fmc` and `edge` for the PCIe overlays as well  (I'm currently figuring out how to attach vcu118 PCIe -> XDMA -> System Bus). The overlays are defined so when you inherit the class you don't need to redefine them. The second ddr node is defined and uses the overlay in `bringup/TestHarness.scala`. 
```scala
  val ddr2Node = dp(DDROverlayKey).last.place(DDRDesignInput(dp(PeripheryTSIHostKey).head.targetMasterPortParams.base, ddr2Wrangler.node, ddr2PLL)).overlayOutput.ddr
```

I imagine similarly we can instantiate a PCIe `edge` node in some similar fashion.

abejgonzalez, 2021-08-30:
@michael-etzkorn called it out correctly. The DDR(1) **overlay** is **placed** and **attached** by default in the main test harness:

https://github.com/ucb-bar/chipyard/blob/9d055fdac638ab90735cbde42fd2d86355eb260b/fpga/src/main/scala/vcu118/TestHarness.scala#L79-L90

The 2nd DDR **overlay** though defined in the main test harness is only **placed** and **attached** in the bringup harness. Currently, we don't have a nice API to have everything specified in the main test harness then configure it solely on the config (its a nice feature that we haven't worked on yet). Thus, if you want to attach more things to the FPGA (use more overlays) then you need to create a different test harness for it. Here is a reference to where the DDR2 is placed and attached:

https://github.com/ucb-bar/chipyard/blob/9d055fdac638ab90735cbde42fd2d86355eb260b/fpga/src/main/scala/vcu118/bringup/TestHarness.scala#L59-L92

abejgonzalez, 2021-08-30:
> Yeah, this is also true of `fmc` and `edge` for the PCIe overlays (I'm currently figuring out how to attach vcu118 PCIe -> XDMA -> System Bus).
> I imagine similarly we can instantiate a PCIe `edge` node in some similar fashion.

One thing to note about attaching things. The `CanHaveMasterTLMemPort` mixin was used to expose a TL node out of ChipTop. Unfortunately, this was done because the diplomatic interface doesn't extend out of the ChipTop (the chip is one diplomatic area and the harness is another). If you want to connect to the other busses then you have to do something similar.

abejgonzalez, 2021-08-30:
Closing since the main issue seems to be addressed

closed 2021-08-30 by abejgonzalez

sant0s12, 2022-06-11:
@michael-etzkorn
> I'm currently figuring out how to attach vcu118 PCIe -> XDMA -> System Bus

Any update on this?

michael-etzkorn, 2022-06-11:
@sant0s12 I have it working on the board with an ugly hack for the clocks. 

Some of my code (without the clock hack) and explaining my hook-up is here: https://github.com/sifive/fpga-shells/issues/165 

This has most of the idea. 

The clocking I still haven't quite figured out. For now, I'm generating a second clock that has the same frequency as the axi_aclk (similar to how the DDR2 in the bringup test harness instantiates a second clock sink node), but the axi_aclk is supposed to clock the axi register logic directly (and not doing so can cause issues when using separate PLLs). I'm still not quite sure how to use the axi_aclk to feed the axi buffers with the diplomatic clock nodes. 

Perhaps you can post something on the google group or on Stack Overflow and we can start a discussion there. 

Note: the PCIe overlay output instantiates a root complex XDMA and not an endpoint.

sant0s12, 2022-06-12:
@michael-etzkorn Thanks, I posted a question [here](https://stackoverflow.com/questions/72590994/how-to-connect-a-pcie-device-to-a-chipyard-design).