ucb-bar/chipyard #967: How do you use TSI Widget in Bringup VCU118?

michael-etzkorn, 2021-09-01:
Hi I'm trying to understand what was connected over the FMC pins to utilize a TSI Host Widget for testing an SoC on the VCU118. After seeing https://github.com/ucb-bar/testchipip/issues/116 I'm guessing the documentation doesn't exist yet, but I would love a rough English overview of how to take advantage of this widget. In the meantime, I'll look at how to utilize it by reading through the source code.

abejgonzalez, 2021-09-13:
Are you trying to test an SoC prototype on the VCU118 or are you trying to use the VCU118 as a test platform to interact with a separate DUT? The TSI Host Widget helps with the latter. It maps an MMIO TSI interface (see 1) that can be attached from a prototype SoC on the VCU118 to a slow digital SerDes port located on the FMC (see 2). Additionally, it exposes a memory port that connects to 1 channel of the DDR of the VCU118 so that the DUT can use it (the DUT sends memory requests over the slow SerDes port). I would look at 8.2.3.3 of the documentation, here the "Serialized TL" is going over the FMC cable.

1 - https://chipyard.readthedocs.io/en/latest/Advanced-Concepts/Chip-Communication.html#example-test-chip-bringup-communication
2 - https://github.com/ucb-bar/chipyard/blob/9d055fdac638ab90735cbde42fd2d86355eb260b/fpga/src/main/scala/vcu118/bringup/CustomOverlays.scala#L172-L183

michael-etzkorn, 2021-09-13:
I see. I mistakenly thought it had to do with testing the SoC prototype on the VCU118. I assume we can use an ILA for that. I do find the idea of using the VCU118 as a test platform an interesting one. Something to explore later :) Thanks for the info!

closed 2021-09-13 by michael-etzkorn

Dai-dirk, 2024-07-17:
> Are you trying to test an SoC prototype on the VCU118 or are you trying to use the VCU118 as a test platform to interact with a separate DUT? The TSI Host Widget helps with the latter. It maps an [...]

Hi, I want to use the FPGA as a test platform to interact with a separate DUT with its own DDR. However, I don't know how the FPGA can send the data to DUT's DRAM via  "Serialized TL"? According to my understanding, the DUT starts executing the program from 0x8000000. Hence, the FPGA needs to send the bare-metal program to DUT's DRAM at the beginning of 0x8000000. However, I found no address signals for "Serialized TL" signals. I guess the TLserdesser with DUT is responsible for the above process.