ucb-bar/chipyard #969: TLClientNode in TestHarness

michael-etzkorn, 2021-09-02:
I'm trying to understand how the TestHarness connects the DDR to ChipTop. I understand that memTLNode is created from a TLManagerNode
```scala
 val memTLNode = TLManagerNode(memPortParamsOpt.map({ case MemoryPortParams(memPortParams, nMemoryChannels) =>
    Seq.tabulate(nMemoryChannels) { channel =>
      val base = AddressSet.misaligned(memPortParams.base, memPortParams.size)
      val filter = AddressSet(channel * mbus.blockBytes, ~((nMemoryChannels-1) * mbus.blockBytes))

     TLSlavePortParameters.v1(
       managers = Seq(TLSlaveParameters.v1(
         address            = base.flatMap(_.intersect(filter)),
         resources          = device.reg,
         regionType         = RegionType.UNCACHED, // cacheable
         executable         = true,
         supportsGet        = TransferSizes(1, mbus.blockBytes),
         supportsPutFull    = TransferSizes(1, mbus.blockBytes),
         supportsPutPartial = TransferSizes(1, mbus.blockBytes))),
         beatBytes = memPortParams.beatBytes)
   }
 }).toList.flatten)
```

But I thought the client would be requests for memory from the CPU. But TestHarness instantiates its own TLClientNode that connects to the manager. I'm trying to understand why a Client node is instantiated here and maybe it comes from my lack of understanding of the `inParams` line. 

```scala
  val ddrNode = dp(DDROverlayKey).head.place(DDRDesignInput(dp(ExtTLMem).get.master.base, dutWrangler.node, harnessSysPLL)).overlayOutput.ddr

  // connect 1 mem. channel to the FPGA DDR
  val inParams = topDesign match { case td: ChipTop =>
    td.lazySystem match { case lsys: CanHaveMasterTLMemPort =>
      lsys.memTLNode.edges.in(0)
    }
  }
  val ddrClient = TLClientNode(Seq(inParams.master))
  ddrNode := ddrClient
```

michael-etzkorn, 2021-09-02:
My thought process atm is that the cpu connects to the manager end, and then the client end is what makes the request to ddr.

michael-etzkorn, 2021-09-02:
This is related to things like `CanHaveSlaveTLPort` and I'm guessing if we want a slave port on the front bus, we instatiate a TLClientNode which kicks off its request when the external master talks to the corresponding TLManagerNode.

jerryz123, 2021-09-03:
The code you pasted is basicaly copying the port params from the ChipTop into the DDR client node, such that they are configured the same way.

michael-etzkorn, 2021-09-03:
I think I'm clear enough on instantiating a corresponding outer client for the inner manager that I'll close this issue.

closed 2021-09-03 by michael-etzkorn