ucb-bar/chipyard #970: Got stuck when booting linux on VCU118 with MegaBoom

xiaoQ77, 2021-09-03:
Following the prototyping flow documents, I could boot linux on VCU118 with the rocket-core and the LargeBoom core. However, when I switched to the MegaBoom core, it got stuck. I notice that @pengwubj has posted this issue before. Has it been solved?

---reference----
Thanks very much for this enhencement. It works with rocket-core to boot linux . But when I switch to boom-core, the kernel stuck at init process as below:
Could you help on this issue. Thank you very much.
Platform Name       : freechips,rocketchip-unknown
Platform Features   : timer,mfdeleg
Platform HART Count : 1
Boot HART ID        : 0
Boot HART ISA       : rv64imafdcsux
BOOT HART Features  : pmp,scounteren,mcounteren
BOOT HART PMP Count : 16
Firmware Base       : 0x80000000
Firmware Size       : 92 KB
Runtime SBI Version : 0.2

MIDELEG : 0x0000000000000222
MEDELEG : 0x000000000000b109
PMP0    : 0x0000000080000000-0x000000008001ffff (A)
PMP1    : 0x0000000000000000-0x00000001ffffffff (A,R,W,X)
[    0.000000] OF: fdt: No chosen node found, continuing without
[    0.000000] OF: fdt: Ignoring memory range 0x80000000 - 0x80200000
[    0.000000] Forcing kernel command line to: console=hvc0 earlycon=sbi
[    0.000000] Linux version 5.7.0-rc3-58539-g5f5fd87 (gcc version 9.2.0 (GCC), GNU ld (GNU Binutils) 2.32) #1 SMP Fri Apr 9 17:29:05 CST 2021
[    0.000000] earlycon: sbi0 at I/O port 0x0 (options '')
[    0.000000] printk: bootconsole [sbi0] enabled
[    0.000000] initrd not found or empty - disabling initrd

_Originally posted by @pengwubj in https://github.com/ucb-bar/chipyard/issues/822#issuecomment-818319502_

abejgonzalez, 2021-09-15:
If I recall correctly, there was a bug in RISC-V Linux that was exposed when running on MegaBOOM. Maybe this is it? You need to add a local flush to the TLB at the end of this function: https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/tree/arch/riscv/mm/init.c#n394. Have you tried running w/ LargeBoom?

abejgonzalez, 2021-09-21:
Turns out this is a branch of Linux with the fix added. Supposedly this is now in upstream Linux.

Cuicunhao, 2021-12-05:
Hello, excuse me, I'd like to ask some questions, I also follow the prototype flow documents, but when I run “./marshal -v -d build br-base.json” , My computer will crash when this program runs to“make ARCH=riscv CROSS_COMPILE=riscv64-unknown-linux-gnu- vmlinux Image -j”. Moreover, when I set up the sdcard, there is no apfs type in gdisk. Can you help me? Thank you very much.

abejgonzalez, 2021-12-06:
There could be a multitude of reasons why `make ...` breaks so without more information on the specific error, I can't help you. Make sure you look through the log that is generated from the build and try to solve the error it throws. Some resources to use are this GH issues area and our mailing list: https://groups.google.com/forum/#!forum/chipyard.

As for the `apfs` issue in `gdisk`, doing a search in the mailing list brought up this: https://groups.google.com/u/1/g/chipyard/c/SBdQCsqDfok/m/C8Rbs3lAAgAJ

Hope that helps