ucb-bar/chipyard #971: No firrtl blackbox resources when running custom project?

michael-etzkorn, 2021-09-06:
I'm not quite sure where to look. I'm not using a build.sbt, and I'm just placing my folder within the vcu118 folder similar to how bringup works. I added the following to the Makefile. 
```
ifeq ($(SUB_PROJECT),xdma)
        SBT_PROJECT       ?= fpga_platforms
        MODEL             ?= PCIeVCU118FPGATestHarness
        VLOG              ?= PCIeVCU118FPGATestHarness
        MODEL_PACKAGE     ?= chipyard.fpga.vcu118.xdma
        CONFIG            ?= PCIeRocketConfig
        CONFIG_PACKAGE    ?= chipyard.fpga.vcu118.xdma
        GENERATOR_PACKAGE ?= chipyard
        TB                ?= none # unused
        TOP               ?= ChipTop
        BOARD             ?= vcu118
        FPGA_BRAND        ?= xilinx
endif
```

After running `make verilog SUB_PROJECT=xdma` none of the blackbox resources like `IOCell.v` appear in the firrtl black box resource files. Only `EICG_wrapper.v` appears in `sim_files.f`. I'm going to try looking at the Makefiles more carefully to see how it gets copied over. I don't think it's an issue with `tapeout` because it's a subproject of `fpga` in `vcu118`.

michael-etzkorn, 2021-09-06:
I think it might have to do with the folder being symbolically linked? I'll check that in the morning.

michael-etzkorn, 2021-09-06:
Is not related to symbolic links. Interestingly, barstools is doing the following transform for the test harness. Probably related?
```
 {
    "class":"barstools.tapeout.transforms.stage.HarnessTopAnnotation",
    "harnessTop":"-faf"
  },
```

lsteveol, 2021-09-06:
```
	$(call run_scala_main,tapeout,barstools.tapeout.transforms.GenerateTopAndHarness,-o $(TOP_FILE) -tho $(HARNESS_FILE) -i $(FIRRTL_FILE) --syn-top $(TOP) --harness-top $(VLOG_MODEL) -faf $(ANNO_FILE) -tsaof $(TOP_ANNO) -tdf $(sim_top_blackboxes) -tsf $(TOP_FIR) -thaof $(HARNESS_ANNO) -hdf $(sim_harness_blackboxes) -thf $(HARNESS_FIR) $(REPL_SEQ_MEM) $(HARNESS_CONF_FLAGS) -td $(build_dir) -ll $(FIRRTL_LOGLEVEL)) && touch $(sim_top_blackboxes) $(sim_harness_blackboxes)
```

https://github.com/ucb-bar/chipyard/blob/master/common.mk#L131

I don't see where you are setting the `VLOG_MODEL`, so the make file is using `-faf` for that variable.

michael-etzkorn, 2021-09-06:
Ah good catch. Yeah I set it as `VLOG` not `VLOG_MODEL` that ought to fix it

closed 2021-09-06 by michael-etzkorn

leduchuybk, 2022-11-04:
hi @michael-etzkorn 
i met similar issue recently.
i tried to generate hdl code for only the module i am developing.
For example, generating hdl code for `vcu118mig` in fpga_platforms project. i added the following in Makefile

`ifeq ($(SUB _PROJECT), vcu118mig)
SBT PROJECT?= fpga_platforms
MODEL ?= vcu118mig
VLOG_MODEL ?= vcu118mig
MODEL_PACKAGE
?= sifive.fpgashells.ip.xilinx.vcu118mig
CONFIG ?= vcu118mig
CONFIG_PACKAGE ?= sifive
GENERATOR_PACKAGE ?= chipyard
TB ?= none # unused
TOP ?= vcu118mig
BOARD ?= vcu118
FPGA_BRAND ?= xilinx
endif`

after that, i run: make SUB_PROJECT=vcu118mig verilog
to generate code.
i want to ask:
is this a formal way to develop a new peripheral module in chipyard?
can i develop similar module in this way: genrerate hdl code to test my module => integrate it in the soc system.

michael-etzkorn, 2022-11-06:
> is this a formal way to develop a new peripheral module in chipyard?
> can i develop similar module in this way: genrerate hdl code to test my module => integrate it in the soc system.

As of now, there's no cookie-cutter way to do simulations of complete FPGA environments. For simulations, I usually generate the verilog and instantiate ChipTop into a test harness outside of chipyard entirely. I leave the entire project in the `fpga_platforms` project's fpga/src/main/scala/vcu118 folder. I then modified the makefile similar to what I have above. My model package is not the sifive.fpgashells one, it's a custom package containing anything new. I also continued to use ChipTop just overriding the `BuildSystem` key with my customized DigitalTop which instantiates a customized System. From there, I generate the verilog to plug into simulation or generate the bitstream for fpga prototyping as needed.