ucb-bar/chipyard #978: AXI4IdIndexer error in documentation?

michael-etzkorn, 2021-09-09:
**Impact**: documetnation

The documentation 
```
9.5.8. AXI4IdIndexer
The AXI4 equivalent of TLSourceShrinker. This limits the number of AWID/ARID bits in the slave AXI4 interface. Useful for connecting to external or black box AXI4 ports.

Arguments:

idBits: Int - The number of ID bits on the slave interface.
Example Usage:

// master.node may have >16 unique IDs
// slave.node will only see 4 ID bits
slave.node := AXI4IdIndexer(4) := master.node
Notes:

The AXI4IdIndexer will create a user field on the slave interface, as it stores the ID of the master requests in this field. If connecting to an AXI4 interface that doesn’t have a user field, you’ll need to use the AXI4UserYanker.
```
This claims that idBits is the total number of ids on the slave interface, but looking at the code, it seems to be the width of the number of bits? So AXI4IdIndexer(4) would be 16 IDs. `idBits` according to the code says its the number of bits for the ID to be used. So `1 << idBits` IDs total.   
```
/** This adapter limits the set of FIFO domain ids used by outbound transactions.
  *
  * Extra AWID and ARID bits from upstream transactions are stored in a User Bits field called AXI4ExtraId,
  * which values are expected to be echoed back to this adapter alongside any downstream response messages,
  * and are then prepended to the RID and BID field to restore the original identifier.
  *
  * @param idBits is the desired number of A[W|R]ID bits to be used
  */
      }
```

I could be wrong on this, but looking at this snippet of code in `IdIndexer.scala` makes me believe idBits refers to the width of the bits used to represent an ID.

```scala
   // Create one new "master" per ID
      val masters = Array.tabulate(1 << idBits) { i => AXI4MasterParameters(
         name      = "",
         id        = IdRange(i, i+1),
         aligned   = true,
         maxFlight = Some(0))
      }
```

**What is the current behavior?**

**What is the expected behavior?**

**Other information**

jerryz123, 2021-09-09:
The documentation seems clear here

> idBits: Int - The number of ID bits on the slave interface.

michael-etzkorn, 2021-09-10:
Yeah, I think I misread this.

closed 2021-09-10 by michael-etzkorn