ucb-bar/chipyard #980: Running other versions of rocker-chip

Rahul-Kande, 2021-09-13:
Hi, 
I am trying to set up chipyard with different versions of rocker-chip other than the default(a7b016e) that is currently there as part of my project work.  I tried to do this by simply checking out the version I want (3f03ba81) in chipyard/generators/rocket-chip but I don't think it is as straightforward as this. The `make` command in sims/vcs threw a lot of errors (screenshot attached). 

If support for all the versions is not possible, the rocket-chip I am interested in now is 3f03ba81.  Please advise on how to proceed. 

[image]

jerryz123, 2021-09-13:
Chipyard is closely tied to specific commits of rocketchip, so changing the rocketchip commit will likely cause build failures, as you are seeing.

What is the reason you want to use that specific rocketchip commit?

Rahul-Kande, 2021-09-13:
Hi, 

Thanks for the reply. 

Got it. I am trying to implement a [tool](https://github.com/compsec-snu/difuzz-rtl) that used this specific version of the rocket-chip. Thats why I wanted to use the same version on chipyard also. 

Ill try to run the rocket-chip standalone from that commit version without using the chipyard then.

closed 2021-09-14 by Rahul-Kande