ucb-bar/chipyard #985: How to test RoCC Accelerator alone

xiongwu12345, 2021-09-18:
Hi, I designed a custom accelerator using the RoCC interface. And now I want to test the accelerator independently on an FPGA board, so I need to write a test driver and only instantiate one custom accelerator. But I am confused about how to just instantiate a custom accelerator without the rocket core. 
I  instantiated a custom accelerator like this :
[image]

But I got this error:
[image]

And I found that the RoCC accelerators are Mount on Rocket core using this grammar.
[image]

So it seems that I need to give the custom accelerator a "parameter p", but how should I instantiate a custom accelerator clearly without other modules? And what is the logic behind these parameters?

Thanks.

jerryz123, 2021-09-18:
The parameters object passed in contains global configuration information that may be relevant to the accelerator.

If you want to create an accelerator-only testharness, I would copy+modify TestHarness.scala to have a version which directly instantiates the accelerator, passing in the `Parameters` object which is available in TestHarness.

xiongwu12345, 2021-09-19:
>  I would copy+modify TestHarness.scala to have a version which directly instantiates the accelerator

Thanks. And how can I get this specific TestHarness.scala file?

jerryz123, 2021-10-02:
TestHarness.scala is available in the chipyard source code, but you'll have to make some modifications to produce an accelerator-only TestHarness.