ucb-bar/chipyard #987: Is it possible to use a 32bit address peripheral with a 64-bit core?

michael-etzkorn, 2021-09-20:
This is more of a general rocket-chip question than chipyard specific, but I noticed that if I hook up my peripheral, it won't index addresses by 32 bit, only by 64 bit. In other words, I can't increment the address by 4, only by 8. Is there a way to maintain 32-bit aligned addressing for peripherals? Maybe it's because I've set the slave node to be executable? I wasn't sure what that option did, but making the node 64-bit aligned makes sense. It seems like 4-byte alignment is possible based on the GCD example.

```scala
  regmap(
    0x00 -> Seq(
      RegField.r(2, status)), // a read-only register capturing current status
    0x04 -> Seq(
      RegField.w(params.width, x)), // a plain, write-only register
    0x08 -> Seq(
      RegField.w(params.width, y)), // write-only, y.valid is set on write
    0x0C -> Seq(
      RegField.r(params.width, gcd))) // read-only, gcd.ready is set on read
```

jerryz123, 2021-09-20:
What do you mean "it won't index addresses by 32 bit"

michael-etzkorn, 2021-09-21:
In simulation, incrementing the awaddr on the slave port by 8 works fine, increments of 4 seems to cause the peripheral to receive "0" in wdata instead of data placed on wdata at the slave port.  My guess is 64 bit aligned addressing occurs for executable slave nodes. [This screenshot of the signals showcases how 4 and c get dropped](https://puu.sh/IcOwM/b6e4782918.png)

lsteveol, 2021-09-21:
I the WSTRB on the bottom is 0 for the ones you miss?

michael-etzkorn, 2021-09-21:
Yes. WSTRB is going to 0 while the last nibble is '4' and 'c'. WSTRB is held to `f` on the slave port.  I don't know if `executable` is relevant here. It seems like it's more so for the device tree from what tracing of it I could do (it goes into `Resources` as `x` so its hard to follow after that).

lsteveol, 2021-09-21:
What are the beatbytes of the AXI slave node you have defined for the `regmap`

michael-etzkorn, 2021-09-21:
I should say I'm not using a regmap. It's a slave node that has a blackbox peripheral that implements its own regmap. I only pointed out gcd because it seems to show that 32-bit aligned addressing is possible. beatBytes is set to 4 since the slave is AXI Lite. Maybe it has to do with using `AddressSet.misaligned` instead of `List(AddressSet())`?

michael-etzkorn, 2021-09-21:
```scala
sbus.toSlave(Some(portName)){
      (myPeripheralTop.regCfgSlv 
        := AXI4Buffer() 
        := AXI4UserYanker(capMaxFlight=Some(2)) 
        := TLToAXI4()
        := TLFragmenter(4, p(CacheBlockBytes), holdFirstDeny = true)  
        := TLWidthWidget(sbus.beatBytes)
        // := TLSourceShrinker(2) )
    }
```
Here's how it is connected to the SBUS. I'm looking into what widget loses / modifies WDATA from slave port to the slave node.

lsteveol, 2021-09-21:
What is `CahceBlockBytes` in this case?

michael-etzkorn, 2021-09-21:
I believe it's 64. Maybe setting that to 32 would fix things.

lsteveol, 2021-09-21:
`CacheBlockBytes` is generally used by the MBUS. Can you try replacing the `p(CacheBlockBytes)` with `sbus.beatBytes`

michael-etzkorn, 2021-09-21:
It seems that the `WidthWidget` is what sets the non-aligned a-channel data to 0. `sub.beatBytes` is 8, so I get the same error about not supporting TransferSizes[1,64]

michael-etzkorn, 2021-09-24:
I figured it might be the `aligned` parameter in the AXI4MasterPortParams, but that didn't do anything. Still dropping data for 32-bit aligned addresses that aren't 64-bit aligned 🙃

lsteveol, 2021-09-24:
Can you post all of the code related to this path? This acts so much like the parameters believe you are talking to a 64bit bus, but someone it's only generating 32bits of data.

michael-etzkorn, 2021-09-24:
I've posted the issue here now. https://github.com/chipsalliance/rocket-chip/issues/2883

I think the relevant code from WidthWidget.scala is
```scala
      val keepBits  = log2Ceil(inBytes)
      val dropBits  = log2Ceil(outBytes)
      val sel = in.bits match {
        case a: TLBundleA => a.address(keepBits-1, dropBits)
        case b: TLBundleB => b.address(keepBits-1, dropBits)
        case c: TLBundleC => c.address(keepBits-1, dropBits)
        case d: TLBundleD => {
          val sel = sourceMap(d.source)
          val hold = Mux(first, sel, RegEnable(sel, first)) // a_first is not for whole xfer
          hold & ~limit // if more than one a_first/xfer, the address must be aligned anyway
        }
      }

      val index  = sel | count
      def helper(idata: UInt, width: Int): UInt = {
        val mux = VecInit.tabulate(ratio) { i => idata((i+1)*outBytes*width-1, i*outBytes*width) }
        mux(index)
      }

      out.bits := in.bits
      out.valid := in.valid
      in.ready := out.ready

      // Don't put down hardware if we never carry data
      edgeOut.data(out.bits) := (if (edgeIn.staticHasData(in.bits) == Some(false)) 0.U else helper(edgeIn.data(in.bits), 8))

```
The WidthWidget decides to select the upper bits using address(keepBits, dropBits-1). Here, keepBits is `log2Ceil(8)=3` and dropBits is `log2Ceil(4) = 2` so it uses the 2nd LSB address bit to determine which data to keep. This seems odd to me because it'll choose the top half whenever a 32-bit access occurs. I think the way to resolve this (and how I've done so in the verilog) is to simply always choose the lower half. I'll probably modify the sel match case to do the same thing. If it causes any side effects, Ill see it as I advance in the verification process.

closed 2021-09-24 by michael-etzkorn

lsteveol, 2021-09-25:
did you figure it out?

michael-etzkorn, 2021-09-25:
The width widget is expecting a 64 bit A channel to transfer data in the upper 32 bits of the A channel's data line if the A-channel address is offset by 4. So, I at least have isolated the problem to the Width Widget's expectation of writes.

Right now I'm just always passing the lower data (but this might break store word instructions to the peripheral from the core). I think a better solution may be setting the beat bytes on the slave port to 4 though that might break this require in Width Widget:

```
require (edgeOut.manager.beatBytes > edgeIn.manager.beatBytes)
```

Either way the answer to my question is more so: the WidthWidget does not expect 32 bit aligned addresses with data in their lowerhalf.

lsteveol, 2021-09-26:
Can you try to put an AXI fragmenter instead of a TLFragmenter?

michael-etzkorn, 2021-09-26:
I'm skeptical this would work because I believe the fragmentation has to happen before switching to the AXI4 protocol (which is still TL).  Either way, the Width Widget would send over the upper-word in the 64-bit A channel data line. 
Assuming your suggestion is: 
```
sbus.toSlave(Some(portName)){
    (myPeripheralTop.regCfgSlv 
      := AXI4Buffer() 
      := AXI4Fragmenter()
      := AXI4UserYanker(capMaxFlight=Some(2)) 
      := TLToAXI4()
      := TLWidthWidget(sbus.beatBytes)
      // := TLSourceShrinker(2)
     )
  }
```

Because I don't believe the AXI4Fragmenter can process TL channels.