ucb-bar/chipyard #996: How to add QSFP1 node and QSFP2 node in vcu118's testharness?

zhuxuanlong, 2021-09-30:
**Impact**: rtl | software | unknown | other

**Description**

**What is a motivating example for changing the behavior?**

Hi,
`QSFP1VCU118PlacedOverlay` and `QSFP2VCU118PlacedOverlay` have been implemented in `VCU118ShellBasicOverlays`. However, the related node is not created in the testharness. For ethernet, how to connect to sbus? How to perform IOBinders and HarnessBinders of QSFP1 and QSFP2?
Thank you.

abejgonzalez, 2021-10-04:
Not all Overlays that are present in the `VCU118ShellBasicOverlays` have been instantiated in the Chipyard `TestHarness` (the default `TestHarness` is setup to boot Linux with the minimal set of Overlays). You can create a new `TestHarness` module (like the bringup version) with the Overlays you want added. 

> For ethernet, how to connect to sbus?

This is a bit trickier. In Chipyard we made the decision to break the diplomatic graph spanning `ChipTop` and the `TestHarness`, thus you have to expose a TL port to the SBUS through the ChipTop. The best thing to look at is the following area that instantiates the DDR:

**FPGA prototyping code (TestHarness):**

https://github.com/ucb-bar/chipyard/blob/9d055fdac638ab90735cbde42fd2d86355eb260b/fpga/src/main/scala/vcu118/TestHarness.scala#L81-L93

**FPGA prototyping code (HarnessBinder):**

https://github.com/ucb-bar/chipyard/blob/9d055fdac638ab90735cbde42fd2d86355eb260b/fpga/src/main/scala/vcu118/HarnessBinders.scala#L33-L45

**Exposing the TL memory port to the FPGA area:**

https://github.com/ucb-bar/chipyard/blob/9d055fdac638ab90735cbde42fd2d86355eb260b/generators/chipyard/src/main/scala/System.scala#L52-L90

https://github.com/ucb-bar/chipyard/blob/9d055fdac638ab90735cbde42fd2d86355eb260b/fpga/src/main/scala/vcu118/IOBinders.scala#L46-L52

In the `CanHaveMasterTLPort` trait, a connection to the MBUS is created and its IOs are exposed in the IOBinder. Then the HarnessBinder connects that IO exposed out of ChipTop to the TestHarness (which in turn connects to the Overlay).

You would have to do something similar on your end.

zhuxuanlong, 2021-10-08:
> Not all Overlays that are present in the `VCU118ShellBasicOverlays` have been instantiated in the Chipyard `TestHarness` (the default `TestHarness` is setup to boot Linux with the minimal set of [...]

Thank you for your reply. The example of vc707pcie enables the pcie module to interconnect with sbus and ibus. However, this trait does not show how to implement its harnessbinder or iobinder interface.

`trait HasSystemXilinxVC707PCIeX1 { this: BaseSubsystem =>
  val xilinxvc707pcie = LazyModule(new XilinxVC707PCIeX1)
  private val cname = "xilinxvc707pcie"
  sbus.coupleFrom(s"master_named_$cname") { _ :=* TLFIFOFixer(TLFIFOFixer.all) :=* xilinxvc707pcie.crossTLOut(xilinxvc707pcie.master) }
  sbus.coupleTo(s"slave_named_$cname") { xilinxvc707pcie.crossTLIn(xilinxvc707pcie.slave) :*= TLWidthWidget(sbus.beatBytes) :*= _ }
  sbus.coupleTo(s"controller_named_$cname") { xilinxvc707pcie.crossTLIn(xilinxvc707pcie.control) :*= TLWidthWidget(sbus.beatBytes) :*= _ }
  ibus.fromSync := xilinxvc707pcie.crossIntOut(xilinxvc707pcie.intnode)
}`

Can I refer to this example? Because the network port is different from ddr. It may be connected to sbus as client or mater.