ucb-bar/chipyard #1026: Simulation with varilator will triger assertion failed while simu拉提哦那、

ddxoxpp, 2021-11-01:


jerryz123, 2021-11-01:
Can you upload or send me the verilator waveform of the failing case?

ddxoxpp, 2021-11-02:
> Can you upload or send me the verilator waveform of the failing case?

Hi Jerry, I have sent the waveform to you by email.

ddxoxpp, 2021-11-02:
And it won't trigger assertion failed  with default RocketConfig with verilator simulation

jerryz123, 2021-11-02:
Is there another way to send me the failing waveform? I was unable to access the file you uploaded.

ddxoxpp, 2021-11-02:
The waveform is [https://drive.google.com/file/d/1UMWzUThMc3RqQg56Sy8zdJdFRTl_S0pZ/view?usp=sharing](url)
And the generated src is [https://drive.google.com/file/d/13Z8Szhlnvt9c6KOivQUrSs1o14EU72FK/view?usp=sharing](url)

ddxoxpp, 2021-11-02:
I share this by google drive：
The waveform is https://drive.google.com/file/d/1UMWzUThMc3RqQg56Sy8zdJdFRTl_S0pZ/view?usp=sharing
And the generated src is https://drive.google.com/file/d/13Z8Szhlnvt9c6KOivQUrSs1o14EU72FK/view?usp=sharing

ddxoxpp, 2021-11-04:
How is the progress，Jerry？ I have looked at waveform and the dcache's prober source id is 8 which is right when the config is MegaBoomConfig.

jerryz123, 2021-11-04:
What commit of boom are you using?

jerryz123, 2021-11-04:
Please apply the patch described here: https://github.com/riscv-boom/riscv-boom/issues/459

It has been fixed on a dev branch, but we need to do more testing before updating.

closed 2021-11-04 by jerryz123