ucb-bar/chipyard #1037: How to compile instructions for Hwacha?

haeunlee22, 2021-11-08:
Hello,

Since website for Hwacha has been deprecated, I thought maybe here is right place to ask questions about Hwacha.

It is written in website below
https://chipyard.readthedocs.io/en/latest/Generators/Hwacha.html

To run "./scripts/build-toolchains.sh esp-tools" to install toolchains for Hwacha.

As far as I understand, we need to run "./scripts/build-toolchains.sh esp-tools" and install risc-v-gnu-toolchain for cross compilation.

What confuses me is that, how does command for compiling Hwacha custom instructions and standard instructions differ?

Thank you.

a0u, 2021-11-08:
The esp-tools toolchain includes a patched version of binutils that adds support for the Hwacha custom extension:
https://github.com/ucb-bar/esp-binutils-gdb/tree/riscv-binutils-2.35

To build Hwacha assembly code, you must pass `-march=rv64gcxhwacha` to `gcc` or GNU `as`, as demonstrated by the Makefile for compiling the example Hwacha benchmarks:
https://github.com/ucb-bar/esp-tests/blob/master/benchmarks/Makefile#L56

haeunlee22, 2021-11-09:
Thank you for your answer.
I am a little confused though. There is riscv-tools directory under toolchians directory. Am I supposed to install riscv-gnu-toolchain at riscv-tools directory as well? I have tried several approaches which didn't yet work.

1) At first attempt, I install riscv-gnu-toolchain outside of chipyard repository and ran ./scripts/build-toolchains.sh esp-tools
- In this case I faced a linking issue when I typed "make linux", which was about cannot find lstdc++ and lgcc_s
- I tried to solve this issue for few days, but ended up concluding it is caused by having multiple riscv-gnu-toolchains installed. I couldn't solve this issue.

2) At second attempt, I installed chipyard repository and ran ./scripts/build-toolchains.sh esp-tools.
- Then I ran configure and make at 'chipyard/toolchains/esp-tools/riscv-gnu-toolchain/riscv-binutils' as you replied.
- However, in this case I can't find riscv64-unknown-elf-gcc at chipyard/esp-tools-install/bin, but I can find other files such as riscv64-unknown-elf-gdb. (I can't find riscv64-unknown-elf-gcc anywhere at chipyard directory)

So, I think the proper order might be like below but isn't so sure.

```
1. Clone chipyard directory
2. Go to chipyard/toolchains/riscv-tools/riscv-gnu-toolchain and do riscv-gnu-toolchain setup
3. Run ./scripts/build-toolchains.sh esp-tools at chipyard directory.
4. Go to chipyard/toolchains/esp-tools/riscv-gnu-toolchain/riscv-binutils and run configure & make as documented.
```

Is there anything I am thinking wrong?

Thank you in advance :)

haeunlee22, 2021-11-09:
Thank you very much!

closed 2021-11-09 by haeunlee22

ashutgupta28, 2023-05-10:
@haeunlee22 @a0u 
I tried to run C code on hawacha compiler. 
First I installed chipyard repo from: https://chipyard.readthedocs.io/en/stable/Chipyard-Basics/Initial-Repo-Setup.html 
and also ran ./build-setup.sh esp-tools from the script.

I compiled my C code by giving `-march=rv64gcxhwacha` to `riscv64-unknown-linux-gcc`. and my code got compiled . But when I am trying to execute the file it is showing me illegeal instruction.

The command I gave to compile my C code is `riscv64-unknown-linux-gcc -march=rv64gcxhwacha example.c`. 
After this the a.out executable is generated. But couldnot execute this. 

What should I do to compile my C code on Hawacha and then execute on RISC V.

zhongpanwu, 2024-05-13:
@asb HI, do you solve the problem, please let me, I am facing the same issue. Thanks for any reply