ucb-bar/chipyard #1042: Hwacha cannot execute after compilation

haeunlee22, 2021-11-13:
Hello,

I have set up riscv64-unknown-linux-gnu-gcc and riscv64-unknown-elf-gcc using toolchians provided. I succeed in compilation, however I fail in executing produced a.out.

For exmaple, assume I have following code hello.c.

#include <stdio.h>

int main() {
    printf("A\n");
    asm("vgetcfg 3, 1"); // or asm("li a4, 0x8000000000000203"); asm("vgetcfg a4");  as in esp-test example
    printf("B\n");
}

I can compile above code with "riscv64-unknown-linux-gnu-gcc -march=rv64gcxhwacha -static hello.c" and compilation works fine and I get a.out file. However, a.out fails to execute in various conditions I have tried. 

1. If I execute "spike --extension=hwacha pk a.out", it gets me "An illegal instruction was executed!". It seems like I get illegal instruction error everytime I try to execute hwacha vector instructions. I tried some other c codes, but they work fine. 

2. I have also tried to execute it using fedora.base image at firesim, but here I also get illegal instruction exceptions.

After all these attempts, I am not sure whether there is any working way to simulate programs with hwacha instructions. Do I need any more options for compilation or linking? 
Any help would really be appreciated.
Many thanks in advance.

a0u, 2021-11-13:
To enable Hwacha and other RoCC instructions, supervisor software must write a non-zero value to the `XS` field of the `mstatus` or `sstatus` CSRs.

This is handled during crt initialization for the esp-tests benchmarks:
https://github.com/ucb-bar/esp-tests/blob/7aa38c892b8e172dccb0e43dafa58cd09816c868/benchmarks/common/crt.S#L50-L52

For `pk`, you will have to patch `mstatus_init()` as follows and then rebuild by running `make` in `toolchains/esp-tools/riscv-pk/build`.

```diff
--- a/machine/minit.c
+++ b/machine/minit.c
@@ -34,6 +34,7 @@ static void mstatus_init()
   if (supports_extension('V'))
     mstatus |= MSTATUS_VS;
 
+  mstatus |= MSTATUS_XS;
   write_csr(mstatus, mstatus);
```

You will also need a similar patch for the Linux kernel (firesim/linux@93d7ca94dcb6a5d663d4fcccbc1764b3250a77d9), but this should already be applied for kernels built by FireMarshal.  Just be aware that Linux does not implement context switching of accelerator state, and Hwacha presently cannot handle restarting from page faults, so all pages must be guaranteed to be resident using `mlockall(2)`.

haeunlee22, 2021-11-14:
Thank you for you answer!

As  far as I understand, I need to add "mstatus |= MSTATUS_XS;" at minit.c of chipyard/toolchains/esp-tools/riscv-pk and rebuild pk. I tried to rebuild pk and spike, however following code below still causes an error. 

```
#include <stdio.h>

int main() {
  asm("li a4, 0x8000000000000203");
  asm("vsetcfg a4");
  printf("Hello RISCV!\n");
  return 0;
}
```

I am not sure whether "asm("vsetcfg 3, 1");" or two lines of code above are correct examples of vsetcfg (I just copied examples from test repo or ISA manual). However, it fails to run, and other instructions like asm("vadd vv2, vv0, vv1") also cannot run I think because vector configuration is not set. I think it's not problem with my usage of vsetcfg but I am not very sure. Do you think there might be any other problems? I will attach my output below. 

```
user@x10:~/Share_Local/user$ riscv64-unknown-linux-gnu-gcc -march=rv64gcxhwacha -static hello.c
user@x10:~/Share_Local/user$ spike --extension=hwacha pk a.out
bbl loader
z  0000000000000000 ra 00000000000105b8 sp 0000003ffffffa00 gp 000000000006d110
tp 000000000006e790 t0 000000000006df60 t1 0000000000000000 t2 000000000006df08
s0 0000003ffffffa10 s1 0000000000010988 a0 0000000000000001 a1 0000003ffffffb58
a2 0000003ffffffb68 a3 0000000000000000 a4 0000000000000203 a5 00000000000103f4
a6 000000000006d238 a7 0000000000000000 s2 0000000000000000 s3 0000000000000000
s4 0000000000000000 s5 0000000000000000 s6 0000000000000000 s7 0000000000000000
s8 0000000000000000 s9 0000000000000000 sA 0000000000000000 sB 0000000000000000
t3 000000000000002e t4 000000000000009f t5 0000000000000000 t6 000000000006df28
pc 0000000000010400 va/inst 000000000007200b sr 8000000200006020
An illegal instruction was executed!
```

Thank you!

a0u, 2021-11-14:
`sr 8000000200006020` indicates that `XS` has not been set, as bits 15 and 16 are both clear (not `0x800000020001e020`).

If you are rebuilding manually, check that you have reinstalled `pk` to the toolchain location (`${RISCV}/riscv64-unknown-elf/bin/pk`) by running `make install`.

haeunlee22, 2021-11-16:
Thank you!
Adding "mstatus |= MSTATUS_XS;" at minit.c and rebuilding solved the issue!
Now I can finally test programs with Hwacha instructions :)

haeunlee22, 2021-11-17:
I am sorry to have reopened the issue.
I am having trouble executing the program I have tested in spike on firesim run farm. I am running two programs, one with hwacha instruction and one without in exactly same configuration. The one without hwacha instruction finishes quickly in about 10 minutes, while one with hwacha instructions takes more than several hours and does not seem to terminate. I think fedora I have used might not be able to execute RoCC instructions. Below pictures are how my directory is configured and contents of my workload configuration, hwacha_fedora.json. Do I need to use base image other than "fedora-base.json" or do I need to feed some special options when building workload? I will also attach commands I have run on firesim.

[hwacha_fedora.json]
[image]

[Files in workloads directory]
[image]

[Commands executed]
[image]

Also, I tried to comment out file "~/firesim/target-design/chipyard/software/firemarshal/boards/default/linux/arch/riscv/kernel/process.c" like below, yet after rebuilding workloads, RoCC instructions seemed not to be available.

```
void start_thread(struct pt_regs *regs, unsigned long pc,
        unsigned long sp)
{
        regs->status = SR_PIE;
//#ifdef CONFIG_RISCV_ROCC
        regs->status |= SR_XS_INITIAL;
//#endif
        if (has_fpu) {
                regs->status |= SR_FS_INITIAL;
                /*
                 * Restore the initial value to the FP register
                 * before starting the user program.
                 */
                fstate_restore(current, regs);
        }
        regs->epc = pc;
        regs->sp = sp;
        set_fs(USER_DS);
}
```

Does it mean I cannot use fedora-base.json with hwacha instructions?

Thank you very much.

closed 2021-11-18 by haeunlee22