ucb-bar/chipyard #1048: Difficulties punching out a signal to tie to an LED

michael-etzkorn, 2021-11-17:
I have this signal I'd like to tie to an LED. In my trait, I connect the signal to an `IO(Output(Bool())`. 
```scala
val myModuleOpt = p(myModuleKey).map { k => 
  // More code here

  val led = InModuleBody {
            val led = IO(Output(Bool()))
            led := myLazyModule.module.led 
            led 
           }
  led 
}
```

I have the harness pin tied to led0 in my harness implementation:
```scala
    val led_io = IO(Output(Bool()))
    _outer.xdc.addPackagePin(led_io, "AT32")
    _outer.xdc.addIOStandard(led_io, "LVCMOS12")
```

I then try to connect the two using io and harness binders:

```scala
class WithLEDIO extends ComposeIOBinder({
    (system: CanHaveMyModule) => {
         val led : Seq[Bool] = 
         system.myModuleOpt.map({ n =>  
            val led = IO(Output(Bool()))
            led := n
            led
         }).toSeq
        (led, Nil) 
    }
})
```

```scala
class WithLED extends ComposeHarnessBinder({
  (system: CanHaveMyModule, th: BaseModule with HasHarnessSignalReferences, led: Seq[Bool] ) => {
    th match { case vcu118th: myVCU118FPGATestHarnessImp => {
      vcu118th.led_io := led.head // also tried led(0)
    }}
  }
})
```

But this causes a null pointer exception in the harness binder. I've managed to punch out signals like this before, but only when placed in a bundle and wrapped in ClockedIO. Is it possible to punch out a single signal like this? Do I need to use `ComposeIOLazyBinder`. I'll keep looking over the ComposeIOBinder implementation and see if I can piece together what I need to do. Any help is appreciated!

jerryz123, 2021-11-17:
Try 'led.foreach { l => vcu118th.led := l }'

michael-etzkorn, 2021-11-17:
I think it has to do with the fact 
```scala
  _outer.topDesign match { case d: HasTestHarnessFunctions =>
    d.harnessFunctions.foreach(_(this))
  }
```
in `VCU118FPGATestHarnessImp` evaluates before `led_io` is instatiated...

jerryz123, 2021-11-17:
Hmm, can you paste the exception message?

There is a code smell here though, in that the IOBinder/HarnessBinder stuff expects that the diplomatic graph should not extend beyond the DUT, but the VCU118 TH violates that. Not sure if this is the cause of what you are seeing though,

michael-etzkorn, 2021-11-17:
```scala

[error] java.lang.NullPointerException
[error]         ...
[error]         at chipyard.fpga.vcu118.Prototype.WithLED$$anonfun$$lessinit$greater$12.$anonfun$new$34(HarnessBinders.scala:159)
[error]         at chipyard.fpga.vcu118.Prototype.WithLED$$anonfun$$lessinit$greater$12.$anonfun$new$34$adapted(HarnessBinders.scala:157)
[error]         at chipyard.harness.ComposeHarnessBinder$$anonfun$$lessinit$greater$4.$anonfun$new$1(HarnessBinders.scala:73)
[error]         at chipyard.harness.ComposeHarnessBinder$$anonfun$$lessinit$greater$4.$anonfun$new$1$adapted(HarnessBinders.scala:71)
[error]         at chipyard.harness.HarnessBinder$$anonfun$$lessinit$greater$2$$anonfun$apply$2.$anonfun$applyOrElse$1(HarnessBinders.scala:56)
[error]         at chipyard.harness.HarnessBinder$$anonfun$$lessinit$greater$2$$anonfun$apply$2.$anonfun$applyOrElse$1$adapted(HarnessBinders.scala:49)
[error]         at chipyard.harness.ApplyHarnessBinders$.$anonfun$apply$1(HarnessBinders.scala:40)
[error]         at chipyard.harness.ApplyHarnessBinders$.$anonfun$apply$1$adapted(HarnessBinders.scala:39)
[error]         at scala.collection.Iterator.foreach(Iterator.scala:941)
[error]         at scala.collection.Iterator.foreach$(Iterator.scala:941)
[error]         at scala.collection.AbstractIterator.foreach(Iterator.scala:1429)
[error]         at scala.collection.IterableLike.foreach(IterableLike.scala:74)
[error]         at scala.collection.IterableLike.foreach$(IterableLike.scala:73)
[error]         at scala.collection.AbstractIterable.foreach(Iterable.scala:56)
[error]         at chipyard.harness.ApplyHarnessBinders$.apply(HarnessBinders.scala:39)
[error]         at chipyard.fpga.vcu118.VCU118FPGATestHarnessImp.<init>(TestHarness.scala:137)
[error]         at chipyard.fpga.vcu118.Prototype.AXI4PCIeVCU118FPGATestHarnessImp.<init>(TestHarness.scala:227)
[error]         at chipyard.fpga.vcu118.Prototype.AXI4PCIeVCU118FPGATestHarness.module$lzycompute(TestHarness.scala:224)
[error]         at chipyard.fpga.vcu118.Prototype.AXI4PCIeVCU118FPGATestHarness.module(TestHarness.scala:224)
[error]         at chipyard.fpga.vcu118.Prototype.AXI4PCIeVCU118FPGATestHarness.module(TestHarness.scala:170)
[error]         at freechips.rocketchip.stage.phases.PreElaboration.$anonfun$transform$1(PreElaboration.scala:38)
[error]         ... (Stack trace trimmed to user code only, rerun with --full-stacktrace if you wish to see the full stack trace)
```
If I make `led_io` `lazy val` or move it to before the Harness Binders are applied in `TestHarness.scala` in `vcu118` (which I don't really want to do for maintanability reasons), I instead get:
```scala
[error] LazyModule.scala:278: Unable to name port Bool(IO <UNNAMED> in AXI4VCU118DigitalTop) in chipyard.fpga.vcu118Prototype.AXI4DigitalTopModule@23368e41, try making it a public field of the Module in class freechips.rocketchip.diplomacy.LazyModuleImpLike
[error] There were 1 error(s) during hardware elaboration.
[error] chisel3.internal.ChiselException: Fatal errors during hardware elaboration
[error]         ...
[error]         ... (Stack trace trimmed to user code only, rerun with --full-stacktrace if you wish to see the full stack trace)

Exception: sbt.TrapExitSecurityException thrown from the UncaughtExceptionHandler in thread "run-main-0"
```

I'd add `--full-stacktrace`, but having a hard time passing that to the `run_sbt` stuff in the makefile.

michael-etzkorn, 2021-11-17:
My feeling is bundle bridge sources might resolve this, but I'm still quite unfamiliar with them at this time.

jerryz123, 2021-11-17:
Are you sure `myModuleOpt.isDefined`? Are you building a config where it is set to `None`?

What happens if you do 
```
  if (led.size > 0) vcu118th.led_io := led.head
```

michael-etzkorn, 2021-11-18:
I'm certain myModuleOpt is defined. `led.size` is greater than 1. I think `led.head` isn't the null pointer. It's `vcu118th.led_io` that's excepting (unless `led_io` is lazy, which works, but then there's this naming issue)

jerryz123, 2021-11-18:
It is strange for `vcu118th.led_io` to be causing the null pointer exception, since it is just normal Chisel hardware in the vcu118th LazyModuleImp definition. 
Did you place the declaration of `led_io` before the call to `ApplyHarnessBinders`, which should also be in the body of the vcu118th LazyModuleImp definition?

jerryz123, 2021-11-18:
And making it led_io lazy does not strike me as the right solution here.

michael-etzkorn, 2021-11-18:
I'm extending `VCU118TestHarnessImp` which calls `ApplyHarnessBinders`, so `led_io` is after that. `Bringup`'s extension of `VCU118` does a similar trick of extending this harness. I can make it non-lazy if I put it directly in `vcu118/TestHarness.scala` before `ApplyHarnessBinders`, but I once again run into the "unable to name port" issue.

jerryz123, 2021-11-18:
Oh I see, I didn't realize you were extending the VCU118TestHarnessImp. I don't believe we support that flow future, although it seems like a reasonable thing to add support for in the future.

Does just applying the following patch cause the naming error? It seems to work for me, and is how I expect a new IO to be implemented (obviously missing the HarnessBinder to drive it, but I don't think the driver of the IO affects the naming issue?).

```diff --git a/fpga/src/main/scala/vcu118/TestHarness.scala b/fpga/src/main/scala/vcu118/TestHarness.scala
index ab6897c9..a0b51d72 100644
--- a/fpga/src/main/scala/vcu118/TestHarness.scala
+++ b/fpga/src/main/scala/vcu118/TestHarness.scala
@@ -101,6 +101,11 @@ class VCU118FPGATestHarnessImp(_outer: VCU118FPGATestHarness) extends LazyRawMod
   _outer.xdc.addPackagePin(reset, "L19")
   _outer.xdc.addIOStandard(reset, "LVCMOS12")
 
+  val led_io = IO(Output(Bool()))
+  _outer.xdc.addPackagePin(led_io, "AT32")
+  _outer.xdc.addIOStandard(led_io, "LVCMOS12")
+  led_io := false.B
+
   val resetIBUF = Module(new IBUF)
   resetIBUF.io.I := reset
 
```

michael-etzkorn, 2021-11-18:
The naming issue seems to have been resolved with `suggestName` in my module option. 
 
```scala
val myModuleOpt = p(myModuleKey).map { k => 
  // More code here

  val led = InModuleBody {
            val led = IO(Output(Bool())).suggestName("led_out")
            led := myLazyModule.module.led 
            led 
           }
  led 
}
```

It also seems I can place this in an extension of `VCU118FPGATestHarnessImp` if I make it a `lazy val`. I'm still new to Scala and Chisel so not sure on the repercussions of that, but thanks for helping me debug!

closed 2021-11-18 by michael-etzkorn