ucb-bar/chipyard #1049: Update QEMU for binutils 2.36 (Ubuntu 21.04)

allisonrandal, 2021-11-18:
**Impact**: software

**Description**
I recently hit the exact problem reported in this patch on the qemu-stable list, as I was compiling QEMU as part of Chipyard (at commit 5129bc7da5093f1d289c2fed27673cf51024eb0d) on Ubuntu 21.04:

https://lists.nongnu.org/archive/html/qemu-stable/2020-12/msg00015.html

> Recent binutils changes dropping unsupported options [1] caused a build
> issue in regard to the optionroms.
> 
>   ld -m elf_i386 -T /<<PKGBUILDDIR>>/pc-bios/optionrom//flat.lds -no-pie \
>     -s -o multiboot.img multiboot.o
>   ld.bfd: Error: unable to disambiguate: -no-pie (did you mean --no-pie ?)
> 
> This isn't really a regression in ld.bfd, filing the bug upstream
> revealed that this never worked as a ld flag [2] - in fact it seems we
> were by accident setting --nmagic).
> 
> Since it never had the wanted effect this usage of LDFLAGS_NOPIE, should be
> droppable without any effect. This also is the only use-case of
> LDFLAGS_NOPIE
> in .mak, therefore we can also remove it from being added there.
> 
> [1]: https://sourceware.org/git/?p=binutils-gdb.git;a=commit;h=983d925d
> [2]: https://sourceware.org/bugzilla/show_bug.cgi?id=27050#c5

The patch was applied in this commit: https://gitlab.com/qemu-project/qemu/-/commit/bbd2d5a8120771ec59b86a80a1f51884e0a26e53

And partially reverted in this commit: https://gitlab.com/qemu-project/qemu/-/commit/ffd205ef2901bd

Because Chipyard is pinning the `toolchains/qemu` git submodule at v5.0.0, it has neither of these fixes, but if Chipyard updates the git submodule to qemu v6.0.1, it will get both of these fixes.

To test the new version of QEMU, I updated the git submodule to v6.0.1 locally, and re-ran `scripts/build-toolchains.sh`. It gave me the following error and exited without building QEMU:

```
==>  PLEASE REMOVE qemu URL-REWRITING from scripts/build-toolchains.sh. It is no longer needed!
```

So, I removed the url rewriting (put in place to avoid git.qemu.org, which no longer exists) using the attached patch, and the toolchain build completes successfully.

[patch_drop_url_rewriting.txt](https://github.com/ucb-bar/chipyard/files/7564537/patch_drop_url_rewriting.txt)
:

**What is a motivating example for changing the behavior?**

QEMU v5.0.0 fails to compile Ubuntu 21.04 or any later version, because binutils 2.36 no longer accepts the invalid ld flag `-no-pie`.

DorianXGH, 2022-02-11:
In the same vein, QEMU v5.0.0 fails to build on archlinux because of an issue that was fixed a long time ago.
I think updating QEMU in this project is needed

jerryz123, 2022-02-11:
Thanks, can you open a PR with that patch?