ucb-bar/chipyard #1051: Rule of main memory access limitation (from DMA side)

crystalbreezy1, 2021-11-25:
Currently I have designed a DMA controller (as a MMIO device) and integrated it into the chipyard SOC. To test it, I have written a c program which reads data (with a specific size) from MMIO and write it to a specific main memory address. Since I have used a 256KB backing scratchpad as my main memory, the main memory address is 0x80000000 to 0x80040000.
What I first did is reading a specific size of data from MMIO and write he data from the beginning address 0x80000000, which first works fine when the data size is small. However, when I increase the transfer data size to 1024 bytes, I find that any memory read access command towards DMA MM registers after the transfer was stuck. By looking into the VCS signal behaviors, I found that the requested DMA transfer has been completed successfully, but the following memory read command cannot be observed (I have traced from DMA ports all the way back to system bus, still can't find it). However, if I increase the scratchpad size or change my starting address to 0x80020000, the 1024 byte transfer can be executed correctly. 
In this case, I wonder if the problem here is that I have accidentally accessd and modified the main memory region where CPU holds intermdiate variables for program execution (such as stacking). To avoid having this type of error again, may I know where is the rule that defines which part of the main memory will be used to hold these untouchable values? (I believe this should be related to the compiler, but still don't know where to find this information).

michael-etzkorn, 2021-12-09:
If you do an objdump with `riscv64-unknown-elf-objdump` and pass `-h` (or `--headers`), you should get a print out of the Size, Virtual Memory Address, and Load Memory Address of each section. You wouldn't want to write over `text` and writing over certain `bss` and `data` could influence your program. 

`riscv64-unknownelf-objdump -d -h DMAtest.elf > DMAtest.dis`

For example: 

```
Sections:
Idx Name          Size      VMA               LMA               File off  Algn
0 .text         00051346  0000000860000000  0000000860000000  00001000  2**8
                  CONTENTS, ALLOC, LOAD, READONLY, CODE
1 .text.startup 0000153c  0000000860051346  0000000860051346  00052346  2**1
                  CONTENTS, ALLOC, LOAD, READONLY, CODE
2 .rodata       00052dc0  0000000860062b00  0000000860062b00  00063b00  2**3
                  CONTENTS, ALLOC, LOAD, READONLY, DATA
3 .data         000050b4  0000000860100000  00000008600b61f8  000b8000  2**3
                  CONTENTS, ALLOC, LOAD, DATA
4 .stack0       0010000c  00000008601050b4  00000008601050b4  000bd0b4  2**0
                  ALLOC
5 .bss          03b7e910  00000008602050c0  00000008602050c0  000bd0b4  2**6
                  ALLOC
6.heap         00300000  0000000863d839d0  0000000863d839d0  000bd0b4  2**0
                  ALLOC
```
I've removed some sections for simplicity and my addressing is a bit funky, but hopefully this shows you how you can find the address of each section of memory.

crystalbreezy1, 2021-12-15:
That's helpful! Thank you very much!

closed 2021-12-15 by crystalbreezy1