ucb-bar/chipyard #1055: How to link another Iinker script and run on chipyard successfully?

Hsiang0816, 2021-12-09:
Hi,

I want to run the test suit in [https://github.com/CTSRD-CHERI/Test-Suite-Transient-Execution](url) with chipyard. 

But the test suit has its own linker script, like below:

```
OUTPUT_ARCH( "riscv" )

SECTIONS
{
  . = 0x80000000;
  .text.init : { *(.text.init) }
  . = ALIGN(0x1000);
  .text : { *(.text) }
  . = ALIGN(0x1000);
  .melt_data : { *(.melt_data) }
  . = ALIGN(0x1000);
  .table1 : { *(.table1) }
  . = ALIGN(0x1000);
  .table2 : { *(.table2) }
  . = ALIGN(0x1000);
  .melt_utext : { *(.melt_utext) }
  . = ALIGN(0x1000);
  .melt_udata : { *(.melt_udata) }
  . = ALIGN(0x20000);
  .acode : { *(.acode) }
  . = ALIGN(0x1000);
  .data : { *(.data) }
  . = ALIGN(0x4000000);
  .spec_btb_attack : { *(.spec_btb_attack) }
  . = ALIGN(0x1000);
  .utext : { *(.utext) }
  . = ALIGN(0x1000);
  .udata : { *(.udata) }
  . = ALIGN(0x1000);
  .tohost : { *(.tohost) }
  _end = .;
```
I Know if I want to run program successfully, It must follow the steps below:
```
$ riscv64-unknown-elf-gcc -fno-common -fno-builtin-printf -specs=htif_nano.specs -c hello.c
$ riscv64-unknown-elf-gcc -static -specs=htif_nano.specs hello.o -o hello.riscv
```
And this code also uses chipyard's linker script **htif.ld** .

So I tried the following steps:
```
$ riscv64-unknown-elf-gcc -fno-common -fno-builtin-printf -specs=htif_nano.specs -c melt_us.S
$ riscv64-unknown-elf-gcc melt_us.o -static -Triscv_link.ld -nostdlib -o melt_us.riscv
$ make CONFIG=SmallBoomConfig run-binary-debug BINARY=./c_code/melt_us.riscv
```
This is the output:
```
using random seed 1639298419
This emulator compiled with JTAG Remote Bitbang client. To enable, use +jtag_rbb_enable=1.
Listening on port 36413
== Loading device model file '/home/N26091526/chipyard/generators/testchipip/src/main/resources/dramsim2_ini/DDR3_micron_64M_8B_x4_sg15.ini' == 
== Loading system model file '/home/N26091526/chipyard/generators/testchipip/src/main/resources/dramsim2_ini/system.ini' == 
===== MemorySystem 0 =====
CH. 0 TOTAL_STORAGE : 4096MB | 1 Ranks | 16 Devices per rank
DRAMSim2 Clock Frequency =666666666Hz, CPU Clock Frequency=100000000Hz
warning: tohost and fromhost symbols not in ELF; can't communicate with target
```
And after a long time there was no result. So I think I made a mistake in a certain step.

Does anyone know how to solve this kind of problem? Thank you very much!

michael-etzkorn, 2021-12-11:
It does look like you have `.tohost` defined, but could try defining `.fromhost` as well to squash the FESVR warning. You could also try moving `.tohost` up in the linker script (perhaps between `.text` and `.text.init`). Because its possible the whole binary doesn't fit into memory. 

The linker script looks like it's expecting 1GB of data space, and I'm not sure if that much memory is used by default in the `SmallBoomConfig`. I'd also try the `SmalRocketConfig` since I know that one should print its PC on writebacks if it's started up (though I imagine Boom does this too).

Hsiang0816, 2021-12-13:
Hi @michael-etzkorn, I try to dump the elf file to assembly code with sections, like below: 
```
Sections:
Idx Name          Size      VMA               LMA               File off  Algn
  0 .text         000000b2  0000000080000000  0000000080000000  00001000  2**1
                  CONTENTS, ALLOC, LOAD, READONLY, CODE
  1 .melt_data    00000080  0000000080001000  0000000080001000  00002000  2**12
                  CONTENTS, ALLOC, LOAD, DATA
  2 .table1       00000018  0000000080002000  0000000080002000  00003000  2**12
                  CONTENTS, ALLOC, LOAD, DATA
  3 .table2       00000030  0000000080003000  0000000080003000  00004000  2**12
                  CONTENTS, ALLOC, LOAD, DATA
  4 .melt_utext   0000001e  0000000080004000  0000000080004000  00005000  2**12
                  CONTENTS, ALLOC, LOAD, READONLY, CODE
  5 .melt_udata   00000080  0000000080005000  0000000080005000  00006000  2**12
                  CONTENTS, ALLOC, LOAD, DATA
  6 .tohost       00000008  0000000080005080  0000000080005080  00006080  2**0
                  CONTENTS, ALLOC, LOAD, DATA
  7 .riscv.attributes 00000033  0000000000000000  0000000000000000  00006088  2**0
                  CONTENTS, READONLY
```
I tried to add `.fromhost` to the linker script, but because my source code does not use `.fromhost`, it does not seem to be loaded into the elf file. I also tried to move `.tohost` up in the linker script (perhaps between `.text` and `.text.init`), the problem is still not solved.

I try to dump the signal that executed on Boom with my source code, and I observed that the pc is stuck at 0x1006a without executing my source code.

michael-etzkorn, 2021-12-13:
Hmm, it doesn't look like the contents are that large. You can try passing `-static` in for linking. It seems like it's hanging around the `wfi` instruction which might be related to HTIF / FESVR. You can also try `make CONFIG=SmallBoomConfig debug` and then pass in the binary after? 

```
make CONFIG=SmallBoomConfig debug
./simv-SmallBoomConfig-debug +verbose melt_us.riscv
```

I doubt that fixes it though. There probably needs to be some merging of `htif.ld` with your linker script 

I know Abe and some others from Berkeley have worked on running meltdown and other attacks on BOOM, so maybe they'll see this and be able to give further suggestions.

michael-etzkorn, 2021-12-28:
>And this code also uses chipyard's linker script htif.ld .

Looking at this again, I still think it's a linker script related issue. You can try adding
```
    .htif ALIGN(0x40) : {
        *(.htif)
    }
```
like `htif.ld` does. And maybe include the `PROVIDE` lines that `htif.ld` does such as `PROVIDE (__boot_hart = 0);` -- this __boot_hart macro might be why the bootrom is stuck at wfi. 

The linker script you'll want to adapt from is located here: `chipyard/toolchains/libgloss/util/htif.ld`

Hsiang0816, 2021-12-30:
Thank you for your help!

I have tried to merge the htif.ld with my linker script, and it can run on BOOM successfully.

closed 2021-12-30 by Hsiang0816