ucb-bar/chipyard #1056: hwo to Instantiate the Dcache ?

dalingmeng, 2021-12-10:
Implicit parameters need to be passed when instantiating the Dcache module ，I don't know which implicit parameter should be used to instantiate Dcache module 。
      Which class of object should the passed implicit parameter be equal to ?
[image]

michael-etzkorn, 2021-12-11:
I've not tried directly emitting the DCache, but Diplomacy might make things a bit tricky here. 

As for how to populate `Parameters`, you can probably just use any of the Rocket Configs. 

Something like

```scala
// I'd use chipyard.RocketConfig, but this doesn't seem like you're working within chipyard?
val topP : Parameters = new freechips.rocketchip.system.DefaultConfig 
// Then pass topP into DCache module
```
The real potential problem here is: you are trying to instantiate the LazyModuleImp (`DCacheModule`) with a dynamically created LazyModule (the `new DCache`). I'm not sure if that's possible. 

You may need to do something like this:

```
       val topP : Parameters = new freechips.rocketchip.system.DefaultConfig
        val dcache = LazyModule(new DCache()(topP))
        (new chisel3.stage.ChiselStage).emitVerilog(
            dcache.module )
```

FYI: you'll probably need to import all of Diplomacy. 
```scala
   import freechips.rocketchip.diplomacy._
```

dalingmeng, 2021-12-13:
I I tried to use your method to instantiate a rocket core, but the compilation shows the following error, showing that there is no TileKey parameter, but this parameter is defined in the basetile file, and I have imported this package, I don’t know how to solve this Mistake.

[image]

michael-etzkorn, 2021-12-13:
Looks like `TileKey` gets set by the abstract Tile class, so it seems there's not a straightforward way to just emit the `DCache` as a stand-alone module. You may have better success generating the `ExampleRocketSystem` and then just looking at the `DCache` module in `ExampleRocketSystem.v`. If you want to pursue generating the standalone DCache further, you can try setting `TileKey` in a custom config extended off `DefaultConfig`. 

Something like

```scala
class AddTileKey extends Config((site, here, up) => {
  case TileKey => {
     TileParams(core = /* CoreParams here.  */
     dcache = Some(/* DCacheParams here*/ )
     hartId = 0
     clockSinkParams = /* ClockSinkParamsI -- maybe can remain uninitialized?*/ )
  } 
} 

class CustomConfig extends Config {
    new AddTileKey ++
    new DefaultConfig
} 
```

But I'm not even sure if I have the syntax right on the TileKey population or if this is even a valid approach (you'll likely run into more problems). Also, I don't even see where `DCache.scala` is using `TileKey` so its not like you can just override with `RocketTilesKey` there. 

In summary: I don't believe there is any out of the box support for generating just components of the core. You're better off trying to make a whole tile or a whole system and extracting the Verilog you want to look / test from that file.

dalingmeng, 2021-12-16:
我只打算使用Decade模块，不生成rocke核心和其他的模块 。现在我只想把Dcache模块从rocket核里面单独抽取出来，然后做些修改让它能被单独实例化。 我知道有些参数是传递的隐式参数，如果我把这些传递过来的隐式参数全部定义到Dcahe的文件里面，因此就不需要传递隐式参数了。 我知道这些隐式参数层层传递，看起来很复杂，我把相关的参数全部定义到Dcache中，这个方法可行么？

dalingmeng, 2021-12-16:
I only plan to use the Decade module and not generate the rocke core and other modules. Now I just want to extract the Dcache module separately from the rocket core, and then make some modifications so that it can be instantiated separately. I know that some parameters are passed implicit parameters. If I define all these passed implicit parameters in the Dcahe file, there is no need to pass implicit parameters. I know that these implicit parameters are passed layer by layer. It seems very complicated. I define all the relevant parameters in Dcache. Is this method feasible?

jerryz123, 2021-12-16:
I haven't tested this, but I believe if you run `make TOP=DCache`, it will split out the verilog for the dcache from everything else and dump that in a file in `generated-src`

jerryz123, 2023-02-17:
Closing due to inactivity. This flow is supported easily now with CIRCT.

closed 2023-02-17 by jerryz123