ucb-bar/chipyard #1059: VCU118：I can't boot boom on VCU118

pig-pig-yang, 2021-12-12:
I prepared the bitstream according to the documentation and used 2020.1 Vivado to load the bitstream file. Secondly, the br-base-nodisk-flat file was generated according to the instructions. After loading the SD card, the following problems occurred in the RESET FPAG UART.

`INIT
CMD0
sd_cmd: timeout
ERROR
`

michael-etzkorn, 2021-12-13:
Are you connecting a PMOD to microSD card converter to the board? The sd card is over the PMOD pins, not the sd card slot that comes with the board (that's only for the zynq system controller).

If you are using the PMOD pins, it might be some issue with the SD card. I know I had an SD card that no matter what didn't seem to be read by the FPGA bootrom.

pig-pig-yang, 2021-12-13:
I didn't connect a PMOD to microSD card converter to the board.  What should i prepare for it?
Thank you for your reply.

michael-etzkorn, 2021-12-13:
Yeah, the microSD card connects over PMOD pins. The relevant fpga-shells code is here https://github.com/sifive/fpga-shells/blob/f9fb9fd338e5fca2ff5116b1d01506c424280d70/src/main/scala/shell/xilinx/VCU118NewShell.scala#L56

What you'll need to do is buy a Pmod MicroSD. They cost about $10. I believe the one Diligent sells has been reliable.

pig-pig-yang, 2021-12-14:
I had bought a microSD card connects over PMOD pins. However, It didn't turn out well.  I couldn't figure out what the problem was.
[image]
The terminal stops output at this point.

pig-pig-yang, 2021-12-14:
[image]

jamesdunn, 2021-12-16:
I replied to your question on the Chipyard mailing list but will echo here. We can close this issue and continue discussion on the mailing list.

My hunch is that the SD card isn't being formatted correctly. The partitions are Apple HFS/APFS when I would expect them to be ext4. [This link](https://www.youtube.com/watch?v=_xWCz1JJwTg&t=1h50m38s) describes the process for formatting the card.

closed 2021-12-16 by jamesdunn

michael-etzkorn, 2021-12-16:
> I replied to your question on the Chipyard mailing list but will echo here. We can close this issue and continue discussion on the mailing list.
>
> My hunch is that the SD card isn't being formatted [...]

These are actually the recommended file systems from [the documentation](https://chipyard.readthedocs.io/en/latest/Prototyping/VCU118.html). Unless I'm missing something in the screenshot, partitioning with the apple filesystems in this way should be readable by the FPGA bootrom. I'd have to double-check my SD cards, but I think they all use APFS for the first partition and HFS for the second partition.

zld932, 2021-12-27:
I have the same problem with you. Have you solved it?

Yuxin-Yu, 2022-08-09:
> I have the same problem with you. Have you solved it?

Hi,Bro.I got the samle problem.Have you solved it ?

Yuxin-Yu, 2022-08-09:
> I have the same problem with you. Have you solved it?

I have the same problem with you. Have you solved it?

Yuxin-Yu, 2022-08-09:
> I prepared the bitstream according to the documentation and used 2020.1 Vivado to load the bitstream file. Secondly, the br-base-nodisk-flat file was generated according to the instructions. After loading the SD card, the following problems occurred in the RESET FPAG.
> 
> `INIT CMD0 sd_cmd: timeout ERROR ` [image]

I have the same problem with you. Have you solved it?

pig-pig-yang, 2022-08-10:
> I have the same problem with you. Have you solved it?

Whether you purchased Pmod MicroSD, as well as check your pin allocation

pig-pig-yang, 2022-08-10:
> 

https://github.com/sifive/fpga-shells/blob/f9fb9fd338e5fca2ff5116b1d01506c424280d70/src/main/scala/shell/xilinx/VCU118NewShell.scala#L56-L61

pig-pig-yang, 2022-08-10:
> Whether you purchased Pmod MicroSD, as well as check your pin allocation
https://github.com/sifive/fpga-shells/blob/f9fb9fd338e5fca2ff5116b1d01506c424280d70/src/main/scala/shell/xilinx/VCU118NewShell.scala#L56-L61

leduchuybk, 2022-09-05:
@pig-pig-yang i met same problem. after i changed the pin cs and mosi on the sdcard module extension. it works. from what i saw in the code: fpga-shells/src/main/scala/shell/xilinx/SDIOOverlay.scala line 39 and 40.
`UIntToAnalog(sd_spi_cs, io.spi_dat(3), true.B)
UIntToAnalog(sd_spi_dq(0), io.spi_cs, true.B)`
they switched these pin. i do not know the reason why. but it seems that these pin are wrongly assigned

michael-etzkorn, 2022-09-05:
It naturally will depend on your pmod sd card module. The PMOD microSD card module from Digilent works without needing to switch any pins. If you purchashed from a different vendor, check your schematic and align the pins correctly in the overlay. 

Another important point mentioned upstream: the VCU118's SD card slot is not the slot connected the SDIO (it's for some board management, it has no pins connected to the FPGA). The SD card should be connected over the female PMOD connector.

michael-etzkorn, 2022-10-11:
Are you connecting to an SD card over the PMOD pins? The built-in sd card
slot on the vcu118 is not for the FPGA.

Did you make sure your binary starts at sector 34?

jiayang-douding, 2025-05-27:
Hello, I also encountered the same issue when booting Linux on the VCU118. Have you resolved it? Below is the log output from my serial port.

[image]

[image]