ucb-bar/chipyard #1065: Error while generating Arty bitstream

12ff7a6, 2021-12-21:
Hello,

I'm getting errors while generating Arty bitstream and I'm using docker image. can someone give me a hint?

> root@0daba79434b7:~/chipyard/fpga# make SUB_PROJECT=arty bitstream   
Running with RISCV=/root/chipyard/esp-tools-install
mkdir -p /root/chipyard/fpga/generated-src/chipyard.fpga.arty.ArtyFPGATestHarness.TinyRocketArtyConfig
cd /root/chipyard && java -Xmx8G -Xss8M -XX:MaxPermSize=256M -Djava.io.tmpdir=/root/chipyard/.java_tmp -jar /root/chipyard/generators/rocket-chip/sbt-launch.jar -Dsbt.sourcemode=true -Dsbt.workspace=/root/chipyard/tools  ";project fpga_platforms; runMain chipyard.Generator  --target-dir /root/chipyard/fpga/generated-src/chipyard.fpga.arty.ArtyFPGATestHarness.TinyRocketArtyConfig --name chipyard.fpga.arty.ArtyFPGATestHarness.TinyRocketArtyConfig --top-module chipyard.fpga.arty.ArtyFPGATestHarness --legacy-configs chipyard.fpga.arty:TinyRocketArtyConfig"
OpenJDK 64-Bit Server VM warning: Ignoring option MaxPermSize; support was removed in 8.0
[info] welcome to sbt 1.4.9 (Ubuntu Java 11.0.11)
[info] loading settings for project chipyard-build from plugins.sbt ...
[info] loading project definition from /root/chipyard/project
[info] loading settings for project chipyardRoot from build.sbt ...
[info] loading settings for project hardfloat from build.sbt ...
[info] loading settings for project rocketConfig from build.sbt ...
[info] loading settings for project firrtl_interpreter from build.sbt ...
[info] loading settings for project treadle from build.sbt ...
[info] loading settings for project chisel_testers from build.sbt ...
[info] loading settings for project testchipip from build.sbt ...
[info] loading settings for project icenet from build.sbt ...
[info] loading settings for project hwacha from build.sbt ...
[info] loading settings for project boom from build.sbt ...
[info] loading settings for project cva6 from build.sbt ...
[info] loading settings for project sodor from build.sbt ...
[info] loading settings for project gemmini from build.sbt ...
[info] loading settings for project mdf from build.sbt ...
[info] loading settings for project barstoolsMacros from build.sbt ...
[info] loading settings for project sim-build from plugins.sbt ...
[info] loading project definition from /root/chipyard/sims/firesim/sim/project
[info] loading settings for project firesim from build.sbt ...
[info] loading settings for project targetutils from build.sbt ...
[info] loading settings for project midas from build.sbt ...
[info] loading settings for project chisel3-build from plugins.sbt ...
[info] loading project definition from /root/chipyard/tools/chisel3/project
[info] loading settings for project chisel from build.sbt ...
[info] loading settings for project firrtl-build from plugins.sbt ...
[info] loading project definition from /root/chipyard/tools/firrtl/project
[info] loading settings for project firrtl from build.sbt ...
[info] resolving key references (55373 settings) ...
[info] set current project to chipyardRoot (in build file:/root/chipyard/)
[info] set current project to fpga_platforms (in build file:/root/chipyard/)
[info] compiling 69 Scala sources to /root/chipyard/tools/treadle/target/scala-2.12/classes ...
[error] /root/chipyard/tools/treadle/src/main/scala/treadle/Driver.scala:6:8: object ExecutionOptionsManager is not a member of package firrtl
[error] import firrtl.{AnnotationSeq, ExecutionOptionsManager, HasFirrtlOptions}
[error]        ^
[error] /root/chipyard/tools/treadle/src/main/scala/treadle/Driver.scala:32:20: type ComposableOptions is not a member of package firrtl
[error]     extends firrtl.ComposableOptions {
[error]                    ^
[error] /root/chipyard/tools/treadle/src/main/scala/treadle/Driver.scala:34:41: not found: type ExecutionOptionsManager
[error]   def vcdOutputFileName(optionsManager: ExecutionOptionsManager): String = {
[error]                                         ^
[error] /root/chipyard/tools/treadle/src/main/scala/treadle/Driver.scala:73:9: not found: type ExecutionOptionsManager
[error]   self: ExecutionOptionsManager =>
[error]         ^
[error] /root/chipyard/tools/treadle/src/main/scala/treadle/Driver.scala:77:3: not found: value parser
[error]   parser.note("treadle-options")
[error]   ^
[error] /root/chipyard/tools/treadle/src/main/scala/treadle/Driver.scala:79:3: not found: value parser
[error]   parser
[error]   ^
[error] /root/chipyard/tools/treadle/src/main/scala/treadle/Driver.scala:87:3: not found: value parser
[error]   parser
[error]   ^
[error] /root/chipyard/tools/treadle/src/main/scala/treadle/Driver.scala:95:3: not found: value parser
[error]   parser
[error]   ^
[error] /root/chipyard/tools/treadle/src/main/scala/treadle/Driver.scala:103:3: not found: value parser
[error]   parser
[error]   ^
[error] /root/chipyard/tools/treadle/src/main/scala/treadle/Driver.scala:111:3: not found: value parser
[... 130 lines cut ...]
[error]   ^
[error] /root/chipyard/tools/treadle/src/main/scala/treadle/vcd/VCDConfig.scala:48:3: not found: value parser
[error]   parser
[error]   ^
[error] /root/chipyard/tools/treadle/src/main/scala/treadle/vcd/VCDConfig.scala:56:3: not found: value parser
[error]   parser
[error]   ^
[error] /root/chipyard/tools/treadle/src/main/scala/treadle/vcd/VCDConfig.scala:64:3: not found: value parser
[error]   parser
[error]   ^
[error] /root/chipyard/tools/treadle/src/main/scala/treadle/vcd/VCDConfig.scala:73:3: not found: value parser
[error]   parser
[error]   ^
[error] /root/chipyard/tools/treadle/src/main/scala/treadle/vcd/VCDConfig.scala:82:57: no arguments allowed for nullary constructor Object: ()Object
[error] class VCDOptionsManager extends ExecutionOptionsManager("vcd") with HasVCDConfig
[error]                                                         ^
[warn] /root/chipyard/tools/treadle/src/main/scala/treadle/vcd/VCDConfig.scala:5:15: Unused import
[warn] import firrtl.ExecutionOptionsManager
[warn]               ^
[warn] four warnings found
[error] 55 errors found
[error] (treadle / Compile / compileIncremental) Compilation failed
[error] Total time: 15 s, completed Dec 21, 2021, 5:16:47 PM
/root/chipyard/common.mk:105: recipe for target 'generator_temp' failed
make: *** [generator_temp] Error 1

12ff7a6, 2021-12-21:
update:
have already generated the Arty bitstream by using local host.

12ff7a6, 2021-12-21:
update:
have already booted on Arty A7 100T. But didn't receive any response in terminal. is A7 35T a must? or have I done something wrong?

abejgonzalez, 2021-12-30:
I don't believe Arty has a UART connection to send a terminal since it is a pretty small device/setup. We only support what is in the docs which is JTAG: https://chipyard.readthedocs.io/en/1.5.0/Prototyping/Arty.html

alexdlukens, 2021-12-31:
@12ff7a6 Are you uploading a program to the FPGA? After generating the bitstream, you need to give the FPGA something to run. There is no default / hello world application included. I use [sifive/freedom-e-sdk](https://github.com/sifive/freedom-e-sdk) and a JTAG debugger to send programs to the FPGA.

JACKLIAO0, 2022-01-03:
> 

hi , alexdlukens

I have generated  .mcs file and download it to Arty 100T board , also a hello world program have download to the flash ,and I successfully saw the "hello".

but when I upload a new binary program via freedom-e-sdk , I encountered a problem like this :
 Error: Debug Module did not become active. dmcontrol=0x0
Warn : target riscv.cpu examination failed
Info : starting gdb server for riscv.cpu on 3333
Info : Listening on port 3333 for gdb connections
Error: Target not examined yet

what can I do to resolve the problem?

@abejgonzalez 
@alexdlukens 

hope for your reply.
Thanks

alexdlukens, 2022-01-03:
@JACKLIAO0 It seems that your computer could not connect to the JTAG Debug module created by Chipyard. What JTAG debugger are you using to connect to the FPGA? The the one I used with the Freedom-E-SDK project is the [Olimex JTAG-TINY Debugger](https://www.mouser.com/ProductDetail/Olimex-Ltd/MSP430-JTAG-TINY-V2?qs=hO3QYTFrOd0M1%2FnQAJ6X5g%3D%3D). 

I outlined the steps I took to get the FPGA connected in [one of my reports](https://alukens.com/project/spring-2021-independent-research-results/research_final_paper.pdf). Try looking at sections 4.2 and 4.5, they might be useful to you for setting up the pinout and configuration settings of Freedom-E-SDK. Table 4.1 shows the wire connections to be made between the debugger and the FPGA.

If the rest of your setup with the JTAG debugger and FPGA is identical to mine, it may be a slight difference between the different FPGA models we are using that is causing an error.

Cheers

JACKLIAO0, 2022-01-12:
@alexdlukens  
  Thanks for your reply , I have seen your reports .  now I use branch arty-spi-flash , I have set frequency of bus as 32.0MHZ and set baud rate as 57600,but   on the terminal still print the character like "?" ,  how can I solve the problem? 
[image]

[image]
[image]

thanks again,
best

closed 2022-01-25 by 12ff7a6

strongerandstr, 2022-01-30:
Hi, I've met the same trouble as the compilation errors when I run the command  $make CONFIG=Sodor1StageConfig  , can you tell me how to solve this issue?

JACKLIAO0, 2022-04-09:
> 

Have you resolved the problem, maybe you can add a clockGate in debug module .

JACKLIAO0, 2022-04-09:
> Hi, I've met the same trouble as the compilation errors when I run the command $make CONFIG=Sodor1StageConfig , can you tell me how to solve this issue?

"Debug.connectDebugClockAndReset(sys.debug, sys.clock)"  ,  I have added this in the top system, then the problem have been solved.

JACKLIAO0, 2022-04-12:
> > > Hi, I've met the same trouble as the compilation errors when I run the command $make CONFIG=Sodor1StageConfig , can you tell me how to solve this issue?
> > 
> > 
> > "Debug.connectDebugClockAndReset(sys.debug, sys.clock)" , I have added this in the top system, then the problem have been solved.
> 
> hi, JACKLIAO0. I have the same problem when I upload a new binary program via freedom-e-sdk : "Error: Target not examined yet" How can I add a clockGate in debug module? "Debug.connectDebugClockAndReset(sys.debug, sys.clock)" Should this sentence be added to ArtyShell.scala? I don't understand the difference between WithArtyJTAGHarnessBinder(HarnessBinders.scala) and connectDebugJTAG(ArtyShell.scala). It seems that this connectDebugJTAG function is not used.

"Debug.connectDebugClockAndReset(sys.debug, sys.clock)"  this code should be added in you top system file.

Yuxin-Yu, 2022-08-16:
> hi, JACKLIAO0. I have the same problem when I upload a new binary program via freedom-e-sdk : "Error: Target not examined yet" How can I add a clockGate in debug module? [...]

Hi, have you solved it? Is it added in ArtyShell.scala?

Yuxin-Yu, 2022-08-17:
> "Debug.connectDebugClockAndReset(sys.debug, sys.clock)" this code should be added in you top system file.

你好，请问能说下具体是加在哪个文件里面吗？

JACKLIAO0, 2022-08-17:
> 

Add it in your own FPGATop File, you can refer to sifive freedom soc.

Yuxin-Yu, 2022-08-17:
> hi , alexdlukens
>
> I have generated .mcs file and download it to Arty 100T board , also a hello world program have download to the flash ,and I successfully saw the "hello".
>
> but when I upload a new [...]

Confirm again that adding "Debug.connectDebugClockAndReset(sys.debug, sys.clock)" solves this problem, I also encountered the problem of "dmcontrol=0x00"

Yuxin-Yu, 2022-08-17:
@JACKLIAO0 你好，请问你最后解决了dmcontrol=0x0的问题吗

Yuxin-Yu, 2022-08-18:
@JACKLIAO0 Hi Bro,Hello, as I understand, I added the Debug.connectDebugClockAndReset code to the WithArtyResetHarnessBinder like this((I am using the Nexys development board):
[image]
But there is a problem,:
[image]
how to solve this? thanks

Yuxin-Yu, 2022-08-18:
The problem is solved, it is the resetn signal problem of mmcm in ArtyShell, this signal should be equal to ~ck_rst.
https://github.com/sifive/fpga-shells/blob/f9fb9fd338e5fca2ff5116b1d01506c424280d70/src/main/scala/shell/xilinx/ArtyShell.scala#:~:text=ip_mmcm.io.resetn%20%20%3A%3D%20ck_rst