ucb-bar/chipyard #1066: Execute  '' make CONFIG=GCDTLRocketConfig BINARY=../../tests/gcd.riscv run-binary '' get Error

lpppeipei, 2021-12-23:
when I   execute  " cd sims/verilator
                          '' make CONFIG=GCDTLRocketConfig BINARY=../../tests/gcd.riscv run-binary ''

It shows "
make: *** [/home/liupeipei/chipyard/common.mk:175: run-binary] Error 255 "

abejgonzalez, 2021-12-30:
We need more information on this issue to be able to help. In the interim, you can try to look at the output logs of the tests and see if there is any indication of what is going on (`*.log` and `*.out` files in the simulation directory/sub-directories).

lpppeipei, 2021-12-31:
> We need more information on this issue to be able to help. In the interim, you can try to look at the output logs of the tests and see if there is any indication of what is going on (`*.log` and `*.out` files in the simulation directory/sub-directories).

It is OK now ~ Thanks.  I  firstly  run " riscv64-unknown-elf-gcc -fno-common -fno-builtin-printf -specs=htif_nano.specs -c  gcd.c
riscv64-unknown-elf-gcc -static -specs=htif_nano.specs gcd.o -o gcd.riscv ",then run   '' make CONFIG=GCDTLRocketConfig BINARY=../../tests/gcd.riscv run-binary ",so everything  is OK .

But  I  note that it shows  "Compiling this program with "make"  produces a gcd.riscv executable."  in the Chipyard Documentation .

What  does  the "make" mean ? And How can  I  use  "make"  to get a gcd.riscv file "?

SihaoLiu, 2022-01-01:
So you did execute this command "make CONFIG=GCDTLRocketConfig BINARY=../../tests/gcd.riscv run-binary"

make is just

lpppeipei, 2022-01-04:
> So you did execute this command "make CONFIG=GCDTLRocketConfig BINARY=../../tests/gcd.riscv run-binary"
> 
> This is make command that runs GCD binary. Have you ever use make command before?

Thanks for your reply. I have not used " make"  command before.

closed 2022-03-01 by lpppeipei