ucb-bar/chipyard #1068: Unresolved dependencies

quyleanh, 2021-12-30:
**Impact**: software

**Tell us about your environment:**
*Chipyard Version:* 1.5.0 
*OS:* 
```
NAME="Red Hat Enterprise Linux Server"
VERSION="7.5 (Maipo)"
```
*Other:* 

**What is the current behavior?**

Error log thrown as following.

```
Java HotSpot(TM) 64-Bit Server VM warning: ignoring option MaxPermSize=256M; support was removed in 8.0
[info] welcome to sbt 1.4.9 (Oracle Corporation Java 1.8.0_25)
[info] loading settings for project chipyard-build from plugins.sbt ...
[info] loading project definition from /home/work/xxx/RiscV/Build1/lequy/chipyard/project
[warn]
[warn]  Note: Some unresolved dependencies have extra attributes.  Check that these dependencies exist with the requested attributes.
[warn]      com.eed3si9n:sbt-buildinfo:0.10.0 (sbtVersion=1.0, scalaVersion=2.12)
[warn]      com.github.gseitz:sbt-protobuf:0.6.3 (sbtVersion=1.0, scalaVersion=2.12)
[warn]      com.dwijnand:sbt-compat:1.2.6 (scalaVersion=2.12, sbtVersion=1.0)
[warn]      com.typesafe.sbt:sbt-site:1.4.1 (sbtVersion=1.0, scalaVersion=2.12)
[warn]      com.eed3si9n:sbt-sriracha:0.1.0 (sbtVersion=1.0, scalaVersion=2.12)
[warn]      com.eed3si9n:sbt-assembly:0.15.0 (sbtVersion=1.0, scalaVersion=2.12)
[warn]      com.typesafe.sbt:sbt-ghpages:0.6.2 (sbtVersion=1.0, scalaVersion=2.12)
[warn]
[warn]  Note: Unresolved dependencies path:
[error] sbt.librarymanagement.ResolveException: Error downloading com.eed3si9n:sbt-buildinfo;sbtVersion=1.0;scalaVersion=2.12:0.10.0
[error]   Not found
[error]   Not found
[error]   not found: https://repo1.maven.org/maven2/com/eed3si9n/sbt-buildinfo_2.12_1.0/0.10.0/sbt-buildinfo-0.10.0.pom
[error]   not found: /home/user/xxx/.ivy2/localcom.eed3si9n/sbt-buildinfo/scala_2.12/sbt_1.0/0.10.0/ivys/ivy.xml
[error]   download error: Caught javax.net.ssl.SSLHandshakeException: sun.security.validator.ValidatorException: PKIX path building failed: sun.security.provider.certpath.SunCertPathBuilderException: unable to find valid certification path to requested target (sun.security.validator.ValidatorException: PKIX path building failed: sun.security.provider.certpath.SunCertPathBuilderException: unable to find valid certification path to requested target) while downloading https://repo.scala-sbt.org/scalasbt/sbt-plugin-releases/com.eed3si9n/sbt-buildinfo/scala_2.12/sbt_1.0/0.10.0/ivys/ivy.xml
[error]   download error: Caught javax.net.ssl.SSLHandshakeException: sun.security.validator.ValidatorException: PKIX path building failed: sun.security.provider.certpath.SunCertPathBuilderException: unable to find valid certification path to requested target (sun.security.validator.ValidatorException: PKIX path building failed: sun.security.provider.certpath.SunCertPathBuilderException: unable to find valid certification path to requested target) while downloading https://repo.typesafe.com/typesafe/ivy-releases/com.eed3si9n/sbt-buildinfo/scala_2.12/sbt_1.0/0.10.0/ivys/ivy.xml
```

**What is the expected behavior?**
All dependencies should be downloaded correctly.

**Other information**

Many dependencies are removed from https://repo1.maven.org/maven2/. When I enter https://repo.scala-sbt.org/scalasbt/sbt-plugin-releases/com.eed3si9n/sbt-buildinfo/scala_2.12/sbt_1.0/0.10.0/ivys/ivy.xml, it's redirected to https://scala.jfrog.io/ui/packages. And I cannot directly download from there.

SihaoLiu, 2021-12-30:
Can you try to use sbt >= 1.4 and Java 11.

quyleanh, 2021-12-31:
@SihaoLiu currently I use sbt version 1.4.9. Java version 8.
Is there any problem?

SihaoLiu, 2021-12-31:
@quyleanh Can you provide more details like what command you are trying and the path you execute the command? 
The version of sbt you are using should be good, sometime I had problem with Java 8 and upgrading to Java 11 solves it.
More details needed

quyleanh, 2021-12-31:
@SihaoLiu I just go to vcs folder and make
```shell
cd sims/vcs
make
```

I found the way to bypass the error about dependencies. I wrote a script to download all jar file to avoid error.

hyun530, 2024-02-16:
Hi, Did you solve this problem?

quyleanh, 2024-02-16:
@hyun530 Yes. I did it long time ago. I don't build environment anymore, but the following is my script to get the dependencies. Please try and get the idea then.

```bash
#!/usr/bin/env bash

dep="dependencies.txt"

down_link_temp="https://scala.jfrog.io/ui/api/v1/download?repoKey=sbt-plugin-releases&path="
save_link_temp="${HOME}/.ivy2/local"

cat $dep | while read line || [[ -n $line ]];
do
    IFS=':' read -r -a info <<< $line
    dep_link=${info[0]}
    dep_name=${info[1]}
    dep_vers=${info[2]}

    link_ivys="${down_link_temp}${dep_link}/${dep_name}/scala_2.12/sbt_1.0/${dep_vers}/ivys/ivy.xml"
    link_jars="${down_link_temp}${dep_link}/${dep_name}/scala_2.12/sbt_1.0/${dep_vers}/jars/${dep_name}.jar"
    link_docs="${down_link_temp}${dep_link}/${dep_name}/scala_2.12/sbt_1.0/${dep_vers}/docs/${dep_name}-javadoc.jar"
    link_srcs="${down_link_temp}${dep_link}/${dep_name}/scala_2.12/sbt_1.0/${dep_vers}/srcs/${dep_name}-sources.jar"

    save_ivys="${save_link_temp}${dep_link}/${dep_name}/scala_2.12/sbt_1.0/${dep_vers}/ivys/ivy.xml"
    save_jars="${save_link_temp}${dep_link}/${dep_name}/scala_2.12/sbt_1.0/${dep_vers}/jars/${dep_name}.jar"
    save_docs="${save_link_temp}${dep_link}/${dep_name}/scala_2.12/sbt_1.0/${dep_vers}/docs/${dep_name}-javadoc.jar"
    save_srcs="${save_link_temp}${dep_link}/${dep_name}/scala_2.12/sbt_1.0/${dep_vers}/srcs/${dep_name}-sources.jar"

    path_ivys="${save_link_temp}${dep_link}/${dep_name}/scala_2.12/sbt_1.0/${dep_vers}/ivys/"
    path_jars="${save_link_temp}${dep_link}/${dep_name}/scala_2.12/sbt_1.0/${dep_vers}/jars/"
    path_docs="${save_link_temp}${dep_link}/${dep_name}/scala_2.12/sbt_1.0/${dep_vers}/docs/"
    path_srcs="${save_link_temp}${dep_link}/${dep_name}/scala_2.12/sbt_1.0/${dep_vers}/srcs/"

    # this variables is replaced for com.jsuereth:pgp-library_2.12:2.1.1 dependency only
    # comment above link_* variables decrlaration before use this one
    # link_ivys="${down_link_temp}${dep_link}/${dep_name}/${dep_vers}/ivys/ivy.xml"
    # link_jars="${down_link_temp}${dep_link}/${dep_name}/${dep_vers}/jars/${dep_name}.jar"
    # link_docs="${down_link_temp}${dep_link}/${dep_name}/${dep_vers}/docs/${dep_name}-javadoc.jar"
    # link_srcs="${down_link_temp}${dep_link}/${dep_name}/${dep_vers}/srcs/${dep_name}-sources.jar"

    mkdir -p ${path_ivys} && wget -O ${save_ivys} ${link_ivys}
    mkdir -p ${path_jars} && wget -O ${save_jars} ${link_jars}
    mkdir -p ${path_docs} && wget -O ${save_docs} ${link_docs}
    mkdir -p ${path_srcs} && wget -O ${save_srcs} ${link_srcs}
done
```

The `dependencies.txt` contains

```text
com.eed3si9n:sbt-buildinfo:0.10.0
com.github.gseitz:sbt-protobuf:0.6.3
com.dwijnand:sbt-compat:1.2.6
com.typesafe.sbt:sbt-site:1.4.1
com.eed3si9n:sbt-sriracha:0.1.0
com.eed3si9n:sbt-assembly:0.15.0
com.typesafe.sbt:sbt-ghpages:0.6.2
com.typesafe.sbt:sbt-ghpages:0.6.3
com.typesafe.sbt:sbt-git:0.9.3
org.planet42:laika-sbt:0.8.0
com.typesafe.sbt:sbt-web:1.4.4
net.virtual-void:sbt-dependency-graph:0.9.2
com.typesafe.sbt:sbt-site:1.3.1
org.xerial.sbt:sbt-pack:0.9.3
com.eed3si9n:sbt-unidoc:0.4.1
org.scoverage:sbt-scoverage:1.5.1
org.scalastyle:scalastyle-sbt-plugin:1.0.0
com.simplytyped:sbt-antlr4:0.8.2
com.github.gseitz:sbt-protobuf:0.6.3
ch.epfl.scala:sbt-scalafix:0.9.21
com.typesafe:sbt-mima-plugin:0.6.1
org.scalameta:sbt-mdoc:2.2.5
com.typesafe.sbt:sbt-site:1.3.2
com.lightbend.paradox:sbt-paradox:0.3.0
org.planet42:laika-sbt:0.7.5
com.typesafe.sbt:sbt-web:1.4.2
com.lightbend.paradox:sbt-paradox:0.3.2
com.typesafe.sbt:sbt-git:1.0.0
com.dwijnand:sbt-dynver:4.1.1
com.typesafe.sbt:sbt-site:1.4.0
com.jsuereth:sbt-pgp:2.1.1
com.eed3si9n:sbt-unidoc:0.4.3
com.github.gseitz:sbt-protobuf:0.6.5
com.eed3si9n:sbt-assembly:0.14.10
com.jsuereth:sbt-pgp:2.0.1
```

closed 2024-02-16 by quyleanh