ucb-bar/chipyard #1075: excute  the  command  " make SUB_PROJECT=vcu118 CONFIG=RocketVCU118Config bitstream " occured  error

lpppeipei, 2022-01-06:
In  the Prototyping Flow, after   I  excute  the  command  "./scripts/init-fpga.sh  ;  make SUB_PROJECT=vcu118 CONFIG=RocketVCU118Config bitstream "

After  a while ,It occured 

/bin/bash: line 1: vivado: command not found
make: *** [Makefile:115: /home/liupeipei/chipyard/fpga/generated-src/chipyard.fpga.vcu118.VCU118FPGATestHarness.RocketVCU118Config/obj/VCU118FPGATestHarness.bit] Error 127

Does anyone know why ?  Thanks a lot.

michael-etzkorn, 2022-01-12:
Looks like you don't have `vivado` on your path. If you run Vivado under a different name, in the bitstream makefile rule, you can change `vivado` to whatever you type to run Vivado from the command line.

lpppeipei, 2022-02-28:
> 

Thanks for your reply. Could you give me more  details ? For example, which file  can I  change the vivado path?

lpppeipei, 2022-02-28:
> Thanks for your reply. Could you give me more details ? For example, which file can I change the vivado path?

> Looks like you don't have `vivado` on your path. If you run Vivado under a different name, in the bitstream makefile rule, you can change `vivado` to whatever you type to run Vivado from the command line.

Thanks for your reply. Could you give me more details ? For example, which file can I change the vivado path?

lpppeipei, 2022-02-28:
> Looks like you don't have `vivado` on your path. If you run Vivado under a different name, in the bitstream makefile rule, you can change `vivado` to whatever you type to run Vivado from the command line.

Thanks for your reply. Could you give me more details ? For example, which file can I change the vivado path?

lpppeipei, 2022-02-28:
> Looks like you don't have `vivado` on your path. If you run Vivado under a different name, in the bitstream makefile rule, you can change `vivado` to whatever you type to run Vivado from the command line.

Yes, I  have save the  vivado path problem,but  a new  issue occurs as follows:

{unknown option} {}

    while executing
"source [file join $scriptdir "prologue.tcl"]"
    (file "/home/liupeipei/chipyard/fpga/fpga-shells/xilinx/common/tcl/vivado.tcl" line 7)
INFO: [Common 17-206] Exiting Vivado at Mon Feb 28 17:30:39 2022...
make: *** [Makefile:115: /home/liupeipei/chipyard/fpga/generated-src/chipyard.fpga.vcu118.VCU118FPGATestHarness.RocketVCU118Config/obj/VCU118FPGATestHarness.bit] Error 1

michael-etzkorn, 2022-02-28:
Based on your discussion in the google forum about Artix-7, I'm guessing you have a license issue with trying to run the vcu118 `fpga-shells` script. Chipyard has no support for Artix-7 currently. To continue, you'll have to create an Artix-7 test harness. The Arty harness may serve as an example for how to declare ChipTop and hook up to your board's FPGA pins. If you want to use non-project mode to generate the bitstream, you'll also have to adapt the `fpga-shells` tcl scripts to support your board (this should just be as straightforward as providing a `board.tcl` file for Artix-7), but you could generate the bitstream through project mode.

michael-etzkorn, 2022-02-28:
However, that would fail on `synth_design`. I believe the error you're facing now is because `fpga-shells` hasn't been checked out. Try running:

```
# in the chipyard top level folder
./scripts/init-fpga.sh
```

lpppeipei, 2022-03-01:
> However, that would fail on `synth_design`. I believe the error you're facing now is because `fpga-shells` hasn't been checked out. Try running:
> 
> ```
> # in the chipyard top level folder
> ./scripts/init-fpga.sh
> ```

Thanks for your reply. I tried the command  "./scripts/init-fpga.sh" in  the chipyard top level folder before, but when I  run 

"make SUB_PROJECT=vcu118 CONFIG=RocketVCU118Config bitstream" ,it occurs the same issue.

Maybe  I have  a probelm when I  run " ./scripts/init-fpga.sh",it just exit script  as follows:

liupeipei@ubuntu:~/chipyard$ ./scripts/init-fpga.sh
liupeipei@ubuntu:~/chipyard$ 

Do you  have any ideas ? Thanks.

lpppeipei, 2022-03-03:
It's  OK  now.Thanks  a lot.

wangliwei-intel, 2022-04-16:
@lpppeipei 
Which chipyard version are you using? thanks.

lpppeipei, 2022-04-22:
> @lpppeipei Which chipyard and vivado version are you using? thanks.

1.6.0 .   Thanks for your reply,.it's  OK now.

closed 2022-04-22 by lpppeipei

souravmohapatra, 2022-04-27:
Hello, sorry for reviving this. I have the same problem.

vivado version: v2022.1 (64-bit)
chipyard: 1.6.2

Ran init-fpga script
vivado command available
Using tiny rocket arty config (the predefined one)

`make SUB_PROJECT=arty CONFIG=TinyRocketArtyConfig bitstream`

Getting the following error

`{unknown option} {}

    while executing
"source [file join $scriptdir "prologue.tcl"]"
    (file "/home/mohapatr/install/chipyard/fpga/fpga-shells/xilinx/common/tcl/vivado.tcl" line 7)
INFO: [Common 17-206] Exiting Vivado at Wed Apr 27 23:14:36 2022...
make: *** [Makefile:115: /home/mohapatr/install/chipyard/fpga/generated-src/chipyard.fpga.arty.ArtyFPGATestHarness.TinyRocketArtyConfig/obj/ArtyFPGATestHarness.bit] Error 1
`
Feels like I am missing something silly but can't find it :(

souravmohapatra, 2022-04-27:
I found the issue, there are extra spaces in the Makefile "bitstream" option after the `-board` tcl args. This is fixed by commit af78c9cadf01e3bd0244d06b6c7fe3735031aa13 in the mainline!

Applying that resolves it!