ucb-bar/chipyard #1077: Simulate two processors

zhejianguk, 2022-01-12:
Hi, I have successfully simulated the ``hello-world.riscv'' on a single-rocket-core system using verilator.
For the next step, I need to build a hardware with a single rocket core and a single BOOM core, and boot both of them with ``hello-world.riscv''.

However, when I run `./xxxxx-debug ./hello-world.riscv` in thesims/verilator, only one ``hello-world'' is output from the std_in/std_out. I am wondering how should I run the software on both processors?

Best regards,

SihaoLiu, 2022-01-12:
I don't get you question. How many "Hello world" you are expecting? Is this hello-world a multi-thread program?

zhejianguk, 2022-01-12:
@SihaoLiu, I really appreciate for your quick reply and I am sorry for the confusion. Let me revise my question:

Step 1: We build a dual-core system with 1 Rocket core and 1 Boom core
Step 2: We employed the two pieces of software, say Hello_Rocket.riscv and Hello_Boom.riscv
Step 3: We need to deploy the Hello_Rocket.riscv to Rocket core; and Hello_Boom.riscv to the Boom core (using verilator for simulation)

My question is: how should I proceed the step 3? 

Many thanks for your help, and I wish you all the best in the new 2022!

Best regards,

SihaoLiu, 2022-01-16:
Hi sorry for late reply. I think you will have to have one Hello_Rocket_Boom.riscv and use read_csr to get the hartid. I don't think there will be easy control to deploy a thread to a specific core.

zhejianguk, 2022-01-16:
Much appreciated, Sihao!

Wish you have a nice 2022!

Best regards,

jerryz123, 2023-02-17:
Closing, resolved

closed 2023-02-17 by jerryz123