ucb-bar/chipyard #1081: How to enable FGP (Fine Grained Parallelism) in VCS simulation

SihaoLiu, 2022-01-16:
According to VCS user guide,

```
Two-step flow:
% vcs -fgp -full64 <otherOptions> 
% simv -fgp=num_threads:<value>
```

By adding `-fgp` option to vcs and simv, we might be able to speed up the simulation. I would like to know is it possible for chipyard flow? I tried to add `-fpg` related to `vcs`, the compilation seems working (compile command looks like: 
`vcs -fgp -full64 -CFLAGS " -O3 -std=c++11 -I/home ...`

But when I tried to run binary simulation with : `make CONFIG=... EXTRA_SIM_FLAGS=-fgp=num_threads:8 BINARY=...`

It failed with error:

```
Error-[DPI-UED] C++ Exception detected
  Import DPI routine invoked at file
  '/home/sihao/repo/chipyard/sims/vcs/generated-src/chipyard.TestHarness.MeshDSARocketConfig/SimSerial.v'(line
  53) has C++ exceptions not caught. C++ exceptions shall not propagate out of
  any imported subroutine.
  Fix it in DPI-C code before running simulation.
```

I would like to know would it be easy to enable fine-grained parallelism in current flow? If so, how should I enable it? If not, where should I hack?

Thank you so much in advance!

jerryz123, 2022-01-17:
I believe this is due to how HTIF is parsing arguments. https://github.com/riscv-software-src/riscv-isa-sim/blob/e93b9cbbbcd3ad0a02ae298e9f1a2d98d3ac0153/fesvr/htif.cc#L252 . I believe `getopt_long` is treating `-fgp`... as option characters, equivalent to `-f -g -p`.

I think the best solution to this is to modify the initialization of `testchip_tsi_t` to remove the `-fgp` flag from the argc, argv that is passed to the base class constructor. The inheritance structure is `testchip_tsi_t < tsi_t < htif_t`.

SihaoLiu, 2022-01-17:
I see, thanks! I will take a look.

SihaoLiu, 2022-01-28:
PR https://github.com/ucb-bar/testchipip/pull/141 created

PR #1150 (2022-04-08): VCS Enhancements
**Related PRs / Issues**:

Closes #1081 
Closes #1136 

**Type of change**:
- [x] Bug fix
- [x] New feature
- [ ] Other enhancement

**Impact**:
- [ ] RTL change
- [x] Software change (RISC-V software)
- [ ] Build system change
- [x] Other

**Contributor Checklist**:
- [x] Did you set `main` as the base branch?
- [x] Did you delete any extraneous prints/debugging code?
- [ ] (If applicable) Did you add documentation for the feature?
- [ ] (If applicable) Did you add a test demonstrating the PR?

- [ ] (If applicable) Did you mark the PR as "Please Backport"?

**Release Notes** [... cut]
closed 2022-04-08 by jerryz123