ucb-bar/chipyard #1096: How to get more details about bootrom and how to set Rocketchip to boot from SPFlash specified address

StilluSmile, 2022-01-21:
Based on chipyard I generated a SoC with DRAM and SPIFlash, next I want to map this SoC to FPGA board. I will not use the JTAG interface for the time being, and directly read and load the executable file from the external flash. 
Note that there is a bootrom folder in the rocket-chip subdirectory. Is it possible to modify the starting address after reset by modifying the bootrom? Also, I would like to know if I place the executable at a certain address on the flash, how can I set the CPU to load and execute the executable from the above specified address.
Looking forward to hearing from you~

jerryz123, 2022-01-21:
We recently added a feature to adjust the boot address.
Add the following config fragment to your config
```
new testchipip.WithCustomBootPinAltAddr(<some_addr>) ++
```

The generated chip should have a single external IO called `custom_boot` or something. When that pin is asserted, the core will jump to `<some_addr>` instead of DRAM_BASE. This is intended to be useful for dynamically selecting boot address after tapeout (or after FPGA deployment).

To test this in RTL simulation, pass `EXTRA_SIM_FLAGS="+custom_boot_pin=1` to the make command.

StilluSmile, 2022-01-21:
ok，i will try it.

jerryz123, 2023-02-17:
closing due to inactivity, this works now.

closed 2023-02-17 by jerryz123