ucb-bar/chipyard #1098: FPGA Linux Boot Gets Stuck When Using Medium BOOM

zslwyuan, 2022-01-25:
**Impact**: rtl 

**Tell us about your environment:**
*Chipyard Version:* 1.2.0, Hash: 2c0928
*OS:* CentOS 7.4
*Other:* Followed the steps in FPGA prototyping for VCU118

**What is the current behavior?**
We changed the default 1-wide BOOM settings(boot successfully) into 2-wide BOOM (i.e. MediumBoomConfig) with 3 cores.  Then the Linux boot procedure got stuck after "Starting mdev....ok" or got stuck after boot when we started a multi-thread application.

closed 2022-02-25 by zslwyuan

Yuxin-Yu, 2023-11-13:
Hi bro,I met the same problem when I boot linux with MegaBoom,how do you solve this? @zslwyuan

Yuxin-Yu, 2023-11-14:
Hello, I have encountered the same problem in GigaBoom and MegaBoom, but occasionally restarting can solve it. How did you solve it @zslwyuan