ucb-bar/chipyard #1101: Build error when running "./scripts/build-toolchains.sh riscv-tools"

g07h4xf00, 2022-01-27:
I get the following error when trying to build the repository with "./scripts/build-toolchains.sh riscv-tools" :

/home/mogleari/vc/chipyard/riscv-tools-install/lib/gcc/riscv64-unknown-linux-gnu/9.2.0/../../../../riscv64-unknown-linux-gnu/bin/ld: cannot find -lstdc++ /home/mogleari/vc/chipyard/riscv-tools-install/lib/gcc/riscv64-unknown-linux-gnu/9.2.0/../../../../riscv64-unknown-linux-gnu/bin/ld: cannot find -lgcc_s collect2: error: ld returned 1 exit status make[3]: *** [../Rules:193: /home/mogleari/vc/chipyard/toolchains/riscv-tools/riscv-gnu-toolchain/build/build-glibc-linux-rv64imafdc-lp64d/support/links-dso-program] Error 1 make[3]: *** Waiting for unfinished jobs.... make[3]: Leaving directory '/home/mogleari/vc/chipyard/toolchains/riscv-tools/riscv-gnu-toolchain/riscv-glibc/support' make[2]: *** [Makefile:258: support/others] Error 2 make[2]: Leaving directory '/home/mogleari/vc/chipyard/toolchains/riscv-tools/riscv-gnu-toolchain/riscv-glibc' make[1]: *** [Makefile:9: all] Error 2 make[1]: Leaving directory '/home/mogleari/vc/chipyard/toolchains/riscv-tools/riscv-gnu-toolchain/build/build-glibc-linux-rv64imafdc-lp64d' make: *** [Makefile:243: stamps/build-glibc-linux-rv64imafdc-lp64d] Error 2

The innate ld on my system does recognize flags like "-lstdc++" but it seems like ld binary generated here does not? I'm pretty sure I installed all the dependencies listed.

Prerise, 2022-02-23:
I occur the same question, have you resolved this?

geekLucian, 2022-02-27:
I met the same problem before, and my solution is to run build-toolchains.sh in an environment without riscv toolchain in your $PATH.

geekLucian, 2022-02-28:
I met the same problem before, and my solution is to run build-toolchains.sh in an environment without $RISCV and riscv toolchain in your $PATH.

mbelda, 2022-07-14:
I have the same error when running ./scripts/build-toolchains.sh esp-tools.

Does the arty100t branch support the esp-tools toolchain?

LY-Mei, 2022-08-12:
> I met the same problem before, and my solution is to run build-toolchains.sh in an environment without $RISCV and riscv toolchain in your $PATH.

Yes, I met the same problem and solved it by 2 actions:
1) set $RISCV empty, 
2) remove all the riscv related path in $PATH and $LD_LIBRARY_PATH. 

That's all.

jerryz123, 2023-02-17:
This is no longer an issue in the new conda-based build flow

closed 2023-02-17 by jerryz123