ucb-bar/chipyard #1103: Error while integrating custom FIRRTL transform with Chisel 3.5

tymcauley, 2022-01-31:
**Impact**: other

**Tell us about your environment:**
*Chipyard Version:* `dev` branch / `chisel-3.5-published` branch
*OS:* macOS 12.2
*Other:* 

**What is the current behavior?**

I am able to run custom FIRRTL transforms on the `dev` branch, but not on the `chisel-3.5-published` branch.

**What is the expected behavior?**

I should be able to run FIRRTL transforms on the `chisel-3.5-published` branch.

**Other information**

So, I understand this issue is a bit speculative, since the `chisel-3.5-published` branch isn't even merged into `dev` yet, but I wanted to test out some of Chisel 3.5's features, and noticed this issue while trying to run a custom FIRRTL transform off of that branch.

On the `dev` branch, adding my custom Chisel project as a dependency of the `chipyard` project in `build.sbt` is enough for me to be able to run a custom FIRRTL transform. You can check out this branch to reproduce: https://github.com/tymcauley/chipyard/tree/custom-firrtl-transform-chisel-3.4

Once submodules are set up, run this command:

```
make -C sims/vcs FIRRTL_LOGLEVEL=info EXTRA_FIRRTL_OPTIONS="--custom-transforms customfirrtl.MyCustomFirrtlTransform" verilog
```

You should see this message somewhere near the end of the FIRRTL compiler output:

```
======== Starting Transform customfirrtl.MyCustomFirrtlTransform ========
Found 62 module definitions in the circuit
-------------------------------------------------------------------------
Time: 0.0 ms
Form: UnknownForm
======== Finished Transform customfirrtl.MyCustomFirrtlTransform ========
```

I also created this branch to add the same change to the `chisel-3.5-published` branch: https://github.com/tymcauley/chipyard/tree/custom-firrtl-transform-chisel-3.5

If you run the same `make` command as above, then you get this output from the FIRRTL compiler:

```
[info] running (fork) barstools.tapeout.transforms.GenerateTopAndHarness --allow-unrecognized-annotations ...
[info] ------------------------------------------------------------------------------
[info] Error: Option --custom-transforms failed when given 'customfirrtl.MyCustomFirrtlTransform'. Unable to locate custom transform customfirrtl.MyCustomFirrtlTransform (did you misspell it?)
[info] Try --help for more information.
[info] ------------------------------------------------------------------------------
[error] Nonzero exit code returned from runner: 1
[error] (Compile / runMain) Nonzero exit code returned from runner: 1
[error] Total time: 2 s, completed Jan 31, 2022 2:26:22 PM
make: *** [firrtl_temp] Error 1
```

abejgonzalez, 2022-02-01:
Have you brought up the SBT CLI and seen if the SBT project is added to the barstools classpath? You might also want to experiment a bit with seeing if you can access the particular class in barstools: https://github.com/chipsalliance/firrtl/blob/475c165ccf8a52f79a94766450c23090f15d1393/src/main/scala/firrtl/stage/FirrtlAnnotations.scala#L241

tymcauley, 2022-02-02:
> Have you brought up the SBT CLI and seen if the SBT project is added to the barstools classpath? You might also want to experiment a bit with seeing if you can access the particular class in [...]

Great idea, I didn't think to do that at all! So, for the `chipyard` project, I get the expected result (including my custom FIRRTL transform project):

```
sbt:chipyard> show Compile/fullClasspath
[warn] multiple main classes detected: run 'show discoveredMainClasses' to see the list
[info] * Attributed(/path/to/chipyard/generators/chipyard/target/scala-2.12/chipyard_2.12-1.6.jar)
[info] * Attributed(/path/to/chipyard/generators/testchipip/target/scala-2.12/testchipip_2.12-1.0-SNAPSHOT.jar)
[info] * Attributed(/path/to/chipyard/generators/rocket-chip/src/target/scala-2.12/rocketchip_2.12-1.6.jar)
[info] * Attributed(/path/to/chipyard/generators/rocket-chip/hardfloat/target/scala-2.12/hardfloat_2.12-1.5-SNAPSHOT.jar)
[info] * Attributed(/path/to/chipyard/sims/firesim/sim/midas/targetutils/target/scala-2.12/midas-targetutils_2.12-1.0-SNAPSHOT.jar)
[info] * Attributed(/path/to/chipyard/generators/rocket-chip/macros/target/scala-2.12/rocketmacros_2.12-1.6.jar)
[info] * Attributed(/path/to/chipyard/generators/rocket-chip/api-config-chipsalliance/build-rules/sbt/target/scala-2.12/rocketconfig_2.12-1.6.jar)
[info] * Attributed(/path/to/chipyard/generators/sifive-blocks/target/scala-2.12/sifive_blocks_2.12-1.6.jar)
[info] * Attributed(/path/to/chipyard/generators/boom/target/scala-2.12/boom_2.12-3.0.jar)
[info] * Attributed(/path/to/chipyard/generators/hwacha/target/scala-2.12/hwacha_2.12-1.2.jar)
[info] * Attributed(/path/to/chipyard/generators/sifive-cache/target/scala-2.12/sifive_cache_2.12-1.6.jar)
[info] * Attributed(/path/to/chipyard/tools/barstools/src/target/scala-2.12/iocell_2.12-1.6.jar)
[info] * Attributed(/path/to/chipyard/generators/sha3/target/scala-2.12/sha3_2.12-1.6.jar)
[info] * Attributed(/path/to/chipyard/generators/my-firrtl-transform/target/scala-2.12/customfirrtl_2.12-0.1.jar)
[info] * Attributed(/path/to/chipyard/tools/dsptools/src/target/scala-2.12/dsptools_2.12-1.6.jar)
[info] * Attributed(/path/to/chipyard/tools/rocket-dsp-utils/src/target/scala-2.12/rocket-dsp-utils_2.12-1.6.jar)
[info] * Attributed(/path/to/chipyard/tools/api-config-chipsalliance/src/target/scala-2.12/api-config-chipsalliance_2.12-1.6.jar)
[info] * Attributed(/path/to/chipyard/generators/gemmini/target/scala-2.12/gemmini_2.12-3.1.0.jar)
[info] * Attributed(/path/to/chipyard/generators/icenet/target/scala-2.12/icenet_2.12-1.0.jar)
[info] * Attributed(/path/to/chipyard/generators/tracegen/target/scala-2.12/tracegen_2.12-1.6.jar)
[info] * Attributed(/path/to/chipyard/generators/cva6/target/scala-2.12/cva6_2.12-1.0.jar)
[info] * Attributed(/path/to/chipyard/generators/nvdla/target/scala-2.12/nvdla_2.12-1.6.jar)
[info] * Attributed(/path/to/chipyard/generators/riscv-sodor/target/scala-2.12/sodor_2.12-2.0.jar)
[info] * Attributed(/path/to/chipyard/generators/ibex/target/scala-2.12/ibex_2.12-1.0.jar)
[info] * Attributed(/path/to/chipyard/generators/fft-generator/target/scala-2.12/fft_generator_2.12-1.6.jar)
[info] * Attributed(/path/to/Caches/Coursier/v1/https/repo1.maven.org/maven2/org/scala-lang/scala-library/2.12.10/scala-library-2.12.10.jar)
...
```

But for barstools, I don't see any local dependencies (aside from the project's own classpath):

```
sbt:tapeout> show Compile/fullClasspath
[info] * Attributed(/path/to/chipyard/tools/barstools/target/scala-2.12/classes)
[info] * Attributed(/path/to/Caches/Coursier/v1/https/repo1.maven.org/maven2/org/scala-lang/scala-library/2.12.13/scala-library-2.12.13.jar)
...
```

These results are from this branch: https://github.com/tymcauley/chipyard/commit/fb954eafcd185793b29cdada1e5e9b9a8f34c448

Any ideas on why `barstools` wouldn't take the `dependsOn` setting? I'll play around with the `sbt` settings. Thanks for pointing me to a debug route @abejgonzalez!

abejgonzalez, 2022-02-02:
I have no idea... I expected it to show up there (maybe check the behavior from before)?

tymcauley, 2022-02-02:
Working off of the `dev` branch, I see all of the local dependencies of the `tapeout` project. However, working on your `chisel-3.5-published` branch, I can't add any local dependencies to the `tapeout` project using `dependsOn(...)` (`chipyard` doesn't work, `rocketchip` doesn't work, ...). Not sure why `tapeout` is now unique in this behavior...

abejgonzalez, 2022-02-02:
Hmm. Maybe this is a new SBT bump issue (we bumped the SBT version in the PR as well)? Otherwise, I don't know...

tymcauley, 2022-02-02:
Alright, I have a suspicion that this has to do with the `build.sbt` file in `barstools`, but I need to do a few more experiments.

tymcauley, 2022-02-02:
Okay, it appears that the [`tapeout` project definition in the `barstools` `build.sbt` file](https://github.com/ucb-bar/barstools/blob/314d80729e7b32e3c10c0da6734bbdc9a867916f/build.sbt#L34-L41) is overwriting [the `tapeout` project definition in the Chipyard `build.sbt` file](https://github.com/ucb-bar/chipyard/blob/27fceafdd7949cbbc9f64f940f1643b31ee3439e/build.sbt#L217-L221).

If you comment out the `tapeout` project definition in the `barstools` `build.sbt` file, then all settings in the Chipyard `build.sbt` file take effect, and you can set inter-project dependencies successfully.

Comparing to the `dev` branch, the Chipyard `build.sbt` has this `tapeout` definition:

```
lazy val tapeout = (project in file("./tools/barstools/tapeout/"))
  .dependsOn(chisel_testers, chipyard) // must depend on chipyard to get scala resources
  .settings(commonSettings)
```

However, there's no `build.sbt` file in `./tools/barstools/tapeout/`. It seems like this means that the Chipyard `tapeout` project doesn't import the `build.sbt` file located at `./tools/barstools/`, so the two project definitions don't have to merge.

Looking at [the SBT docs on the subject](https://www.scala-sbt.org/1.x/docs/Multi-Project.html#Appendix%3A+Subproject+build+definition+files), they have these suggestions:

> Style choices:
> - Each subproject’s settings can go into *.sbt files in the base directory of that project, while the root build.sbt declares only minimum project declarations in the form of lazy val foo = (project in file("foo")) without the settings.
> - We recommend putting all project declarations and settings in the root build.sbt file in order to keep all build definition under a single file. However, it’s up to you.

I'm guessing you'd like to keep barstools as being able to run as a standalone project. I'm not sure if there's a good way to ignore those `build.sbt` settings for the Chipyard case.

jerryz123, 2022-02-02:
You can have the subproject SBT read an environment variable to selectively set settings. I've done this when I've needed a subproject to have different SBT settings when run standalone, but its kind of a hack.

tymcauley, 2022-02-02:
I probably just don't have enough context on the barstools project, but does it need to be run standalone or with other non-chipyard projects? If not, it seems like the easiest solution is to just remove the conflicting project definition inside the barstools repo.

tymcauley, 2022-02-02:
Actually, it looks like we can make all of the `barstools` `build.sbt` settings unscoped, rather than defining and configuring a `tapeout` project in there. So we can still have setting like this:

```scala
organization := "edu.berkeley.cs"
version := "0.4-SNAPSHOT"
name := "barstools"
scalaVersion := "2.12.13"
fork := true
Test / scalacOptions ++= Seq("-language:reflectiveCalls")
mainClass := Some("barstools.macros.MacroCompiler")
libraryDependencies ++= Seq(
  "com.typesafe.play" %% "play-json" % "2.9.2",
  "org.scalatest" %% "scalatest" % "3.2.9" % "test",
  "org.apache.logging.log4j" % "log4j-api" % "2.11.2",
  "org.apache.logging.log4j" % "log4j-core" % "2.11.2"
)
```

And that will merge nicely with the settings in Chipyard's `build.sbt`.

tymcauley, 2022-02-02:
Since the solution to this problem might lie in barstools, I just filed this issue there: https://github.com/ucb-bar/barstools/issues/117.

abejgonzalez, 2022-02-02:
I feel like the `freshProject` function call was used to solve similar issues... but its been a while since I've SBT hacked.

abejgonzalez, 2022-02-02:
> I probably just don't have enough context on the barstools project, but does it need to be run standalone or with other non-chipyard projects? If not, it seems like the easiest solution is to just remove the conflicting project definition inside the barstools repo.

I think originally it was designed to be separate but over time it has gotten more and more coupled with Chipyard.

tymcauley, 2022-02-02:
Okay, I feel like the solution I proposed in https://github.com/ucb-bar/barstools/issues/117 should let barstools remain usable as a standalone project while also respecting Chipyard's `build.sbt` settings. If you think that strategy will work, then I can make a PR in barstools.

tymcauley, 2022-02-03:
Closing this issue now that this is being resolved in issue https://github.com/ucb-bar/barstools/issues/117 and PR https://github.com/ucb-bar/barstools/pull/118.

closed 2022-02-03 by tymcauley