ucb-bar/chipyard #1106: How to add HarnessBinder with more IOBinder

bg193, 2022-02-03:
Hello, I want to add a new harness binder have a axi slave port and several interrupt ports like WithTieOffL2FBusAXI+WithTieOffInterrupts.

```
class WithTieOffInterrupts extends OverrideHarnessBinder({
  (system: HasExtInterruptsModuleImp, th: HasHarnessSignalReferences, ports: Seq[UInt]) => {
  .....
  }
})

class WithTieOffL2FBusAXI extends OverrideHarnessBinder({
  (system: CanHaveSlaveAXI4Port, th: HasHarnessSignalReferences, ports: Seq[ClockedIO[AXI4Bundle]]) => {
   ......
  }
})
```
So , my question is how to get two different function ports(axi slave&ext_interrupts) in a harness binder , or call two binders connect to same target module which instantiate only once?

jerryz123, 2022-02-03:
This feature is currently not supported. You can hack in your harness module directly in the `TestHarness` is a workaround

bg193, 2022-02-03:
> This feature is currently not supported. You can hack in your harness module directly in the `TestHarness` is a workaround

Can you give me some advice on this? I haven't found the right way.

bg193, 2022-02-05:
> > This feature is currently not supported. You can hack in your harness module directly in the `TestHarness` is a workaround
> 
> Can you give me some advice on this? I haven't found the right way.

The problem has been solved.

closed 2022-02-05 by bg193