ucb-bar/chipyard #1111: Overlap of standard cells in GDS of TinyRocket generated from the ASAP7 VLSI Flow

baichen318, 2022-02-10:
Hi all,

Recently I encountered a problem with the generated GDS of TinyRocket.

I found that layer `7/0​` in GDS overlapped.

A screenshot below illustrates an abnormal case in my generated GDS of TinyRocket.
[image]

For each standard cell, each component of layer `7/0`​ has approximately the same gap (AND2x2_ASAP7_75t_L), as a screenshot shown below.
[image]

However, multiple standard cells overlap each other in the final GDS.

The VLSI flow to obtain the GDS of TinyRocket strictly follows the document: https://chipyard.readthedocs.io/en/latest/VLSI/Tutorial.html.

I wondered what caused the abnormal case and how I could solve it.

Thank you very much in advance!

Best regards,
Chen

harrisonliew, 2022-02-10:
Hi @baichen318, I have received your email about this but didn't have time to look at it yet. I will try to debug tomorrow and let you know.

harrisonliew, 2022-02-11:
Hi @baichen318,

I have figured out that it actually is due to the ASAP7 PDK r1p5. I was able to reproduce your GDS using the current Chipyard release. The issue is caused by some standard cell LEFs having incorrect boundaries.

On the `dev` branch, the tutorial is updated for the publicly released ASAP7 PDK r1p7 (#934). Here is what my gate layer shows in this new PDK:

[image]

We plan to release Chipyard pointing to this newer PDK very soon.

-Harrison

closed 2022-02-11 by harrisonliew

baichen318, 2022-02-12:
Hi @harrisonliew,

Thank you very much.

Best regards,
Chen