ucb-bar/chipyard #1129: After  run make CONFIG=FFTRocketConfig,  ERROR

lpppeipei, 2022-02-24:
### Background Work

- [X] Yes, I searched the [mailing list](https://groups.google.com/forum/#!forum/chipyard)
- [X] Yes, I searched [prior issues](https://github.com/ucb-bar/chipyard/issues)
- [X] Yes, I searched the [documentation](https://chipyard.readthedocs.io/)

### Chipyard Version and Hash

Release: 1.6.0
Hash: a6a6a6

### OS Setup

Ex: Output of `uname -a` and `lsb_release -a`

### Other Setup

Ex: Prior steps taken / Documentation Followed / etc...

### Current Behavior

[error] Picked up JAVA_TOOL_OPTIONS: -Xmx8G -Xss8M -Djava.io.tmpdir=/home/liupeipei/chipyard/.java_tmp
[error] Nonzero exit code returned from runner: 137
[error] (Compile / runMain) Nonzero exit code returned from runner: 137
[error] Total time: 122 s (02:02), completed Feb 24, 2022, 11:04:43 AM
make: *** [/home/liupeipei/chipyard/common.mk:137: firrtl_temp] Error 1

### Expected Behavior

no  error

closed 2022-02-24 by lpppeipei

1bing2, 2022-04-29:
Have you haddle it?

lpppeipei, 2022-05-05:
> > > Have you haddle it?
> > 
> > 
> > Yes,it's OK now.
> 
> how can you solve this problem? i have a this a problem when i run make CONFIG=SmallBoomConfig. It have a error. error:/home/lee/Desktop/riscv-boom/verilator/chipyard/common.mk:110: recipe for target 'generator_temp' failed make: *** [generator_temp] Error 1 Can you give me a help ,please?

My problem occured because of  not enough memory. Maybe you can refer  to  issue: "recipe for target 'firrtl_temp' failed #455".