ucb-bar/chipyard #1132: Couldn't not find cpu id for hartid[9...15] during Linux Boot (FPGA Prototyping)

zslwyuan, 2022-02-25:
### Background Work

- [X] Yes, I searched the [mailing list](https://groups.google.com/forum/#!forum/chipyard)
- [X] Yes, I searched [prior issues](https://github.com/ucb-bar/chipyard/issues)
- [X] Yes, I searched the [documentation](https://chipyard.readthedocs.io/)

### Chipyard Version and Hash

Release: 1.5.0
Hash: a6a6a6

### OS Setup

Distributor ID:	Ubuntu
Description:	Ubuntu 20.04.3 LTS
Release:	20.04
Codename:	focal

### Other Setup

Go through the standard [FPGA prototyping flow in Chipyard Documentation](https://chipyard.readthedocs.io/en/latest/Prototyping/VCU118.html) but change the number of cores from 1 to 4 / 8 /  16.

### Current Behavior

After programming the FPGA and starting the Linux Boot procedure, we found that the system with 1/4/8 cores can boot normally but when the number of cores increases to 16, we got errors during Linux Boot: "Couldn't not find cpu id for hartid[9...15]".

[image] 

### Expected Behavior

It should boot normally when there are 16 cores. Currently we have not figured out the reason.

jerryz123, 2022-02-25:
Interesting. What happens when you try to boot Linux in spike with -p16. Does that work?

zslwyuan, 2022-02-25:
> Interesting. What happens when you try to boot Linux in spike with -p16. Does that work?

@jerryz123 Thanks Jerry! I think I found the cause. It seems that the problem is in the kernel instead of the hardware design.

When compiling the Linux kernel, it will set a maximum number of CPU cores, as shown in the link below:
https://elixir.bootlin.com/linux/latest/source/arch/riscv/Kconfig#L285

and the default value of the maximum number of cores is 8 for RISC-V.
I think I can rebuild the kernel for later testing.
Besides, maybe an option in the [prototyping documentation](https://chipyard.readthedocs.io/en/latest/Prototyping/VCU118.html) to allow designers to explictly set the kernel configuration would be great. ^_^

zslwyuan, 2022-02-25:
It is interesting. According to the config information shown in the kernel code (line 287: range 2 32), does it means that RISC-V system cannot support hardware with more than 32 cores.

michael-etzkorn, 2022-02-28:
@zslwyuan I'd be curious to know if you could even fit 32 Rocket Tiny Cores onto the vcu118. I'm already impressed to hear 16 cores fit. I imagine you had to adjust the clock frequency to below 100MHz?

zslwyuan, 2022-03-01:
@michael-etzkorn actually we are running the design on a larger FPGA but go through a similar prototyping flow. ^_^