ucb-bar/chipyard #1139: Failing 'useVM=false' RocketConfig Build

francesco-peverelli, 2022-03-07:
### Background Work

- [X] Yes, I searched the [mailing list](https://groups.google.com/forum/#!forum/chipyard)
- [X] Yes, I searched [prior issues](https://github.com/ucb-bar/chipyard/issues)
- [X] Yes, I searched the [documentation](https://chipyard.readthedocs.io/)

### Chipyard Version and Hash

Release: 1.6.0
Hash: dcf8da4

### OS Setup

Red Hat Enterprise Linux Server release 7.9 (Maipo)

### Other Setup

Installed riscv-tools and sourced env.sh
The standard RocketConfig runs correctly and simulates with no errors with verilator

Created the following in generators/chipyard/src/main/scala/config/RocketConfigs.scala:

```
class SmallRocketConfig extends Config( // TO TEST
  new freechips.rocketchip.subsystem.WithNSmallCores(1) ++         // single rocket-core no VM, fp
  new chipyard.config.AbstractConfig)
```
Created this in generators/rocket-chip/src/main/scala/subsystem/Configs.scala:

```
class WithNSmallCores(n: Int, overrideIdOffset: Option[Int] = None) extends Config((site, here, up) => {
  case RocketTilesKey => {
    val prev = up(RocketTilesKey, site)
    val idOffset = overrideIdOffset.getOrElse(prev.size)
    val small = RocketTileParams(
      core = RocketCoreParams(useVM = false, fpu = None),
      btb = None,
      dcache = Some(DCacheParams(
        rowBits = site(SystemBusKey).beatBits,
        nSets = 64,
        nWays = 1,
        nTLBSets = 1,
        nTLBWays = 4,
        nMSHRs = 0,
        blockBytes = site(CacheBlockBytes))),
      icache = Some(ICacheParams(
        rowBits = site(SystemBusKey).beatBits,
        nSets = 64,
        nWays = 1,
        nTLBSets = 1,
        nTLBWays = 4,
        blockBytes = site(CacheBlockBytes))))
    List.tabulate(n)(i => small.copy(hartId = i + idOffset)) ++ prev
  }
})
```

### Current Behavior

when running `make CONFIG=SmallRocketConfig` in sims/verilator
```
[error] Exception in thread "main" firrtl.passes.CheckWidths$BitsWidthException:  @[PTW.scala 291:61]: [target ~TestHarness|PTW] High bit 26 in bits operator is larger than input width 21 in bits(r_req.addr, 26, 0).
[error] Nonzero exit code returned from runner: 1 
[error] (Compile / runMain) Nonzero exit code returned from runner: 1 
[error] Total time: 19 s, completed Mar 7, 2022 11:12:15 AM  
make: *** [firrtl_temp] Error 1 
```

### Expected Behavior

Expected the config to build successfully

### Other Information

I'm rather new to Chipyard and RocketChip, I found no information to debug this issue in the docks, and generally how to tell if the configs/parameter combinations are sound. Any help would be greatly appreciated
[small_config.log](https://github.com/ucb-bar/chipyard/files/8196897/small_config.log)

francesco-peverelli, 2022-03-08:
I have been able to build the config by adding `new chipyard.config.WithL2TLBs(0) ++ ` to the config. I still get the warning you see in the log, but the asm tests pass for this config. Is this correct?

jerryz123, 2022-03-08:
That seems like a good workaround. Let's keep this issue open for now, this should be fixed in upstream rocket chip

AlviseDeFaveri, 2023-03-16:
+1! Thanks @francesco-peverelli for the suggestion, you saved me from a long night :)

ashutgupta28, 2023-12-20:
Hi I am facing the same Issue, Is there the issue resolved ?

```
[error] Exception in thread "main" firrtl.passes.CheckWidths$BitsWidthException:  @[PTW.scala 291:61]: [target ~TestHarness|PTW] High bit 26 in bits operator is larger than input width 21 in bits(r_req.addr, 26, 0).
[error] Nonzero exit code returned from runner: 1 
[error] (Compile / runMain) Nonzero exit code returned from runner: 1 
```