ucb-bar/chipyard #1144: Missing dependencies ?

DorianXGH, 2022-03-21:
### Background Work

- [X] Yes, I searched the [mailing list](https://groups.google.com/forum/#!forum/chipyard)
- [X] Yes, I searched [prior issues](https://github.com/ucb-bar/chipyard/issues)
- [X] Yes, I searched the [documentation](https://chipyard.readthedocs.io/)

### Chipyard Version and Hash

Release: master
Hash: 51fea

### OS Setup

Linux pompeius 5.16.13-arch1-1 #1 SMP PREEMPT Tue, 08 Mar 2022 20:07:36 +0000 x86_64 GNU/Linux
LSB Version:	n/a
Distributor ID:	Arch
Description:	Arch Linux
Release:	rolling
Codename:	n/a

### Current Behavior

In sims/verilator/, `make CONFIG=SmallBoomConfig`
Doesn't compile, doesn't find apache commons dependencies : java.lang.NoClassDefFoundError: org/apache/commons/lang3/Validate
Adding 
```scala
"org.apache.commons" % "commons-lang3" % "3.12.0",
"org.apache.commons" % "commons-text" % "1.9"
```
In the right dependency Seq, fixes the issue.

### Expected Behavior

Compiles.

### Other Information

I don't know whether or not the apache commons are *supposed* to be preinstalled and already in the CLASSPATH, but it doesn't seem so on arch.
I am posting the issue to ask whether or not to make a pull request with the aforementioned additions to the build.sbt.

abejgonzalez, 2022-03-22:
I personally don't know of anyone that has tried using Arch for Chipyard tasks. Did you try installing similar things found in the `-req.sh` scripts here: https://github.com/ucb-bar/chipyard/blob/main/scripts/ubuntu-req.sh  https://github.com/ucb-bar/chipyard/blob/main/scripts/centos-req.sh?

DorianXGH, 2022-03-22:
Oh, like I said in the issue, it's not any package listed on these scripts. On Arch, it seems like the apache commons are in several separate packages. I think listing these as a dependency in the build system in build.sbt is appropriate, as they ARE a dependency, and it should be distribution independent, but then again I come from the Arch school of thought where every dependency should be listed as such x) (no implicit dependency)
The problem is fixed ***for me*** but I wanted to know if this was by design or if you would welcome a pull request adding the dependency to the build.sbt

abejgonzalez, 2022-03-22:
I would go ahead and add a PR for it. We can run it through our CI flow and see if anything breaks (but I think things will be fine). Thanks!

commit 20fb29a0 (2022-07-21): Merge pull request #1147 from ucb-bar/mergify/copy/main/pr-1146

Added missing apache commons dependencies (fixes #1144) (copy #1146)
closed 2022-06-15 by sagark