ucb-bar/chipyard #1148: build the riscv-tools toolchain on RHEL6

Leon924, 2022-04-04:
### Background Work

- [X] Yes, I searched the [mailing list](https://groups.google.com/forum/#!forum/chipyard)
- [X] Yes, I searched [prior issues](https://github.com/ucb-bar/chipyard/issues)
- [X] Yes, I searched the [documentation](https://chipyard.readthedocs.io/)

### Chipyard Version and Hash

Release: git checkout 1.6.2

### OS Setup

Distributor ID:	RedHatEnterpriseServer
Description:	Red Hat Enterprise Linux Server release 6.10 (Santiago)
Release:	6.10
Codename:	Santiago

### Other Setup

1. manually installed gcc-4.8.5 in /usr/local/bin/x86_64-unknown-linux-gnu-gcc
2. manually installed bison3.0.4 in my homedir and add it to PATH variable
3. manually installe boost-1.53 library in my homedir ‘xx/boostlib-1.53’ as BOOST_ROOT，and setup 
`export BOOST_INCLUDE=$BOOST_ROOT/include/boost
export BOOST_LIB=$BOOST_ROOT/lib
export BOOST_LIBRARYDIR=$BOOST_ROOT/lib
export CPLUS_INCLUDE_PATH=$BOOST_ROOT/include/boost
export LIBRARY_PATH=$LD_LIBRARY_PATH$BOOST_ROOT/lib
export BOOST_IGNORE_SYSTEM_PATHS=1`

### Current Behavior

when I build toolchain of riscv-tools: 
` ./scripts/build-toolchains.sh riscv-tools --ignore-qemu |  tee -i build-toolchain.log`
The first error  comes with 
`Makefile:1637: recipe for target 'auto-load.o' failed
gmake[3]: *** [auto-load.o] Error 1`
the detailed log is below 
[build-toolchain.log](https://github.com/ucb-bar/chipyard/files/8409564/build-toolchain.log)

### Expected Behavior

So  Is there any method to fix this？ What's the root cause of this peoblem?

a0u, 2022-04-05:
The actual error is:
```
  CXX    auto-load.o
...
chipyard/toolchains/riscv-tools/riscv-gnu-toolchain/build/../riscv-gdb/gdb/auto-load.c: In function 'int collect_matching_scripts(void**, void*)':
chipyard/toolchains/riscv-tools/riscv-gnu-toolchain/build/../riscv-gdb/gdb/auto-load.c:1211:66: error: 're_exec' was not declared in this scope
   if (script->language == data->language && re_exec (script->name))
```

An earlier [`configure` check](https://github.com/riscv-collab/riscv-binutils-gdb/blob/c3eb4078520dad8234ffd7fbf893ac0da23ad3c8/gdb/configure.ac#L1430-L1457) for system regex support  indicates that a recent enough version of glibc  is detected, which is being used over the included libiberty regex implementation:
```
checking for GNU regex... yes
```

So `re_exec()` is expected to be declared in `<regex.h>`, which should have been included by `gdb/auto-load.c` through [`gdb_regex.h`](https://github.com/riscv-collab/riscv-binutils-gdb/blob/c3eb4078520dad8234ffd7fbf893ac0da23ad3c8/gdb/gdb_regex.h#L26-L27).

This discrepancy might be caused by a mismatch between `gcc` and `g++`.   `gcc` (`CC`) is used for the autoconf test, whereas `g++` (`CXX`) is used to build gdb.  It's unclear from the issue description how GCC 4.8.5 was manually installed and what provides `g++`.

On RHEL, I recommend that you obtain a more current version of GCC through `devtoolset` instead of building your own toolchain:
https://www.softwarecollections.org/en/scls/rhscl/devtoolset-8/

Leon924, 2022-04-05:
Thanks，Albert.
1. About munally installed gcc-4.8.5
I download the [gcc-4.8.5](http://ftp.gnu.org/gnu/gcc/gcc-4.8.5/gcc-4.8.5.tar.gz) tarball source, and compile , the installed it to `/usr/local/` folder. 
secondly use 
`update-alternative --install /usr/bin/gcc gcc /usr/local/bin/x86_64-unknown-linux-gnu-gcc 100 ; 
update-alternative --config gcc` 
to choose gcc-4.8.5 as a prior version.
and I check gcc and g++, both of them are 4.8.5

2. I  found that `/usr/local/` has glibc2.14 and glic2.15, So I remove them and try with gcc/g++-4.8.5 again. It's still the same error 
[buildriscv.log](https://github.com/ucb-bar/chipyard/files/8420804/buildriscv.log)

3. As you recomend, I installed devtoolset-8 full library, and enable it in bash shell. gcc/g++ version changes into 8.3.1，it output the same error also. below is log 
[buildriscv-devtoolset.log](https://github.com/ucb-bar/chipyard/files/8420805/buildriscv-devtoolset.log)

Leon924, 2022-04-08:
I have remove gcc-4.8.5 and related lib , the try with devtoolset-8 (gcc-9.1.1) again. 
The new error comes out:
`g++: /rror: unrecognized command line option '-rdynamic'`
How can I fix this on  RHEL6.10?

Leon924, 2022-04-08:
fixed

a0u, 2022-04-11:
I wasn't able to reproduce the errors on RHEL 6 ELS (6.10).

The build succeeds for me on a clean EC2 instance using devtoolset-8.
```sh
# RHSCL repository ID is different for AWS
sudo yum --enablerepo='rhui-REGION-rhel-server-rhscl' install devtoolset-8 rh-python36
sudo yum install autoconf automake mpfr-devel gmp-devel gawk bison flex texinfo patchutils zlib-devel expat-devel
. /opt/rh/devtoolset-8/enable
. /opt/rh/rh-python36/enable
```

I did need to install bison 3.8.2 from source, as the system version (2.7.1) was too old to build riscv-glibc.  (Also note that 3.8 has as regression that breaks the glibc build.)
```sh
curl -L https://ftp.gnu.org/gnu/bison/bison-3.8.2.tar.gz | tar -xzvf -
cd bison-3.8.2
./configure --prefix="${HOME}/local"
make install
```

device-tree-compiler is later needed for riscv-isa-sim:
```sh
curl -L 'https://git.kernel.org/pub/scm/utils/dtc/dtc.git/snapshot/dtc-1.6.1.tar.gz' | tar -xzvf -
cd dtc-1.6.1
make install NO_PYTHON=1 PREFIX="${HOME}/local"
```

And finally:
```sh
export PATH="${HOME}/local/bin:${PATH}"
cd chipyard
./scripts/build-toolchains.sh --ignore-qemu
```

a0u, 2022-04-11:
riscv-gcc's configure script tests for `-rdynamic` support.  Check `chipyard/toolchains/riscv-tools/riscv-gnu-toolchain/build/build-gcc-newlib-stage1/gcc/config.log` to confirm that the intended compiler is being used.
Make sure to start from a clean environment.

```
configure:3855: checking for x86_64-pc-linux-gnu-gcc
configure:3882: result: gcc
configure:4151: checking for C compiler version
configure:4160: gcc --version >&5
gcc (GCC) 8.3.1 20190311 (Red Hat 8.3.1-3)
Copyright (C) 2018 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

...
configure:4748: checking for C++ compiler version
configure:4757: g++ --version >&5
g++ (GCC) 8.3.1 20190311 (Red Hat 8.3.1-3)
Copyright (C) 2018 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

...
configure:29908: checking for -rdynamic
configure:29918: result: yes
```

bchetwynd, 2022-07-27:
I don't know if this will help, but we worked through some of the issues with compiling the RISCV toolchain on RHEL7. 

The README.md @ https://github.com/mit-ll/CEP has some instructions.