ucb-bar/chipyard #1158: Toolset not compiling on Mac M1

sandeepwadhwa-ai, 2022-04-10:
### Background Work

- [X] Yes, I searched the [mailing list](https://groups.google.com/forum/#!forum/chipyard)
- [X] Yes, I searched [prior issues](https://github.com/ucb-bar/chipyard/issues)
- [X] Yes, I searched the [documentation](https://chipyard.readthedocs.io/)

### Chipyard Version and Hash

Release: 1.6.2
Hash: a6a6a6

### OS Setup

Ex: Output of `uname -a` 
Darwin Sandeeps-MacBook-Air.local 21.4.0 Darwin Kernel Version 21.4.0: Fri Mar 18 00:47:26 PDT 2022; root:xnu-8020.101.4~15/RELEASE_ARM64_T8101 arm64

and `lsb_release -a`
Docent exist on Mac OS.

### Other Setup

Ex: Prior steps taken / Documentation Followed / etc...

make version 4.2 installed. 

### Current Behavior

cd /Users/sandy/ComputerArchitecture/chipyard/toolchains/riscv-tools/riscv-gnu-toolchain/build/../riscv-newlib/libgloss/riscv && autoconf
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/gm4: unrecognized option `--gnu'
Try `/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/gm4 --help' for more information.
autom4te: error: gm4 failed with exit status: 1
make[4]: *** [Makefile:248: /Users/sandy/ComputerArchitecture/chipyard/toolchains/riscv-tools/riscv-gnu-toolchain/build/../riscv-newlib/libgloss/riscv/configure] Error 1
make[4]: Leaving directory '/Users/sandy/ComputerArchitecture/chipyard/toolchains/riscv-tools/riscv-gnu-toolchain/build/build-newlib/riscv64-unknown-elf/libgloss/riscv'
make[3]: *** [Makefile:132: stmp-bsp] Error 2
make[3]: Leaving directory '/Users/sandy/ComputerArchitecture/chipyard/toolchains/riscv-tools/riscv-gnu-toolchain/build/build-newlib/riscv64-unknown-elf/libgloss'
make[2]: *** [Makefile:9557: all-target-libgloss] Error 2
make[2]: Leaving directory '/Users/sandy/ComputerArchitecture/chipyard/toolchains/riscv-tools/riscv-gnu-toolchain/build/build-newlib'
make[1]: *** [Makefile:908: all] Error 2
make[1]: Leaving directory '/Users/sandy/ComputerArchitecture/chipyard/toolchains/riscv-tools/riscv-gnu-toolchain/build/build-newlib'
make: *** [Makefile:464: stamps/build-newlib] Error 2
make: *** Waiting for unfinished jobs....
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/gm4: unrecognized option `--gnu'
Try `/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/gm4 --help' for more information.
autom4te: error: gm4 failed with exit status: 1
make[4]: *** [Makefile:248: /Users/sandy/ComputerArchitecture/chipyard/toolchains/riscv-tools/riscv-gnu-toolchain/build/../riscv-newlib/libgloss/riscv/configure] Error 1
make[4]: Leaving directory '/Users/sandy/ComputerArchitecture/chipyard/toolchains/riscv-tools/riscv-gnu-toolchain/build/build-newlib-nano/riscv64-unknown-elf/libgloss/riscv'
make[3]: *** [Makefile:132: stmp-bsp] Error 2
make[3]: Leaving directory '/Users/sandy/ComputerArchitecture/chipyard/toolchains/riscv-tools/riscv-gnu-toolchain/build/build-newlib-nano/riscv64-unknown-elf/libgloss'
make[2]: *** [Makefile:9557: all-target-libgloss] Error 2
make[2]: Leaving directory '/Users/sandy/ComputerArchitecture/chipyard/toolchains/riscv-tools/riscv-gnu-toolchain/build/build-newlib-nano'
make[1]: *** [Makefile:908: all] Error 2
make[1]: Leaving directory '/Users/sandy/ComputerArchitecture/chipyard/toolchains/riscv-tools/riscv-gnu-toolchain/build/build-newlib-nano'
make: *** [Makefile:481: stamps/build-newlib-nano] Error 2
~

### Expected Behavior

Should compile with Done! message

a0u, 2022-04-10:
Follow the riscv-gnu-toolchain README to install the required dependencies:
https://github.com/riscv-collab/riscv-gnu-toolchain#prerequisites

The version of GNU m4 shipped with Xcode is evidently too old.

sandeepwadhwa-ai, 2022-04-11:
Firstly two .c files - "rltty.c" and "terminal.c" in 'readline' directory need to be changed.
I added in "#include <sys/ioctl.h>" in both files to circumvent compile errors. 

The page talks of using these three commands to install the relevant packages. I tried using the M1-Airbook in both modes(i.e native ARM and x86 emulation through Rosetta emulation). 

$brew install python3 gawk gnu-sed gmp mpfr libmpc isl zlib expat
$ brew tap discoteq/discoteq
$ brew install flock

Still have the same error in both modes towards the last. 
my make --version output is here.
----------
make --version
GNU Make 4.3
Built for x86_64-apple-darwin21.4.0
Copyright (C) 1988-2020 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.

------------ Console output where the error occurs. This is for x86 emulation mode ---------
make[4]: Nothing to be done for 'all'.
make[4]: Leaving directory '/Users/sandy/ComputerArchitecture/chipyard/toolchains/riscv-tools/riscv-gnu-toolchain/build/build-newlib/riscv64-unknown-elf/libgloss/doc'
make[4]: Entering directory '/Users/sandy/ComputerArchitecture/chipyard/toolchains/riscv-tools/riscv-gnu-toolchain/build/build-newlib-nano/riscv64-unknown-elf/libgloss/riscv'
cd /Users/sandy/ComputerArchitecture/chipyard/toolchains/riscv-tools/riscv-gnu-toolchain/build/../riscv-newlib/libgloss/riscv && autoconf
make[4]: Entering directory '/Users/sandy/ComputerArchitecture/chipyard/toolchains/riscv-tools/riscv-gnu-toolchain/build/build-newlib/riscv64-unknown-elf/libgloss/riscv'
cd /Users/sandy/ComputerArchitecture/chipyard/toolchains/riscv-tools/riscv-gnu-toolchain/build/../riscv-newlib/libgloss/riscv && autoconf
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/gm4: unrecognized option `--gnu'
Try `/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/gm4 --help' for more information.
autom4te: error: gm4 failed with exit status: 1
make[4]: *** [Makefile:248: /Users/sandy/ComputerArchitecture/chipyard/toolchains/riscv-tools/riscv-gnu-toolchain/build/../riscv-newlib/libgloss/riscv/configure] Error 1
make[4]: Leaving directory '/Users/sandy/ComputerArchitecture/chipyard/toolchains/riscv-tools/riscv-gnu-toolchain/build/build-newlib-nano/riscv64-unknown-elf/libgloss/riscv'
make[3]: *** [Makefile:132: stmp-bsp] Error 2
make[3]: Leaving directory '/Users/sandy/ComputerArchitecture/chipyard/toolchains/riscv-tools/riscv-gnu-toolchain/build/build-newlib-nano/riscv64-unknown-elf/libgloss'
make[2]: *** [Makefile:9557: all-target-libgloss] Error 2
make[2]: Leaving directory '/Users/sandy/ComputerArchitecture/chipyard/toolchains/riscv-tools/riscv-gnu-toolchain/build/build-newlib-nano'
make[1]: *** [Makefile:908: all] Error 2
make[1]: Leaving directory '/Users/sandy/ComputerArchitecture/chipyard/toolchains/riscv-tools/riscv-gnu-toolchain/build/build-newlib-nano'
make: *** [Makefile:481: stamps/build-newlib-nano] Error 2
make: *** Waiting for unfinished jobs....
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/gm4: unrecognized option `--gnu'
Try `/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/gm4 --help' for more information.
autom4te: error: gm4 failed with exit status: 1
make[4]: *** [Makefile:248: /Users/sandy/ComputerArchitecture/chipyard/toolchains/riscv-tools/riscv-gnu-toolchain/build/../riscv-newlib/libgloss/riscv/configure] Error 1
make[4]: Leaving directory '/Users/sandy/ComputerArchitecture/chipyard/toolchains/riscv-tools/riscv-gnu-toolchain/build/build-newlib/riscv64-unknown-elf/libgloss/riscv'
make[3]: *** [Makefile:132: stmp-bsp] Error 2
make[3]: Leaving directory '/Users/sandy/ComputerArchitecture/chipyard/toolchains/riscv-tools/riscv-gnu-toolchain/build/build-newlib/riscv64-unknown-elf/libgloss'
make[2]: *** [Makefile:9557: all-target-libgloss] Error 2
make[2]: Leaving directory '/Users/sandy/ComputerArchitecture/chipyard/toolchains/riscv-tools/riscv-gnu-toolchain/build/build-newlib'
make[1]: *** [Makefile:908: all] Error 2
make[1]: Leaving directory '/Users/sandy/ComputerArchitecture/chipyard/toolchains/riscv-tools/riscv-gnu-toolchain/build/build-newlib'
make: *** [Makefile:464: stamps/build-newlib] Error 2
--------------------------

a0u, 2022-04-12:
Try installing a newer GNU m4:
```sh
brew install m4
```

----

It also turns out that the gcc-newlib-stage1 bootstrap later fails on arm64 with a link error:
```
Undefined symbols for architecture arm64:
  "_host_hooks", referenced from:
      gt_pch_save(__sFILE*) in libbackend.a(ggc-common.o)
      gt_pch_restore(__sFILE*) in libbackend.a(ggc-common.o)
      toplev::main(int, char**) in libbackend.a(toplev.o)
ld: symbol(s) not found for architecture arm64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
```

Apply this hack from riscv-software-src/homebrew-riscv@5b72f7c19c6a54f2b36b4e7c4b3be0cc63b2e51a to `chipyard/toolchains/riscv-tools/riscv-gnu-toolchain/riscv-gcc`:

```diff
diff --git a/gcc/config.host b/gcc/config.host
index 816a0f06cb7..a6595977957 100644
--- a/gcc/config.host
+++ b/gcc/config.host
@@ -93,8 +93,6 @@ esac
 case ${host} in
   *-darwin*)
     # Generic darwin host support.
-    out_host_hook_obj=host-darwin.o
-    host_xmake_file="${host_xmake_file} x-darwin"
     ;;
 esac
```

If you're not using a case-sensitive filesystem, you'll also run into problems with building riscv-glibc later for the riscv64-unknown-linux-gnu toolchain.

jerryz123, 2023-02-17:
We don't officially support native compile on M1, unfortunately. The standard conda-flow is x86 only

closed 2023-02-17 by jerryz123