ucb-bar/chipyard #1164: a error i meet when run vlsi flow

1bing2, 2022-04-29:
### Background Work

- [X] Yes, I searched the [mailing list](https://groups.google.com/forum/#!forum/chipyard)
- [X] Yes, I searched [prior issues](https://github.com/ucb-bar/chipyard/issues)
- [X] Yes, I searched the [documentation](https://chipyard.readthedocs.io/)

### Chipyard Version and Hash

Release: 1.5.0
Hash: a6a6a6

### OS Setup

Ex: Output of `uname -a` and `lsb_release -a`

### Other Setup

Ex: Prior steps taken / Documentation Followed / etc...

### Current Behavior

when i run vlsi flow on chipyard,i get a error.
i run the command ：make buildfile CONFIG=TinyRocketConfig

error:Total time: 59 s, completed Apr 28, 2022 9:41:02 PM
/home/lee/Desktop/riscv-boom/verilator/chipyard/common.mk:110: recipe for target 'generator_temp' failed
make: *** [generator_temp] Error 1

### Expected Behavior

i have tried a lot of methods.but they don't work.

harrisonliew, 2022-04-29:
It looks like you had trouble running the Chisel Generator itself, and this is not related to the VLSI flow. Did you satisfy the [environment requirements](https://chipyard.readthedocs.io/en/stable/Chipyard-Basics/Initial-Repo-Setup.html#requirements)? If so, can you post the stack trace from sbt?

1bing2, 2022-04-30:
> It looks like you had trouble running the Chisel Generator itself, and this is not related to the VLSI flow. Did you satisfy the [environment requirements](https://chipyard.readthedocs.io/en/stable/Chipyard-Basics/Initial-Repo-Setup.html#requirements)? If so, can you post the stack trace from sbt?

i have satisfy the environment  requirements.but when i run make CONFIG=SmallBoomConfig .it have this error
error:Total time: 26 s, completed Apr 30, 2022 4:37:08 PM
/home/lee/Desktop/riscv-boom/verilator/chipyard/common.mk:110: recipe for target 'generator_temp' failed
make: *** [generator_temp] Error 1

harrisonliew, 2022-04-30:
We need to know what failed, not just that the Make target failed to help you. If you look at the `generator_temp` target: https://github.com/ucb-bar/chipyard/blob/main/common.mk#L109-L116 you can see that it just runs a `mkdir` and then `sbt`. Can you post the sbt stack trace/error log? Or do you have trouble even running `mkdir` or `sbt`?

1bing2, 2022-05-01:
ok!
My Error Log put into attachment
Thank you!
[MY error.txt](https://github.com/ucb-bar/chipyard/files/8598175/MY.error.txt)

harrisonliew, 2022-05-01:
Thanks for attaching the log.

So it looks like you do not have the correct dsptools and testchipip submodules. Did you run the `scripts/init-submodules-no-riscv-tools.sh` script? If so, can you check that the commits of those submodules match the ones you see here: https://github.com/ucb-bar/chipyard/tree/<tag>/tools (replace `<tag>` with whichever release of Chipyard you are using)?

1bing2, 2022-05-02:
> 

i have run the  scripts/init-submodules-no-riscv-tools.sh script and it comes the results are:
Setting up official Chipyard release: 1.6.2
Skipping submodule 'fpga/fpga-shells'
Skipping submodule 'generators/gemmini'
Skipping submodule 'generators/sha3'
Skipping submodule 'sims/firesim'
Skipping submodule 'software/coremark'
Skipping submodule 'software/firemarshal'
Skipping submodule 'software/nvdla-workload'
Skipping submodule 'software/spec2017'
Skipping submodule 'toolchains/esp-tools/riscv-gnu-toolchain'
Skipping submodule 'toolchains/esp-tools/riscv-isa-sim'
Skipping submodule 'toolchains/esp-tools/riscv-pk'
Skipping submodule 'toolchains/esp-tools/riscv-tests'
Skipping submodule 'toolchains/libgloss'
Skipping submodule 'toolchains/qemu'
Skipping submodule 'toolchains/riscv-tools/riscv-gnu-toolchain'
Skipping submodule 'toolchains/riscv-tools/riscv-gnu-toolchain-prebuilt'
Skipping submodule 'toolchains/riscv-tools/riscv-isa-sim'
Skipping submodule 'toolchains/riscv-tools/riscv-openocd'
Skipping submodule 'toolchains/riscv-tools/riscv-pk'
Skipping submodule 'toolchains/riscv-tools/riscv-tests'
Skipping submodule 'vlsi/hammer-cadence-plugins'
Skipping submodule 'vlsi/hammer-mentor-plugins'
Skipping submodule 'vlsi/hammer-synopsys-plugins'

harrisonliew, 2022-05-03:
Can you verify that you have the correct commit for your `dsptools` and `testchipip` submodules? Also, it looks like you are trying to elaborate the included FFT generator, is that correct?

1bing2, 2022-05-04:
i have verified these modules,but these modules don't loss anything and my FFT generator is correct . i have compare my generator with the generator on the https://github.com/ucb-bar/chipyard/tree/main/generators.but they both same.

michael-etzkorn, 2022-05-04:
Did you try building anything in your cloned directory before changing commits or running any initialization scripts? That can impact the whole cloned directory. Since Chipyard doesn't really have a proper clean rule at this time, you can try cloning a fresh repo and running the init script again. You shouldn't need to rebuild the toolchain if you leave that as built in the stale repo.

1bing2, 2022-05-05:
> nce Chipyard doesn't really have a proper clean rule at this time, you can try cloning a fresh repo and running the init script again. You shouldn't need to rebuild the toolchain if you leave that as built in the stale repo.

Can you describe it in detail.I don't konw it very well

michael-etzkorn, 2022-05-05:
My suggestion is to remove any doubt that it is an issue with the setup ordering. It's possible to confuse the initialization script if you start running make rules before the set-up scripts. Chipyard also [doesn't support switching versions](https://chipyard.readthedocs.io/en/stable/Chipyard-Basics/Initial-Repo-Setup.html#upgrading-chipyard-release-versions) without a fresh repo. If you're confident you didn't checkout to different commits between build attempts or didn't attempt to run a build before running the initialization scripts, this probably will not fix your build issue, but it can help remove doubt that there was a problem with ordering during setup. 

Since you believe you already followed [requirements](https://chipyard.readthedocs.io/en/stable/Chipyard-Basics/Initial-Repo-Setup.html#requirements) correctly:

follow [repo setup](https://chipyard.readthedocs.io/en/stable/Chipyard-Basics/Initial-Repo-Setup.html#setting-up-the-chipyard-repo), but with a second clone of the repo. Your riscv-tools will still be set-up in the first repo (or you can remove that and start over completely). 

```
git clone https://github.com/ucb-bar/chipyard.git chipyard2
cd chipyard2
# checkout latest official chipyard release
# note: this may not be the latest release if the documentation version != "stable"
git checkout 1.6.2
./scripts/init-submodules-no-riscv-tools.sh
```

Only *after* the script tool finishes, run `make CONFIG=SmallBoomConfig` within the verilator sim folder. 

You can then look to set-up the VLSI flow and run that: https://chipyard.readthedocs.io/en/stable/VLSI/ASAP7-Tutorial.html?highlight=vlsi#initial-setup

1bing2, 2022-05-09:
Thank you a lot! I will have a try