ucb-bar/chipyard #1169: Error about `a.bits.user.amba_prot` signals

vhg0299, 2022-05-30:
### Background Work

- [X] Yes, I searched the [mailing list](https://groups.google.com/forum/#!forum/chipyard)
- [X] Yes, I searched [prior issues](https://github.com/ucb-bar/chipyard/issues)
- [X] Yes, I searched the [documentation](https://chipyard.readthedocs.io/)

### Chipyard Version and Hash

Release: 1.5.0
Hash: a6a6a6

### OS Setup

Ex: Output of `uname -a` and `lsb_release -a`

### Other Setup

Ex: Prior steps taken / Documentation Followed / etc...

### Current Behavior

I integrated a custom RISC-V core that uses AXI interface into the Chipyard and built a simulator successfully.

Now, I want to build the core config bitstream for Arty. (The core uses AXI interface)

When `ddrClientBundle` is defined: `val ddrClientBundle = Wire(new HeterogeneousBag(bundles.map(.cloneType)))`, I get error about `a.bits.user.amba_prot` signals

So there was a `$RefNotlnitializedException` in FIRTL pass.

```
[error] firrti.passes.PassExceptions:
lerror].firrtl.passes.Checklnitialization.$RefNotInitializedException:@[HarnessBinders.scala.]:[moduleArtyFPGATestHarness]Reference_WIRE_2 is not fully initialized.
[error]:_WIRE_2.0.a.bits.user.amba_prot.fetch <= VOID
```

### Expected Behavior

Does anyone have any ideas about this error?

michael-etzkorn, 2022-05-30:
Yeah, I worked around this by exposing the AXI ports instead of TL ports when I need to connect them to AXI outside of ChipTop. I think it's because the DDR mem port isn't cloning the BundleMap user fields. 

https://stackoverflow.com/questions/69167589/what-are-these-a-bits-user-amba-prot-signals-and-why-are-they-only-uninitializ

leduchuybk, 2022-07-15:
@vhg0299 
i met similar problem.
can you update your work? have you fixed it?