ucb-bar/chipyard #1173: ASAP7 tutorial power-par error

odxa20, 2022-06-13:
### Background Work

- [X] Yes, I searched the [mailing list](https://groups.google.com/forum/#!forum/chipyard)
- [X] Yes, I searched [prior issues](https://github.com/ucb-bar/chipyard/issues)
- [X] Yes, I searched the [documentation](https://chipyard.readthedocs.io/)

### Chipyard Version and Hash

Release: 1.6.2
Hash: 481398b

### OS Setup

Linux edabox 4.18.0-372.9.1.el8.x86_64 #1 SMP Fri Apr 15 22:12:19 EDT 2022 x86_64 x86_64 x86_64 GNU/Linux

LSB Version:	:core-4.1-amd64:core-4.1-noarch:cxx-4.1-amd64:cxx-4.1-noarch:desktop-4.1-amd64:desktop-4.1-noarch:languages-4.1-amd64:languages-4.1-noarch:printing-4.1-amd64:printing-4.1-noarch
Distributor ID:	RedHatEnterprise
Description:	Red Hat Enterprise Linux release 8.6 (Ootpa)
Release:	8.6
Codename:	Ootpa

### Other Setup

Using genus 21.10, innovus 21.11 and voltus 21.12

### Current Behavior

When running the power-par build target I get the following error

"sim.inputs.execution_flags": [
        "+verbose",
        "+vcdplusfile=/home/odxa20/chipyard/vlsi/output/chipyard.TestHarness.TinyRocketConfig/rv32ui-p-simple.vpd",
        "+dramsim",
        "+dramsim_ini_dir=/home/odxa20/chipyard/generators/testchipip/src/main/resources/dramsim2_ini",
        "+max-cycles=10000000"
    ],
    "sim.inpmake: *** [/home/odxa20/chipyard/vlsi/build/chipyard.TestHarness.TinyRocketConfig-ChipTop/hammer.d:68: /home/odxa20/chipyard/vlsi/build/chipyard.TestHarness.TinyRocketConfig-ChipTop/power-sim-par-input.json] Error 1

### Expected Behavior

Should finish normally

harrisonliew, 2022-06-14:
At first glance, your error dump seems weirdly incomplete... can you post a few files first:
1. The Makefile-generated `sim-inputs.yml` file and `sim-debug-inputs.yml`
2. The Makefile-generated `hammer.d`
3. The Hammer-generated `power-sim-par-input.json` file

odxa20, 2022-06-14:
Hi thank you very much for the reply. I am posting the files you requested
[files.zip](https://github.com/ucb-bar/chipyard/files/8896891/files.zip)

harrisonliew, 2022-06-14:
Nothing appears to be wrong with the input files. Can you post the `hammer-vlsi-<timestamp>.log` file that corresponds to this error?

odxa20, 2022-06-14:
Of course. These are the last two log files that where output before the error
[hammer-vlsi-20220614-104059.log](https://github.com/ucb-bar/chipyard/files/8901311/hammer-vlsi-20220614-104059.log)
[hammer-vlsi-20220614-104018.log](https://github.com/ucb-bar/chipyard/files/8901312/hammer-vlsi-20220614-104018.log)

odxa20, 2022-06-14:
I just ran the command again (make power-par CONFIG=TinyRocketConfig BINARY=$RISCV/riscv64-unknown-elf/share/riscv-tests/isa/rv32ui-p-simple) after doing a make clean. The error message seems to be clearer

"sim.inputs.execution_flags": [
        "+verbose",
        "+vcdplusfile=/home/odxa20/chipyard/vlsi/output/chipyard.TestHarness.TinyRocketConfig/rv32ui-p-simple.vpd",
        "+dramsim",
        "+dramsim_ini_dir=/home/odxa20/chipyard/generators/testchipip/src/main/resources/dramsim2_ini",
        "+max-cycles=10000000"
    ],
    "sim.inpTraceback (most recent call last):
  File "./example-vlsi", line 62, in <module>
    ExampleDriver().main()
  File "/home/odxa20/chipyard/vlsi/hammer/src/hammer-vlsi/hammer_vlsi/cli_driver.py", line 1360, in main
    sys.exit(self.run_main_parsed(vars(parser.parse_args(args))))
  File "/home/odxa20/chipyard/vlsi/hammer/src/hammer-vlsi/hammer_vlsi/cli_driver.py", line 1283, in run_main_parsed
    print(output_str)
BlockingIOError: [Errno 11] write could not complete without blocking
make: *** [/home/odxa20/chipyard/vlsi/build/chipyard.TestHarness.TinyRocketConfig-ChipTop/hammer.d:68: /home/odxa20/chipyard/vlsi/build/chipyard.TestHarness.TinyRocketConfig-ChipTop/power-sim-par-input.json] Error 1

I am attaching all the .log files generated in this fresh run
[logs.zip](https://github.com/ucb-bar/chipyard/files/8902243/logs.zip)

harrisonliew, 2022-06-15:
I see. It looks like Python is choking up trying to write the `sim-to-par` output to `power-sim-par-input.json` quickly followed by a `print` to stdout. It seems like some output buffer in your system is overfilled. We haven't seen this before on our systems, but this definitely needs fixing.

I think this is a case of old sloppy programming. Can you add `f.close()` before the `print(output_str)` in your `/home/odxa20/chipyard/vlsi/hammer/src/hammer-vlsi/hammer_vlsi/cli_driver.py` at line 1283?

odxa20, 2022-06-15:
I made the change you suggested but got the same error

"synthesis.outputs.output_files": [
        "/home/odxa20/chipyard/vlsi/build/chipyard.TestHarness.TinyRocketConfig-ChipTop/syn-rundir/ChipTop.mapped.v"
    ],
    "synthesis.outputs.sdc": "/home/odxa20/chipyard/vlsi/build/chipyard.TestHarness.TinyRocketConfig-ChipTop/syn-rundir/ChipTop.mapped.sdc",
    "synthesis.outputs.seq_cells": "/home/odxa20/chipyard/vlsi/build/chipyard.TestHarness.TinyRocketConfig-ChipTop/syn-rundir/find_regs_cells.json",
    "synthesis.outputs.all_regs": "/home/odxa20/chipyard/vlsi/build/chipyard.TestHarness.TinyRocketConfig-ChipTop/syn-rundir/find_regs_paths.json",
    "synthesis.outputs.sdf_file": "/home/odxa20/chipyard/vlsi/build/chipyard.TestHarness.TinyRocketConfig-ChipTop/syn-rundir/ChipTop.mapped.sdf",
    "par.inputs.input_files": [
        "/home/odxa20/chipyard/vlsi/build/chipyard.TestHarness.TinyRocketConfig-ChipTop/syn-rundir/ChipTop.mapped.v"
    ],
    "par.inputs.top_module": "ChipTop",
    "par.inputs.post_synth_sdc": "/home/odxa20/chipyard/vlsi/build/chipyard.TestHarness.TinyRocketConfig-ChipTop/syn-rundir/ChipTop.mapped.sdc",
    "par.outputs.output_gds": "/home/odxa20/chipyard/vlsi/build/chipyard.TestHarness.TinyRocketConfig-ChipTop/par-rundir/ChipTop.gds",
    "par.outputs.output_netlist": "/home/odxa20/chipyard/vlsi/build/chipyard.TestHarness.TinyRocketConfig-ChipTop/par-rundir/ChipTop.lvs.v",
    "par.outputs.output_sim_netlist": "/home/odxa20/chipyard/vlsi/build/chipyard.TestHarness.TinyRocketConfig-ChipTop/par-rundir/ChipTop.sim.v",
    "par.outputs.hcells_list": [],
    "par.outputs.seq_cells": "/home/odxa20/chipyard/vlsi/build/chipyard.TestHarness.TinyRocketConfig-ChipTop/par-rundir/find_regs_cells.json",
    "par.outputs.all_regs": "/home/odxa20/chipyard/vlsi/build/chipyard.TestHarness.TinyRocketConfig-ChipTop/par-rundir/find_regs_paths.json",
    "par.outputs.sdf_file": "/home/odxa20/chipyard/vlsi/build/chipyard.TestHarness.TinyRocketConfig-ChipTop/par-rundir/ChipTop.par.sdf",
    "par.outputs.spefs": [
        "/home/odxa20/chipyard/vlsi/build/chipyard.TestHarness.TinyRocketConfig-ChipTop/par-rundir/ChipTop.PVT_0P63V_100C.par.spef",
        "/home/odxa20/chipyard/vlsi/build/chipyard.TestHarness.TinyRocketConfig-ChipTop/par-rundir/ChipTop.PVT_0P77V_0C.par.spef"
    ],
    "power.inputs.top_module": "ChipTop",
    "power.inputs.netlist": "/home/odxa20/chipyard/vlsi/build/chipyard.TestHarness.TinyRocketConfig-ChipTop/par-rundir/ChipTop.lvs.v",Traceback (most recent call last):
  File "./example-vlsi", line 62, in <module>
    ExampleDriver().main()
  File "/home/odxa20/chipyard/vlsi/hammer/src/hammer-vlsi/hammer_vlsi/cli_driver.py", line 1361, in main
    sys.exit(self.run_main_parsed(vars(parser.parse_args(args))))
  File "/home/odxa20/chipyard/vlsi/hammer/src/hammer-vlsi/hammer_vlsi/cli_driver.py", line 1284, in run_main_parsed
    print(output_str)
BlockingIOError: [Errno 11] write could not complete without blocking

    "power.inputs.spefs": [
        "/home/odxa20/chipyard/vlsi/build/chipyard.TestHarness.TinyRocketConfig-ChipTop/par-rundir/ChipTop.PVT_0P63V_100C.par.spef",
        "/home/odxa20/chipyard/vlsi/build/chipyard.TestHarness.TinyRocketConfig-ChipTop/par-rundir/ChipTop.PVT_0P77V_0C.par.spef"
    ]
}make: *** [/home/odxa20/chipyard/vlsi/build/chipyard.TestHarness.TinyRocketConfig-ChipTop/hammer.d:71: /home/odxa20/chipyard/vlsi/build/chipyard.TestHarness.TinyRocketConfig-ChipTop/power-par-input.json] Error 1

I am attaching the log files
[logs.zip](https://github.com/ucb-bar/chipyard/files/8907482/logs.zip)

If this is due to a buffer being full is there a way to increase the buffer size (e.g with some kind of ulimit command ?)
[logs.zip](https://github.com/ucb-bar/chipyard/files/8907496/logs.zip)

harrisonliew, 2022-06-15:
Looks like it got a little farther in the output print... but I see, the issue is actually not with the file write (`power-sim-par-input.json` in this case) but actually the `print` to `stdout`. 

Based on this: https://stackoverflow.com/questions/65876753/how-to-make-pythons-print-non-blocking it looks like you can bypass output buffering with a `flush=True` in the `print`. ulimit seems overkill/dangerous just to get around this.

Regardless, I think the `print(output_str)` is quite unnecessary from a usability perspective because it pollutes the terminal buffer for these X-to-Y steps. Can you instead just replace that line with something like `print("Output config written to " + args["output"])` and see if this error goes away?

odxa20, 2022-06-15:
This seems to have fixed it. Thank you very much. Unfortunately I am getting an other error.

`Begin Processing Power Net/Grid for Power Calculation

Rail status:
  0.63V    VDD
  0V    VSS

Ended Processing Power Net/Grid for Power Calculation: (cpu=0:00:00, real=0:00:00, mem(process/total/peak)=2729.66MB/12962.07MB/7500.04MB)

Begin Processing Timing Window Data for Power Calculation
   clock_clock(1000MHz)

  Timing Window Statistics:
    Waveform definitions parsed             : 1
    Clock roots assigned/parsed             : 1/1 (100%)
    Constants assigned/parsed               : 1342/1342 (100%)
    External loads assigned/parsed          : 37/37 (100%)
    Slews assigned/parsed                   : 123224/123224 (100%)
    Nets with slew/Nets in design           : 58532/58532 (100%)
    Slew range                              : 1.375e-12s - 3.96375e-10s

  SPEF    : /home/odxa20/chipyard/vlsi/build/chipyard.TestHarness.TinyRocketConfig-ChipTop/par-rundir/ChipTop.PVT_0P63V_100C.par.spef
    Name matched: 58495/58495
    Annotation coverage for this file      : 58495/58532 (99.9368%)

Ended Processing Timing Window Data for Power Calculation: (cpu=0:00:01, real=0:00:00, mem(process/total/peak)=2750.31MB/12960.06MB/7500.04MB)

Begin Processing VCD Vectors
Parsing VCD file /home/odxa20/chipyard/vlsi/output/chipyard.TestHarness.TinyRocketConfig/rv32ui-p-simple.vpd

Starting Reading VCD variables
2022-Jun-15 21:37:11 (2022-Jun-15 18:37:11 GMT)

----------------------   FATAL   ----------------------
Tcl command failed.
-->
** ERROR: (VOLTUS_POWR-1151): Voltus Power Analysis was unable to complete
during the processing of VCD.
The Error was:
** ERROR: (VOLTUS_POWR-1735): Unable to read the activity file because of the following error: {[/home/odxa20/chipyard/vlsi/output/chipyard.TestHarness.TinyRocketConfig/rv32ui-p-simple.vpd line 1, col 1] syntax error}. Fix the error and run power analysis again.

    while executing
"ChipPwr fnExecute dynamic"
    (file "/tmp/ssv_tmpdir_293729_8dWKPD/voltus_power_293729_3.cmd" line 74)

-------------------------------------------------------

Exit code 2.
Voltus Power Analysis exited unsuccessfully.
** ERROR: (VOLTUS_POWR-1390): Voltus Power Analysis could not complete successfully.
**ERROR: (IMPSE-110):	File '/home/odxa20/chipyard/vlsi/build/chipyard.TestHarness.TinyRocketConfig-ChipTop/power-par-rundir/power.tcl' line 42: 1.
#@ End verbose source /home/odxa20/chipyard/vlsi/build/chipyard.TestHarness.TinyRocketConfig-ChipTop/power-par-rundir/power.tcl
**ERROR: (UI-93):	1`

harrisonliew, 2022-06-16:
Glad to see that fixed it - I'll make a PR to Hammer.

Hm, unfortunately the logs do not capture the `run_simulation` output log part of the VCS simulation, so I'm unable to see if there were any issues with the VPD writing. Did VCS report a successful waveform dump?

Also, you're using Voltus 21.12. Do you know if the `read_activity_file` command in this newer version distriguishes `vpd` as distinct from the `vcd` format? If so, that might explain the syntax error?

odxa20, 2022-06-16:
vcs did not produce an error when running. Is there a way to check the validity of the .vpd file ? We unfortunately don't currently have access to older versions of Voltus. Is there a way to check if the read_activity_file command in this newer version distinguishes vpd and vcd formats ?

harrisonliew, 2022-06-16:
You can try to read the VPD into a waveform viewer. You have a couple options here:
1. If you still have licenses for DVE or WV, the VPD can be read directly
2. If not, you can run the `vpd2vcd` utility (same install dir as `vcs`) and then read it with Verdi or basically any other waveform viewer.

Yes, you can pull up either the Voltus Text Command Reference or in the Voltus terminal you can type `help read_activity_file` and it will list all the documentation.

harrisonliew, 2022-06-16:
> I just checked the Voltus Text Command Reference and there is no mention of vpd files.

In Voltus, VPD files are supposed to be read as VCD format. It would be odd if it no longer supports VPD...

odxa20, 2022-06-16:
In the attached file which I exported from the  Voltus Text Command Reference there is no mention of vpd (maybe it is implied they are treated the same as vcd ?). Does your version explicilty mention vpd files ?

harrisonliew, 2022-06-16:
Right, for many versions now it is implied they are treated the same as VCD. I'm pretty sure it's built on VCD but is a compressed binary format.

You could also try dumping in FSDB format instead. However, I have noticed that Voltus' FSDB reader is a few versions behind VCS' FSDB writer, so you'd need to use an older version of VCS.

odxa20, 2022-06-16:
I just opened the vpd file with wv and it seems fine. No errors and signals are showing. What would be the next step to find the issue ?

odxa20, 2022-06-16:
So I change line 189 of the vlsi/Makefile from `echo "sim.outputs.waveforms: ['$(sim_out_name).vpd']" >> $@` to `echo "sim.outputs.waveforms: ['~/rv32ui-p-simple.vcd']" >> $@` right ?

harrisonliew, 2022-06-16:
No, edit `power.inputs.waveforms`, otherwise you'll need to run the `sim-to-par` step again. This way you can directly run `power-par`.

odxa20, 2022-06-16:
Ok got it. Changed that and will run the power-par command. Once it finished I will post the results

odxa20, 2022-06-16:
I managed to run the `power-par` step with the converted vcd file. From my tests even when running `redo-power-par` the `power-inputs.yml` file gets overwritten so to do this I ran `power-par` once from scratch, then converted the file using vpd2vcd in the output folder and changed the extension in the Makefile to vcd and ran the `redo-power-par` command. Unfortunately more errors arise

```
Starting Calculating power
2022-Jun-16 22:35:21 (2022-Jun-16 19:35:21 GMT)
2022-Jun-16 22:35:22 (2022-Jun-16 19:35:22 GMT): 10%
2022-Jun-16 22:35:22 (2022-Jun-16 19:35:22 GMT): 20%
2022-Jun-16 22:35:22 (2022-Jun-16 19:35:22 GMT): 30%
2022-Jun-16 22:35:22 (2022-Jun-16 19:35:22 GMT): 40%
2022-Jun-16 22:35:22 (2022-Jun-16 19:35:22 GMT): 50%
2022-Jun-16 22:35:22 (2022-Jun-16 19:35:22 GMT): 60%
2022-Jun-16 22:35:22 (2022-Jun-16 19:35:22 GMT): 70%
2022-Jun-16 22:35:22 (2022-Jun-16 19:35:22 GMT): 80%
2022-Jun-16 22:35:22 (2022-Jun-16 19:35:22 GMT): 90%

Finished Calculating power
2022-Jun-16 22:35:22 (2022-Jun-16 19:35:22 GMT)
Ended Power Analysis: (cpu=0:00:05, real=0:00:00, mem(process/total/peak)=3666.79MB/13912.02MB/7728.89MB)

Begin Writing Current Files
  This vectcorbased dynamic simulation is from 0 second to 0.01 second.
  CPU              : 24
  Simulation Period: 1e+07ns
  Step Size        : 500ps

** WARN:  (VOLTUS_POWR-1812): Provided step size is very small compared to the simulation period.
This may have a significant impact on runtime and output file size.
You may want to increase the step size for better performance.

Starting Processing toggles for instances
2022-Jun-16 22:35:28 (2022-Jun-16 19:35:28 GMT)
** ERROR: (VOLTUS_POWR-1390): Voltus Power Analysis could not complete successfully.
**ERROR: (IMPSE-110):	File '/home/odxa20/chipyard/vlsi/build/chipyard.TestHarness.TinyRocketConfig-ChipTop/power-par-rundir/power.tcl' line 42: 1.
#@ End verbose source /home/odxa20/chipyard/vlsi/build/chipyard.TestHarness.TinyRocketConfig-ChipTop/power-par-rundir/power.tcl
**ERROR: (UI-93):	1
```

harrisonliew, 2022-06-16:
Ok, so it looks like it was able to run the `read_activity_file` (line 40) with the VCD file instead of the VPD. I'll make an issue in hammer-cadence-plugins to investigate why the Cadence tools don't support VPD anymore.

From the log, it looks like it was able to analyze the power (it says it finished calculating) but I think the warning could reveal something. Your simulation seems excessively long (0.01 seconds, or 2e7 time steps) - did the simulation timeout, or do you have a custom program that you're running? I have a suspicion that Voltus is unable to process the toggles for so many instances over so many timestamps and may have run out of resources, hence the error.

odxa20, 2022-06-16:
The test program I am using is the one from the tutorial rv32ui-p-simple which is quite small. Nonetheless when looking at the resource monitor once the simulation starts the systems ram (16GB) is immediately filled and swap space starts to get used. After a few seconds I receive the error so your hypothesis about resources seems to be correct. How can this be fixed ?

harrisonliew, 2022-06-16:
We need to fix the simulation timeout first so that your waveforms aren't 0.01s long. Unfortunately, there is a fix for ASAP7 SRAM behavioral models (ucb-bar/hammer#645) that hasn't yet made it into the Chipyard release. Without these fixes, certain simulations fail and thus timeout.

Can you update all of your Hammer and plugins repositories to their latest commits on `master` branch, clean, and try your flow again?

odxa20, 2022-06-16:
Should I delete and clone the repositories from scratch or try to do a merge ?

harrisonliew, 2022-06-16:
In each of the repositories (hammer, hammer-cadence-plugins, etc.), just `git checkout master && git pull` to get the latest versions. Then in your `vlsi/` worksplace, clean everything and run the `power-par` flow again with your VPD -> VCD conversion hack.

Pay special attention to the post-P&R simulation to make sure it shows a `PASS` instead of `TIMEOUT`. If it still times out, do not proceed to power analysis. Re-run the simulation with a much lower [`timeout_cycles` environment variable](https://github.com/ucb-bar/chipyard/blob/51fea1f8c610001558b4f8eda6fe199993b825ef/variables.mk#L29), then attach the VCD and please also try to debug the simulation failure concurrently with your waveform viewer of choice. I will not have time to debug this fully until perhaps next week.

odxa20, 2022-06-16:
Thank you very much for all the help. I updated the repositories and will proceed with performing the tests you described. Since you mentioned that this is probably caused by issues with SRAM models if I were to run the synthesis with synflops argument would that mean that the error will not present itself ? (wondering if I could run the power estimation with the synflops argument to at least see the tutorial through)

harrisonliew, 2022-06-16:
Yes, you can compile to synflops instead of SRAMs. However, you will need to remove the SRAM placement constraints from `example-asap7.yml` or else Innovus will complain that the instances are not found.

Power-wise - you'll get a more accurate result with flops rather than SRAMs since the ASAP7 SRAM are dummy models and contain no power information.

odxa20, 2022-06-16:
Got it thanks again. Once I run the sims I will post again here. Concurrently I will also run the flow in the synflops mode and try to adapt the tutorial to also do power estimation of boom

odxa20, 2022-06-18:
Hi. I created a new chipyard installation (v1.7.0) which contains the latest versions of hammer and hammer-plugins which you specified. I ran the sim-par-debug command and the simulation shows a **PASS**

```
ucli% source /home/odxa20/chipyard/vlsi/build/chipyard.TestHarness.TinyRocketConfig-ChipTop/sim-par-rundir/force_regs.ucli
ucli% run 0.0ns
[UART] UART0 is here (stdin/stdout).
testing $random 86561c0c seed          1
VCD+ Writer R-2020.12_Full64 Copyright (c) 1991-2020 by Synopsys Inc.
ClockDividerN.sv, 1 : // See LICENSE for license details.
ucli% power -gate_level on
ucli% power TestDriver.testHarness.chiptop
ucli% config endofsim noexit
noexit
ucli% run 10000000.0ns
*** PASSED *** Completed after                 5916 simulation cycles
$finish called from file "/home/odxa20/chipyard/vlsi/generated-src/chipyard.TestHarness.TinyRocketConfig/TestDriver.v", line 158.
$finish at simulation time             59155000
Simulation complete, time is 5915500000 fs.
TestDriver.v, 158 :         $finish;
ucli% power -report ucli.saif 1e-9 TestDriver.testHarness.chiptop
ucli% run
           V C S   S i m u l a t i o n   R e p o r t
Time: 5915500000 fs
CPU Time:    729.000 seconds;       Data structure size:  10.9Mb
Sun Jun 19 00:59:56 2022
ucli% exit
Action sim config output written to output.json
```
After doing the hack with the .vpd to .vcd conversion I ran power-par but unfortunately the same error occurs (ram gets filled up and voltus exits after a few seconds). Here is the output

```
Begin Writing Current Files
  This vectcorbased dynamic simulation is from 0 second to 0.01 second.
  CPU              : 24
  Simulation Period: 1e+07ns
  Step Size        : 500ps

** WARN:  (VOLTUS_POWR-1812): Provided step size is very small compared to the simulation period.
This may have a significant impact on runtime and output file size.
You may want to increase the step size for better performance.

Starting Processing toggles for instances
2022-Jun-19 01:06:14 (2022-Jun-18 22:06:14 GMT)
** ERROR: (VOLTUS_POWR-1390): Voltus Power Analysis could not complete successfully.
**ERROR: (IMPSE-110):	File '/home/odxa20/chipyard/vlsi/build/chipyard.TestHarness.TinyRocketConfig-ChipTop/power-par-rundir/power.tcl' line 42: 1.
#@ End verbose source /home/odxa20/chipyard/vlsi/build/chipyard.TestHarness.TinyRocketConfig-ChipTop/power-par-rundir/power.tcl
**ERROR: (UI-93):	1
```
I would greatly appreciate your help when you get the opportunity. If you are attending ISCA2022 I wish you the best. I watched the Chipyard tutorial remotely and am looking forward to watching the paper presentations

odxa20, 2022-06-18:
When looking at the vpd file in waveviewer I can see that its 5915.5ns long which matches with the output of VCS which indicates that the simulation ran for 5916 simulation cycles

harrisonliew, 2022-06-21:
I see that your VCS simulation is now not timed out. However, it looks like Voltus still interprets the waveform as having an end time of 0.01s, like the original error. Are you sure you converted the correct VPD to VCD? If so, are there any options in the `vpd2vcd` utility to truncate the end time of the waveform file?

odxa20, 2022-06-21:
Thank you very much for following up on this. I just ran the flow from scratch to make sure I had the correct vpd file. I opened both vcd and vpd files on waveviewer and they both seem to be 5915.5 ns long. I am attaching the files so you can also confirm this
[sim_out.zip](https://github.com/ucb-bar/chipyard/files/8951052/sim_out.zip)

odxa20, 2022-06-21:
I was also thinking of making a hook according to the documentation to automatically convert the vpd file to vcd

odxa20, 2022-06-21:
I was going through the log files produced and noticed this

```
** ERROR: (VOLTUS_RAIL-4063): Failed to merge the PGV libraries during rail analysis.
Use the -check_compatibility option of the check_pg_library command to check compatibility between the specified libraries before rerunning rail analysis.
The following errors occurred during the merge:
Error : The first library specified is not a TECHNOLOGY library. Please generate TECHNOLOGY library using command "set_pg_library_mode -celltype techonly". Specify the TECHNOLOGY PGV generated as the first PGV in the list. 
Ended Processing Cell Libraries for Rail Analysis: (cpu=0:00:00, real=0:00:06, mem(process/total/peak)=38.24MB/9493.58MB/10135.30MB)
** ERROR: (VOLTUS_RAIL-1050): Failed to merge the PGV libraries during rail analysis. 
Use the -check_compatibility option of the check_pg_library command to check  compatibility between the specified libraries before rerunning rail analysis.
*** Error: In processing command script
Library merging failed.Run OCI extraction flow failed.

1 error was found while processing the file /tmp/ssv_tmpdir_391591_z8nsgv/voltus_rail_391591.cmd

Rail Analysis Statistics:
Warning messages:      0
Error messages:        2

Rail Analysis is unsuccessful due to errors.

Finished Rail Analysis at 21:18:57 06/21/2022 (cpu=0:00:00, real=0:00:10, peak mem=4254.43MB)
Current Voltus IC Power Integrity Solution resource usage: (total cpu=0:29:41, real=0:04:20, mem=4115.53MB)
voltus_rail exited unsuccessfully.
**ERROR: (PRL-387):	"Rail Analysis" failed to finish successfully.
**ERROR: (VOLTUS-1055):	Rail analysis failed to finish successfully; Check voltus log file for more details.
```
After looking in the rail reports folders they are empty. Could this be due to my version of Voltus being newer or is this a known issue ?

harrisonliew, 2022-06-21:
We'll take a look at intelligently specifying the `-end` option and `vpd2vcd` hook insertion.

This looks like something went wrong in the power grid library generation prior to the power/rail analysis. Can you check if `<build_dir>/tech-asap7-cache/tech_pgv` directory exists? If not, can you look through your voltus logs and find where `set_pg_library_mode` and `write_pg_library` was executed to see if there was any error?

odxa20, 2022-06-21:
The directory does exist and has 2 sub-directories for the two different ASAP7 operating points

harrisonliew, 2022-06-21:
Huh. Something might be wrong with generating `techonly.cl`. Can you do the following:
1. Wipe the existing generated libraries (`tech-asap7-cache/tech_pgv` and `tech-asap7-cache/stdcell_pgv`)
2. Re-run the `power-par` action
3. Attach the voltus log that has the power grid library generation (calls `set_pg_library_mode` in it), not the one that executes power/rail analysis

odxa20, 2022-06-21:
I deleted the generated libraries and rerun power-par with redo-power-par target. I am attaching the log file containing `set_pg_library_mode `
[voltus2.log](https://github.com/ucb-bar/chipyard/files/8952283/voltus2.log)

harrisonliew, 2022-06-21:
I'm stumped. The library generation looks correct and successful from the log, and they are specified properly for rail analysis. This was also verified working back with Voltus 19.17.

Can you help debug this with the suggested `check_pg_library` command? I would try this command interactively:
1. Delete `exit` from the last line of `tech_stdcell_pgv.tcl`
2. Run library generation again: `voltus -no_gui -common-ui -init tech_stdcell_pgv.tcl` (same command as on the top of your attached voltus2.log)
3. After libraries are written, run `check_pg_library` with the paths to all of the generated `.cl` files (in the `tech_pgv` + `stdcell_pgv` directories). Try it with various options (look at the command reference), such as `-check_compatibility` as suggested and see what it reports.

odxa20, 2022-06-21:
I am running this command `check_pg_library "/home/odxa20/chipyard/vlsi/build/chipyard.TestHarness.TinyRocketConfig-ChipTop/tech-asap7-cache/tech_pgv/PVT_0P63V_100C/techonly.cl/ /home/odxa20/chipyard/vlsi/build/chipyard.TestHarness.TinyRocketConfig-ChipTop/tech-asap7-cache/stdcell_pgv/PVT_0P77V_0C/stdcells.cl" -check_compatibility` but am getting the following error

```
***  ***

** ERROR: (VOLTUS_LGEN-3228): The absolute path to
"/home/odxa20/chipyard/vlsi/build/chipyard.TestHarness.TinyRocketConfig-ChipTop/tech-asap7-cache/tech_pgv/PVT_0P63V_100C/techonly.cl/.cl"
could not be resolved:
System error code (errno) is 2 (ENOENT):
  No such file or directory

** ERROR: (VOLTUS_LGEN-3228): The command interpreter reported the
following error message:
"(The line number on this message indicates the location within the command
file at which the previously reported error occurred.)"

Session cannot continue due to errors while processing command file.
**ERROR: (VOLTUS-1057):	Library generation failed to finish successfully.
0
```

odxa20, 2022-06-21:
The output of `check_pg_library -check_compatibility "/home/odxa20/chipyard/vlsi/build/chipyard.TestHarness.TinyRocketConfig-ChipTop/tech-asap7-cache/tech_pgv/PVT_0P77V_0C/techonly /home/odxa20/chipyard/vlsi/build/chipyard.TestHarness.TinyRocketConfig-ChipTop/tech-asap7-cache/tech_pgv/PVT_0P63V_100C/techonly /home/odxa20/chipyard/vlsi/build/chipyard.TestHarness.TinyRocketConfig-ChipTop/tech-asap7-cache/stdcell_pgv/PVT_0P77V_0C/stdcells /home/odxa20/chipyard/vlsi/build/chipyard.TestHarness.TinyRocketConfig-ChipTop/tech-asap7-cache/stdcell_pgv/PVT_0P63V_100C/stdcells"` is

```
***  ***
Started PGV Library Generator at 23:50:22 06/21/2022

** INFO:  (VOLTUS_LGEN-3455): The libraries are compatible and therefore
can be merged or linked.
```

odxa20, 2022-06-21:
I noticed in the power-par-rundir directory there is a text file named `rail_missing_files`. Could the missing files that it mentions be causing these errors ?
[rail_missing_files.txt](https://github.com/ucb-bar/chipyard/files/8952739/rail_missing_files.txt)

harrisonliew, 2022-06-21:
Hm, very weird. Looking through your voltus.log again, here's what I see:
- static rail analysis is running into the merge failure you are seeing so far
- dynamic rail analysis (the one that requires the VCD + the ptiavg files) seems to be erroring out on missing ptiavg files before it would run into the merge failure.

For the ptiavg file case - can you look in your `activePowerReports....` directory to see if there are any ptiavg files? Perhaps the directory structure is different - I don't see any active power analysis failures in your log.

As for the library merge failure - I literally have no clue what's happening. Do you have access to an older Voltus version?

odxa20, 2022-06-21:
Indeed this is very weird :) There are ptiavg files in all `activePowerReports....` directories even in the one specified in the text file. Unfortunately we only have access to versions 21 and 20. I could give it a try tomorrow morning with version 20 as it will take some time to download

odxa20, 2022-06-25:
Hi. I'm very sorry for the delay. There were some technical issues with installing the older version on our EDA workstation which have now been resolved. I managed to run power analysis successfully on version 20.11 without any errors so there must be some compatibility problem with newer versions. The only problem is that version 20.11 of Innovus (needed to run power analysis with Voltus 20.11) is extremely slow. The place and route step with Innovous 21.12 finished in 30 minutes whereas Innovus 20.11 finishes in a whopping 6 hours and 36 minutes. During the run I noticed in the performance monitor that only one core was used out of 24. The newer version had no issues using all 24. The tcl script that runs in Innovus has the following command `set_multi_cpu_usage -local_cpu 24` which should work. Do you have any ideas why this could be happening ?

harrisonliew, 2022-06-27:
Wow, that's very odd. I have no idea why that is happening. Is there anything in the log to suggest that the `set_multi_cpu_usage` failed?

Good to know that older version of Voltus worked... but it's frustrating that these tool versions have bugs like this.

odxa20, 2022-07-08:
Again I’m very sorry for the delay. We migrated our EDA tools to a new machine and wasn’t able to run any more tests until now. There is no indication that set_multi_cpu_usage failed. We have now reverted to using the newer versions of the tools for faster design space exploration and using the older version only when rail analysis is needed. This is a temporary workaround. I hope we can fix rail analysis so that it also works with the newer tool versions. If you have any tests in mind that I should run please let me know

odxa20, 2022-07-08:
I have started two runs of the same config and will let you know as soon as they finish

harrisonliew, 2022-09-28:
Superceded by discussion in #1185

closed 2022-09-28 by harrisonliew