ucb-bar/chipyard #1192: Strange corruption in src file

mixotricha, 2022-07-06:
### Background Work

- [X] Yes, I searched the [mailing list](https://groups.google.com/forum/#!forum/chipyard)
- [X] Yes, I searched [prior issues](https://github.com/ucb-bar/chipyard/issues)
- [X] Yes, I searched the [documentation](https://chipyard.readthedocs.io/)

### Chipyard Version and Hash

Release: 1.5.0
Hash: a6a6a6

### OS Setup

Description:	Ubuntu 21.10
Release:	21.10
Codename:	impish

Description:	Ubuntu 21.10
Release:	21.10
Codename:	impish

### Other Setup

Following : https://chipyard.readthedocs.io/en/latest/Chipyard-Basics/Initial-Repo-Setup.html

- cloned fine 
- installed packages ( found I had to install universal-ctags ) ( ok ) 
- ./scripts/init-submodules-no-riscv-tools.sh ( ok ) 
- ./scripts/build-toolchains.sh riscv-tools # for a normal risc-v toolchain ( failed ) 

### Current Behavior

compiler gave up on : ./toolchains/riscv-tools/riscv-gnu-toolchain/riscv-gcc/gcc/ggc-common.c

Reason appears to be that this file was corrupt. Compiler was helpful in that it informed me that it could not make sense of :

  else if (fseek (f, mmi.offset + mmi.size, SEEK_SET) != 0)
    fatal_error (input_location, "can%'t read PCH file: %m");

Changed this to :

fatal_error (input_location, "cannot read PCH file: %m");

For all occurances in file.

After this built fine. 

### Expected Behavior

For build to successfully complete. 

### Other Information

After I fixed file build was fine.

jerryz123, 2022-07-15:
That is strange. As we pull upstream GNU tool chain we don't have control over those repos. 

Does git status in that repo return a clean status?

mixotricha, 2022-07-18:
I went with a clean ubuntu 18.04 LTS in a VM and this error went away and it started to be able to clone fine. I am guessing problems with git itself.

mixotricha, 2022-07-18:
To follow this up it seems like lots of brittleness in the build. I think a lot of this was not enough ram. Had lots of weird segfaults and instability. Seems like I toiled away at this for so long I couldn't even say what I changed aside from ... 
- Lots more ram 
- 18.04 LTS clean in VM
- Holding my tongue in the right way as still weird issues with git and clones of repo's failing 
- Something to do with the order of steps in the init and having env.sh with correct environment variables added manually before building a toolchain 
- Who the heck knows. Just a big heavy fat brutal thing to build :|

closed 2022-09-02 by jerryz123