ucb-bar/chipyard #1195: Support for Virtex VC707

mbelda, 2022-07-12:
### Background Work

- [X] Yes, I searched the [mailing list](https://groups.google.com/forum/#!forum/chipyard)
- [X] Yes, I searched the [documentation](https://chipyard.readthedocs.io/)

### Feature Description

I would need support for the Xilinx VC707 board.
I have seen that the support for the shell is already done (or started at least).
https://github.com/sifive/fpga-shells/blob/master/src/main/scala/shell/xilinx/VC707Shell.scala

So it just needs to be added to chipyard/fpga/src/main

### Motivating Example

I could use the board to deploy a 16-by-16 Gemmini that does not fit in the artys that had support (and I don't have the vcu118..)

gnodipac886, 2022-11-12:
Hi there @mbelda did you ever happen to get Gemmini to run on VC707? Cheers

mbelda, 2022-11-13:
Nop

gnodipac886, 2022-11-14:
Bump, this will be nice to have. I also want to deploy Gemmini :)

Lorilandly, 2022-11-15:
I'm trying to get rocket/hwacha working on VC707. Will report back if there's progress.

gnodipac886, 2022-11-17:
@Lorilandly @mbelda 
I found a working repo here :)
It has Gemmini support too!
https://github.com/eugene-tarassov/vivado-risc-v

Lorilandly, 2022-11-30:
I'm very exited to announce that my vc707 board has successfully booted.

PR #1278 (2022-12-15): Add support for VC707 FPGA board changelog:added

**Related PRs / Issues**: resolves #1195 

**Type of change**:
- [ ] Bug fix
- [x] New feature
- [ ] Other enhancement

**Impact**:
- [ ] RTL change
- [ ] Software change (RISC-V software)
- [ ] Build system change
- [x] Other

**Contributor Checklist**:
- [x] Did you set `main` as the base branch?
- [x] Is this PR's title suitable for inclusion in the changelog and have you added a `changelog:<topic>` label?
- [x] Did you state the type-of-change/impact?
- [x] Did you delete any extraneous prints/debugging code?
- [x] Did you mark the PR with  [... cut]
closed 2022-12-15 by abejgonzalez