ucb-bar/chipyard #1196: VCS ERROR while simulation in vlsi flow

WegdanAli13, 2022-07-12:
### Background Work

- [X] Yes, I searched the [mailing list](https://groups.google.com/forum/#!forum/chipyard)
- [X] Yes, I searched [prior issues](https://github.com/ucb-bar/chipyard/issues)
- [X] Yes, I searched the [documentation](https://chipyard.readthedocs.io/)

### Chipyard Version and Hash

Release: 1.7.1

### OS Setup

centos-release-8

### Other Setup

vcs version "S-2021.09-SP1"

### Current Behavior

Error-[DPI-UED] C++ Exception detected
  Import DPI routine invoked at file
  '/home/mohammwn/chipyard/vlsi/generated-src/chipyard.TestHarness.TinyRocketConfig/SimSerial.v'(line
  53) has C++ exceptions not caught. C++ exceptions shall not propagate out of
  any imported subroutine.
  Fix it in DPI-C code before running simulation.

This error happens when trying the vlsi flow as I want to simulate my design. I use the command "make sim-rtl CONFIG=TinyRocketConfig BINARY=$RISCV/riscv64-unknown-elf/share/riscv-tests/isa/rv32ui-p-simple" as in the ASAP7 example . 

### Expected Behavior

Simulation executables based on vcs run correctly with no errors thrown.

harrisonliew, 2022-07-12:
Can you paste the commands invoked by the make target (i.e. both the VCS compilation and execution)? I wonder if #1081 might be related.

WegdanAli13, 2022-07-14:
"make sim-rtl CONFIG=TinyRocketConfig BINARY=$RISCV/riscv64-unknown-elf/share/riscv-tests/isa/rv32ui-p-simple" 
This ius the command that I used, in vlsi flow I choose that after synthesizing I want to run simulation. Then in my path there is VCS and it calls it with this command .

harrisonliew, 2022-07-14:
Sorry what I mean is, can you paste the VCS commands generated by Hammer? They should be:
`vcs <a bunch of options>`
`simv-<config> <a bunch of options>`

I'm wondering if there are some options passed to the compilation or simulator that are not specified correctly.

harrisonliew, 2022-07-18:
Sure, you can attach the generated `sim-inputs.yml`. However, I want to see the actual VCS commands that are being run. You will need to copy these from your terminal output or the `hammer-vlsi-<timestamp>.log` file.

WegdanAli13, 2022-07-21:
[hammer-vlsi-20220721-110048.log](https://github.com/ucb-bar/chipyard/files/9157919/hammer-vlsi-20220721-110048.log)
 sorry for the late reply, the log file is attached as there are so many lines

harrisonliew, 2022-09-28:
Not sure how I missed replying to this, but I see that you have a VCS and VERDI version mismatch in your log file. Also, upcoming PR #1226 may also help fix this.

harrisonliew, 2022-09-29:
#1226