ucb-bar/chipyard #1199: [error] java.lang.Exception: Unable to find part "chipyard.HeteroConfig" from "ArraySeq(chipyard.HeteroConfig)"

nicolast0604, 2022-07-15:
### Background Work

- [X] Yes, I searched the [mailing list](https://groups.google.com/forum/#!forum/chipyard)
- [X] Yes, I searched [prior issues](https://github.com/ucb-bar/chipyard/issues)
- [X] Yes, I searched the [documentation](https://chipyard.readthedocs.io/)

### Chipyard Version and Hash

Release: 1.5.0
Hash: a6a6a6

### OS Setup

Ex: Output of `uname -a` and `lsb_release -a`
Linux nicolast0604-VirtualBox 5.4.0-122-generic #138~18.04.1-Ubuntu SMP Fri Jun 24 14:14:03 UTC 2022 x86_64 x86_64 x86_64 GNU/Linux

No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 18.04.6 LTS
Release:	18.04
Codename:	bionic

### Other Setup

Ex: Prior steps taken / Documentation Followed / etc...

### Current Behavior

make CONFIG=HeteroConfig
Running with RISCV=/home/nicolast0604/chipyard/esp-tools-install
mkdir -p /home/nicolast0604/chipyard/sims/verilator/generated-src/chipyard.TestHarness.HeteroConfig
cd /home/nicolast0604/chipyard && java -jar /home/nicolast0604/chipyard/generators/rocket-chip/sbt-launch.jar   ";project chipyard; runMain chipyard.Generator  --target-dir /home/nicolast0604/chipyard/sims/verilator/generated-src/chipyard.TestHarness.HeteroConfig --name chipyard.TestHarness.HeteroConfig --top-module chipyard.TestHarness --legacy-configs chipyard:HeteroConfig "
Picked up JAVA_TOOL_OPTIONS: -Xmx8G -Xss8M -Djava.io.tmpdir=/home/nicolast0604/chipyard/.java_tmp
[info] welcome to sbt 1.5.5 (Private Build Java 11.0.15)
[info] loading settings for project chipyard-build from plugins.sbt ...
[info] loading project definition from /home/nicolast0604/chipyard/project
[info] loading settings for project chipyardRoot from build.sbt ...
[info] loading settings for project hardfloat from build.sbt ...
[info] loading settings for project rocketConfig from build.sbt ...
[info] loading settings for project testchipip from build.sbt ...
[info] loading settings for project icenet from build.sbt ...
[info] loading settings for project hwacha from build.sbt ...
[info] loading settings for project boom from build.sbt ...
[info] loading settings for project cva6 from build.sbt ...
[info] loading settings for project ibex from build.sbt ...
[info] loading settings for project sodor from build.sbt ...
[info] loading settings for project gemmini from build.sbt ...
[info] loading settings for project tapeout from build.sbt ...
[info] loading settings for project sim-build from plugins.sbt ...
[info] loading project definition from /home/nicolast0604/chipyard/sims/firesim/sim/project
[info] loading settings for project firesim from build.sbt ...
[info] loading settings for project targetutils from build.sbt ...
[info] loading settings for project midas from build.sbt ...
[info] resolving key references (32290 settings) ...
[info] set current project to chipyardRoot (in build file:/home/nicolast0604/chipyard/)
[info] set current project to chipyard (in build file:/home/nicolast0604/chipyard/)
[info] running chipyard.Generator --target-dir /home/nicolast0604/chipyard/sims/verilator/generated-src/chipyard.TestHarness.HeteroConfig --name chipyard.TestHarness.HeteroConfig --top-module chipyard.TestHarness --legacy-configs chipyard:HeteroConfig
[error] (run-main-0) java.lang.Exception: Unable to find part "chipyard.HeteroConfig" from "ArraySeq(chipyard.HeteroConfig)", did you misspell it or specify the wrong package path?
[error] java.lang.Exception: Unable to find part "chipyard.HeteroConfig" from "ArraySeq(chipyard.HeteroConfig)", did you misspell it or specify the wrong package path?
[error] 	at Chisel.package$throwException$.apply(compatibility.scala:451)
[error] 	at freechips.rocketchip.util.HasRocketChipStageUtils.$anonfun$getConfig$1(GeneratorUtils.scala:19)
[error] 	at scala.collection.IndexedSeqOptimized.foldRight(IndexedSeqOptimized.scala:65)
[error] 	at scala.collection.IndexedSeqOptimized.foldRight$(IndexedSeqOptimized.scala:72)
[error] 	at scala.collection.mutable.ArraySeq.foldRight(ArraySeq.scala:47)
[error] 	at freechips.rocketchip.util.HasRocketChipStageUtils.getConfig(GeneratorUtils.scala:14)
[error] 	at freechips.rocketchip.util.HasRocketChipStageUtils.getConfig$(GeneratorUtils.scala:13)
[error] 	at freechips.rocketchip.stage.phases.PreElaboration.getConfig(PreElaboration.scala:18)
[error] 	at freechips.rocketchip.stage.phases.PreElaboration.transform(PreElaboration.scala:29)
[error] 	at freechips.rocketchip.stage.phases.PreElaboration.transform(PreElaboration.scala:18)
[error] 	at firrtl.options.phases.DeletedWrapper.internalTransform(DeletedWrapper.scala:38)
[error] 	at firrtl.options.phases.DeletedWrapper.internalTransform(DeletedWrapper.scala:15)
[error] 	at firrtl.options.Translator.transform(Phase.scala:248)
[error] 	at firrtl.options.Translator.transform$(Phase.scala:248)
[error] 	at firrtl.options.phases.DeletedWrapper.transform(DeletedWrapper.scala:15)
[error] 	at firrtl.options.DependencyManager.$anonfun$transform$5(DependencyManager.scala:280)
[error] 	at firrtl.Utils$.time(Utils.scala:181)
[error] 	at firrtl.options.DependencyManager.$anonfun$transform$3(DependencyManager.scala:280)
[error] 	at scala.collection.LinearSeqOptimized.foldLeft(LinearSeqOptimized.scala:126)
[error] 	at scala.collection.LinearSeqOptimized.foldLeft$(LinearSeqOptimized.scala:122)
[error] 	at scala.collection.immutable.List.foldLeft(List.scala:89)
[error] 	at firrtl.options.DependencyManager.transform(DependencyManager.scala:269)
[error] 	at firrtl.options.DependencyManager.transform$(DependencyManager.scala:255)
[error] 	at firrtl.options.PhaseManager.transform(DependencyManager.scala:443)
[error] 	at chisel3.stage.ChiselStage.run(ChiselStage.scala:45)
[error] 	at firrtl.options.Stage$$anon$1.transform(Stage.scala:43)
[error] 	at firrtl.options.Stage$$anon$1.transform(Stage.scala:43)
[error] 	at firrtl.options.phases.DeletedWrapper.internalTransform(DeletedWrapper.scala:38)
[error] 	at firrtl.options.phases.DeletedWrapper.internalTransform(DeletedWrapper.scala:15)
[error] 	at firrtl.options.Translator.transform(Phase.scala:248)
[error] 	at firrtl.options.Translator.transform$(Phase.scala:248)
[error] 	at firrtl.options.phases.DeletedWrapper.transform(DeletedWrapper.scala:15)
[error] 	at firrtl.options.Stage.$anonfun$transform$5(Stage.scala:47)
[error] 	at scala.collection.LinearSeqOptimized.foldLeft(LinearSeqOptimized.scala:126)
[error] 	at scala.collection.LinearSeqOptimized.foldLeft$(LinearSeqOptimized.scala:122)
[error] 	at scala.collection.immutable.List.foldLeft(List.scala:89)
[error] 	at firrtl.options.Stage.$anonfun$transform$3(Stage.scala:47)
[error] 	at logger.Logger$.$anonfun$makeScope$2(Logger.scala:137)
[error] 	at scala.util.DynamicVariable.withValue(DynamicVariable.scala:62)
[error] 	at logger.Logger$.makeScope(Logger.scala:135)
[error] 	at firrtl.options.Stage.transform(Stage.scala:47)
[error] 	at firrtl.options.Stage.execute(Stage.scala:58)
[error] 	at firrtl.options.StageMain.main(Stage.scala:71)
[error] 	at chipyard.Generator.main(Generator.scala)
[error] 	at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
[error] 	at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
[error] 	at java.base/jdk.internal.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
[error] 	at java.base/java.lang.reflect.Method.invoke(Method.java:566)
[error] Caused by: java.lang.ClassNotFoundException: chipyard.HeteroConfig
[error] 	at java.base/java.net.URLClassLoader.findClass(URLClassLoader.java:476)
[error] stack trace is suppressed; run last Compile / bgRunMain for the full output
[error] Nonzero exit code: 1
[error] (Compile / runMain) Nonzero exit code: 1
[error] Total time: 18 s, completed Jul 15, 2022, 5:33:26 PM
/home/nicolast0604/chipyard/common.mk:110: recipe for target 'generator_temp' failed
make: *** [generator_temp] Error 1

### Expected Behavior

without errors and generate the simulation

michael-etzkorn, 2022-07-15:
There is no config `HeteroConfig`. Perhaps you meant one of the configs here? https://github.com/ucb-bar/chipyard/blob/main/generators/chipyard/src/main/scala/config/HeteroConfigs.scala

nicolast0604, 2022-07-16:
Yes, I tried many Configs.scala, only RocketConfigs.scala can work. others have the same errors.

michael-etzkorn, 2022-07-16:
> Yes, I tried many Configs.scala, only RocketConfigs.scala can work. others have the same errors. 

You'll need to show us what you're running. The one example you showed is not a valid CONFIG value.

nicolast0604, 2022-07-17:
Hi Michael
I found the root casuse, the config name is wrong in my simulation. closed this.

closed 2022-07-17 by nicolast0604