ucb-bar/chipyard #1200: Invalid build URI (no handler available) /chipyard/sims/firesim/sim/

mbelda, 2022-07-15:
### Background Work

- [X] Yes, I searched the [mailing list](https://groups.google.com/forum/#!forum/chipyard)
- [X] Yes, I searched [prior issues](https://github.com/ucb-bar/chipyard/issues)
- [X] Yes, I searched the [documentation](https://chipyard.readthedocs.io/)

### Chipyard Version and Hash

Branch: arty100t
commit 7a3f7aef34c9045b49cd3127c5b197493e80d944 (HEAD -> arty100t, origin/arty100t)

### OS Setup

Ubuntu 20.04.4

### Other Setup

I followed the chipyard documentation

### Current Behavior

When I try to execute
~~~
make SUB_PROJECT=arty100t bitstream
~~~

on the /chipyard/fpga directory I got the following error.

~~~
[debug]     Load.loadUnit: loadedProjectsRaw took 6511.282458ms
[debug]     Load.loadUnit: cleanEvalClasses took 4.367459ms
[debug]   Load.loadUnit(file:/home/majo/chipyard-arty/chipyard/, ...) took 12361.779661ms
[error] java.lang.RuntimeException: Invalid build URI (no handler available): file:/home/majo/chipyard-arty/chipyard/sims/firesim/sim/
[error] 	at scala.sys.package$.error(package.scala:30)
[error] 	at sbt.internal.Load$.$anonfun$builtinLoader$1(Load.scala:483)
[error] 	at sbt.internal.BuildLoader.$anonfun$apply$3(BuildLoader.scala:244)
[error] 	at scala.Option.getOrElse(Option.scala:189)
[error] 	at sbt.internal.BuildLoader.apply(BuildLoader.scala:244)
[error] 	at sbt.internal.Load$.loadURI$1(Load.scala:548)
[error] 	at sbt.internal.Load$.loadAll(Load.scala:564)
[error] 	at sbt.internal.Load$.loadURI(Load.scala:494)
[error] 	at sbt.internal.Load$.load(Load.scala:473)
[error] 	at sbt.internal.Load$.$anonfun$apply$1(Load.scala:242)
[error] 	at sbt.internal.Load$.timed(Load.scala:1382)
[error] 	at sbt.internal.Load$.apply(Load.scala:242)
[error] 	at sbt.internal.Load$.defaultLoad(Load.scala:55)
[error] 	at sbt.BuiltinCommands$.liftedTree1$1(Main.scala:934)
[error] 	at sbt.BuiltinCommands$.doLoadProject(Main.scala:934)
[error] 	at sbt.BuiltinCommands$.$anonfun$loadProjectImpl$2(Main.scala:887)
[error] 	at sbt.Command$.$anonfun$applyEffect$4(Command.scala:150)
[error] 	at sbt.Command$.$anonfun$applyEffect$2(Command.scala:145)
[error] 	at sbt.Command$.process(Command.scala:189)
[error] 	at sbt.MainLoop$.$anonfun$processCommand$5(MainLoop.scala:243)
[error] 	at scala.Option.getOrElse(Option.scala:189)
[error] 	at sbt.MainLoop$.process$1(MainLoop.scala:243)
[error] 	at sbt.MainLoop$.processCommand(MainLoop.scala:276)
[error] 	at sbt.MainLoop$.$anonfun$next$5(MainLoop.scala:163)
[error] 	at sbt.State$StateOpsImpl$.runCmd$1(State.scala:289)
[error] 	at sbt.State$StateOpsImpl$.process$extension(State.scala:325)
[error] 	at sbt.MainLoop$.$anonfun$next$4(MainLoop.scala:163)
[error] 	at sbt.internal.util.ErrorHandling$.wideConvert(ErrorHandling.scala:23)
[error] 	at sbt.MainLoop$.next(MainLoop.scala:163)
[error] 	at sbt.MainLoop$.run(MainLoop.scala:144)
[error] 	at sbt.MainLoop$.$anonfun$runWithNewLog$1(MainLoop.scala:119)
[error] 	at sbt.io.Using.apply(Using.scala:27)
[error] 	at sbt.MainLoop$.runWithNewLog(MainLoop.scala:112)
[error] 	at sbt.MainLoop$.runAndClearLast(MainLoop.scala:66)
[error] 	at sbt.MainLoop$.runLoggedLoop(MainLoop.scala:51)
[error] 	at sbt.MainLoop$.runLogged(MainLoop.scala:42)
[error] 	at sbt.StandardMain$.runManaged(Main.scala:193)
[error] 	at sbt.xMain$.$anonfun$run$8(Main.scala:102)
[error] 	at scala.util.DynamicVariable.withValue(DynamicVariable.scala:62)
[error] 	at scala.Console$.withIn(Console.scala:230)
[error] 	at sbt.internal.util.Terminal$.withIn(Terminal.scala:560)
[error] 	at sbt.internal.util.Terminal$.$anonfun$withStreams$1(Terminal.scala:350)
[error] 	at scala.util.DynamicVariable.withValue(DynamicVariable.scala:62)
[error] 	at scala.Console$.withOut(Console.scala:167)
[error] 	at sbt.internal.util.Terminal$.$anonfun$withOut$2(Terminal.scala:550)
[error] 	at scala.util.DynamicVariable.withValue(DynamicVariable.scala:62)
[error] 	at scala.Console$.withErr(Console.scala:196)
[error] 	at sbt.internal.util.Terminal$.withOut(Terminal.scala:550)
[error] 	at sbt.internal.util.Terminal$.withStreams(Terminal.scala:350)
[error] 	at sbt.xMain$.run(Main.scala:86)
[error] 	at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
[error] 	at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
[error] 	at java.base/jdk.internal.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
[error] 	at java.base/java.lang.reflect.Method.invoke(Method.java:566)
[error] 	at sbt.internal.XMainConfiguration.run(XMainConfiguration.scala:83)
[error] 	at sbt.xMain.run(Main.scala:46)
[error] 	at xsbt.boot.Launch$$anonfun$run$1.apply(Launch.scala:109)
[error] 	at xsbt.boot.Launch$.withContextLoader(Launch.scala:128)
[error] 	at xsbt.boot.Launch$.run(Launch.scala:109)
[error] 	at xsbt.boot.Launch$$anonfun$apply$1.apply(Launch.scala:35)
[error] 	at xsbt.boot.Launch$.launch(Launch.scala:117)
[error] 	at xsbt.boot.Launch$.apply(Launch.scala:18)
[error] 	at xsbt.boot.Boot$.runImpl(Boot.scala:41)
[error] 	at xsbt.boot.Boot$.main(Boot.scala:17)
[error] 	at xsbt.boot.Boot.main(Boot.scala)
[error] Invalid build URI (no handler available): file:/home/majo/chipyard-arty/chipyard/sims/firesim/sim/
[error] Use 'last' for the full log.
~~~

It doesn't have have any sense because it is a Firesim issue and I am not using FireSim at all.
I want to deploy the bitstream on my own Arty100T FPGA.

### Expected Behavior

I want the bitstream to be generated correctly.

### Other Information

I tried it first with the esp-tools toolchain and now I am using the riscv-tools just in case this branch doesn't have support for Hwacha (I am not using it now, but I might need it) but I still have the error.

mbelda, 2022-07-15:
I just re-run the dependencies at the beginning of Chipyard docs and everything is installed exept for one package that gives an error.

When running `sudo apt-get update` I get
~~~
Err:10 https://dl.bintray.com/sbt/debian  Release                                                                                                      
  Could not handshake: The TLS connection was non-properly terminated.
...
E: The repository 'https://dl.bintray.com/sbt/debian  Release' no longer has a Release file.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
~~~

michael-etzkorn, 2022-07-15:
Scala Build Tool is needed to compile the Scala code (that Chisel lives in) 

If you're running https://github.com/ucb-bar/chipyard/blob/main/scripts/ubuntu-req.sh, I'm not sure you would still see this issue. Are you running from within a VM? That shouldn't matter but noticed the TLS handshake error. 
```
echo "deb https://repo.scala-sbt.org/scalasbt/debian all main" | sudo tee /etc/apt/sources.list.d/sbt.list
echo "deb https://repo.scala-sbt.org/scalasbt/debian /" | sudo tee /etc/apt/sources.list.d/sbt_old.list
curl -sL "https://keyserver.ubuntu.com/pks/lookup?op=get&search=0x2EE0EA64E40A89B84B2DF73499E82A75642AC823" | sudo apt-key add
sudo apt-get update
sudo apt-get install sbt
```

https://www.scala-sbt.org/download.html

But as long as you have any recent SBT version ~1.5 that should be fine. It may be that the arty branch is out of date with main to the point where firesim won't compile properly. 

Did you run `./scripts/init-submodules-no-riscv-tools.sh` from within the arty branch?

michael-etzkorn, 2022-07-15:
I believe targetting arty should work on 1.7.1. IIRC the arty100T branch was adding simulation support for the Arty test harness.

mbelda, 2022-07-18:
The sbt was installed properly with the commands you provide.
Also, I rerun the dependencies setup but changing the config of git from git:// to https:// and everything was cloned fine.
I encountered another error on my way to generate the bitstream but it is not related to this, so I am closing this issue and opening a new one.
Thanks!

closed 2022-07-18 by mbelda