ucb-bar/chipyard #1202: Errors make CONFIG=LargeNVDLARocketConfig

nicolast0604, 2022-07-17:
### Background Work

- [X] Yes, I searched the [mailing list](https://groups.google.com/forum/#!forum/chipyard)
- [X] Yes, I searched [prior issues](https://github.com/ucb-bar/chipyard/issues)
- [X] Yes, I searched the [documentation](https://chipyard.readthedocs.io/)

### Chipyard Version and Hash

Release: 1.5.0
Hash: a6a6a6

### OS Setup

Ex: Output of `uname -a` and `lsb_release -a`
Linux nicolast0604-VirtualBox 5.4.0-122-generic #138~18.04.1-Ubuntu SMP Fri Jun 24 14:14:03 UTC 2022 x86_64 x86_64 x86_64 GNU/Linux

No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 18.04.6 LTS
Release:	18.04
Codename:	bionic

### Other Setup

Ex: Prior steps taken / Documentation Followed / etc...

### Current Behavior

%Error-TIMESCALEMOD: /home/nicolast0604/chipyard/sims/verilator/generated-src/chipyard.TestHarness.LargeNVDLARocketConfig/nvdla_large.preprocessed.v:326937:67: Timescale missing on this module as other modules have it (IEEE 1800-2017 3.14.2.2)
                     /home/nicolast0604/chipyard/sims/verilator/generated-src/chipyard.TestHarness.LargeNVDLARocketConfig/ClockDividerN.sv:8:8: ... Location of module with timescale
    8 | module ClockDividerN #(parameter DIV = 1)(output logic clk_out = 1'b0, input clk_in);
      |        ^~~~~~~~~~~~~
%Error-TIMESCALEMOD: /home/nicolast0604/chipyard/sims/verilator/generated-src/chipyard.TestHarness.LargeNVDLARocketConfig/nvdla_large.preprocessed.v:327591:69: Timescale missing on this module as other modules have it (IEEE 1800-2017 3.14.2.2)
                     /home/nicolast0604/chipyard/sims/verilator/generated-src/chipyard.TestHarness.LargeNVDLARocketConfig/ClockDividerN.sv:8:8: ... Location of module with timescale
    8 | module ClockDividerN #(parameter DIV = 1)(output logic clk_out = 1'b0, input clk_in);
      |        ^~~~~~~~~~~~~
%Error-TIMESCALEMOD: /home/nicolast0604/chipyard/sims/verilator/generated-src/chipyard.TestHarness.LargeNVDLARocketConfig/nvdla_large.preprocessed.v:328110:68: Timescale missing on this module as other modules have it (IEEE 1800-2017 3.14.2.2)
                     /home/nicolast0604/chipyard/sims/verilator/generated-src/chipyard.TestHarness.LargeNVDLARocketConfig/ClockDividerN.sv:8:8: ... Location of module with timescale
    8 | module ClockDividerN #(parameter DIV = 1)(output logic clk_out = 1'b0, input clk_in);
      |        ^~~~~~~~~~~~~
%Error-TIMESCALEMOD: /home/nicolast0604/chipyard/sims/verilator/generated-src/chipyard.TestHarness.LargeNVDLARocketConfig/nvdla_large.preprocessed.v:328606:68: Timescale missing on this module as other modules have it (IEEE 1800-2017 3.14.2.2)
                     /home/nicolast0604/chipyard/sims/verilator/generated-src/chipyard.TestHarness.LargeNVDLARocketConfig/ClockDividerN.sv:8:8: ... Location of module with timescale
    8 | module ClockDividerN #(parameter DIV = 1)(output logic clk_out = 1'b0, input clk_in);
      |        ^~~~~~~~~~~~~
%Error-TIMESCALEMOD: /home/nicolast0604/chipyard/sims/verilator/generated-src/chipyard.TestHarness.LargeNVDLARocketConfig/nvdla_large.preprocessed.v:329063:68: Timescale missing on this module as other modules have it (IEEE 1800-2017 3.14.2.2)
                     /home/nicolast0604/chipyard/sims/verilator/generated-src/chipyard.TestHarness.LargeNVDLARocketConfig/ClockDividerN.sv:8:8: ... Location of module with timescale
    8 | module ClockDividerN #(parameter DIV = 1)(output logic clk_out = 1'b0, input clk_in);
      |        ^~~~~~~~~~~~~
%Error-TIMESCALEMOD: /home/nicolast0604/chipyard/sims/verilator/generated-src/chipyard.TestHarness.LargeNVDLARocketConfig/nvdla_large.preprocessed.v:329574:68: Timescale missing on this module as other modules have it (IEEE 1800-2017 3.14.2.2)
                     /home/nicolast0604/chipyard/sims/verilator/generated-src/chipyard.TestHarness.LargeNVDLARocketConfig/ClockDividerN.sv:8:8: ... Location of module with timescale
    8 | module ClockDividerN #(parameter DIV = 1)(output logic clk_out = 1'b0, input clk_in);
      |        ^~~~~~~~~~~~~
%Error-TIMESCALEMOD: /home/nicolast0604/chipyard/sims/verilator/generated-src/chipyard.TestHarness.LargeNVDLARocketConfig/nvdla_large.preprocessed.v:330180:67: Timescale missing on this module as other modules have it (IEEE 1800-2017 3.14.2.2)
                     /home/nicolast0604/chipyard/sims/verilator/generated-src/chipyard.TestHarness.LargeNVDLARocketConfig/ClockDividerN.sv:8:8: ... Location of module with timescale
    8 | module ClockDividerN #(parameter DIV = 1)(output logic clk_out = 1'b0, input clk_in);
      |        ^~~~~~~~~~~~~
%Error-TIMESCALEMOD: /home/nicolast0604/chipyard/sims/verilator/generated-src/chipyard.TestHarness.LargeNVDLARocketConfig/nvdla_large.preprocessed.v:330758:68: Timescale missing on this module as other modules have it (IEEE 1800-2017 3.14.2.2)
                     /home/nicolast0604/chipyard/sims/verilator/generated-src/chipyard.TestHarness.LargeNVDLARocketConfig/ClockDividerN.sv:8:8: ... Location of module with timescale
    8 | module ClockDividerN #(parameter DIV = 1)(output logic clk_out = 1'b0, input clk_in);
      |        ^~~~~~~~~~~~~
%Error-TIMESCALEMOD: /home/nicolast0604/chipyard/sims/verilator/generated-src/chipyard.TestHarness.LargeNVDLARocketConfig/nvdla_large.preprocessed.v:331283:69: Timescale missing on this module as other modules have it (IEEE 1800-2017 3.14.2.2)
                     /home/nicolast0604/chipyard/sims/verilator/generated-src/chipyard.TestHarness.LargeNVDLARocketConfig/ClockDividerN.sv:8:8: ... Location of module with timescale
    8 | module ClockDividerN #(parameter DIV = 1)(output logic clk_out = 1'b0, input clk_in);
      |        ^~~~~~~~~~~~~
%Error-TIMESCALEMOD: /home/nicolast0604/chipyard/sims/verilator/generated-src/chipyard.TestHarness.LargeNVDLARocketConfig/nvdla_large.preprocessed.v:331814:70: Timescale missing on this module as other modules have it (IEEE 1800-2017 3.14.2.2)
                     /home/nicolast0604/chipyard/sims/verilator/generated-src/chipyard.TestHarness.LargeNVDLARocketConfig/ClockDividerN.sv:8:8: ... Location of module with timescale
    8 | module ClockDividerN #(parameter DIV = 1)(output logic clk_out = 1'b0, input clk_in);
      |        ^~~~~~~~~~~~~
%Error-TIMESCALEMOD: /home/nicolast0604/chipyard/sims/verilator/generated-src/chipyard.TestHarness.LargeNVDLARocketConfig/nvdla_large.preprocessed.v:332380:70: Timescale missing on this module as other modules have it (IEEE 1800-2017 3.14.2.2)
                     /home/nicolast0604/chipyard/sims/verilator/generated-src/chipyard.TestHarness.LargeNVDLARocketConfig/ClockDividerN.sv:8:8: ... Location of module with timescale
[... 84 lines cut ...]
    8 | module ClockDividerN #(parameter DIV = 1)(output logic clk_out = 1'b0, input clk_in);
      |        ^~~~~~~~~~~~~
%Error-TIMESCALEMOD: /home/nicolast0604/chipyard/sims/verilator/generated-src/chipyard.TestHarness.LargeNVDLARocketConfig/nvdla_large.preprocessed.v:359285:70: Timescale missing on this module as other modules have it (IEEE 1800-2017 3.14.2.2)
                     /home/nicolast0604/chipyard/sims/verilator/generated-src/chipyard.TestHarness.LargeNVDLARocketConfig/ClockDividerN.sv:8:8: ... Location of module with timescale
    8 | module ClockDividerN #(parameter DIV = 1)(output logic clk_out = 1'b0, input clk_in);
      |        ^~~~~~~~~~~~~
%Error-TIMESCALEMOD: /home/nicolast0604/chipyard/sims/verilator/generated-src/chipyard.TestHarness.LargeNVDLARocketConfig/nvdla_large.preprocessed.v:359885:70: Timescale missing on this module as other modules have it (IEEE 1800-2017 3.14.2.2)
                     /home/nicolast0604/chipyard/sims/verilator/generated-src/chipyard.TestHarness.LargeNVDLARocketConfig/ClockDividerN.sv:8:8: ... Location of module with timescale
    8 | module ClockDividerN #(parameter DIV = 1)(output logic clk_out = 1'b0, input clk_in);
      |        ^~~~~~~~~~~~~
%Error-TIMESCALEMOD: /home/nicolast0604/chipyard/sims/verilator/generated-src/chipyard.TestHarness.LargeNVDLARocketConfig/nvdla_large.preprocessed.v:360564:70: Timescale missing on this module as other modules have it (IEEE 1800-2017 3.14.2.2)
                     /home/nicolast0604/chipyard/sims/verilator/generated-src/chipyard.TestHarness.LargeNVDLARocketConfig/ClockDividerN.sv:8:8: ... Location of module with timescale
    8 | module ClockDividerN #(parameter DIV = 1)(output logic clk_out = 1'b0, input clk_in);
      |        ^~~~~~~~~~~~~
[... cut ...]
    8 | module ClockDividerN #(parameter DIV = 1)(output logic clk_out = 1'b0, input clk_in);
      |        ^~~~~~~~~~~~~
%Error: Exiting due to 36 error(s)
Makefile:214: recipe for target '/home/nicolast0604/chipyard/sims/verilator/generated-src/chipyard.TestHarness.LargeNVDLARocketConfig/chipyard.TestHarness.LargeNVDLARocketConfig/VTestHarness.mk' failed
make: *** [/home/nicolast0604/chipyard/sims/verilator/generated-src/chipyard.TestHarness.LargeNVDLARocketConfig/chipyard.TestHarness.LargeNVDLARocketConfig/VTestHarness.mk] Error 1

### Expected Behavior

successfully make.

jerryz123, 2022-07-17:
What verilator version do you have. 4.034 is the minimum.

nicolast0604, 2022-07-17:
Verilator 4.104 2020-11-14 rev v4.104

nicolast0604, 2022-07-17:
I found that the folder is empty.
~/chipyard/sims/verilator/generated-src/chipyard.TestHarness.LargeNVDLARocketConfig/chipyard.TestHarness.LargeNVDLARocketConfig/

michael-etzkorn, 2022-07-17:
You can try putting 
```
`timescale 1ns/100fs
```

at the top of `nvdla_large.preprocessed.v`. Looks like it could be a bug with that blackbox not having a timescale.

nicolast0604, 2022-07-18:
I added it from your suggestion but unfortunately still fail with the same erros messages.