ucb-bar/chipyard #1203: The board_part definition was not found for digilentinc.com:arty-a7-100

mbelda, 2022-07-18:
### Background Work

- [X] Yes, I searched the [mailing list](https://groups.google.com/forum/#!forum/chipyard)
- [X] Yes, I searched [prior issues](https://github.com/ucb-bar/chipyard/issues)
- [X] Yes, I searched the [documentation](https://chipyard.readthedocs.io/)

### Chipyard Version and Hash

Branch: arty100t
commit 7a3f7aef34c9045b49cd3127c5b197493e80d944 (HEAD -> arty100t, origin/arty100t)

### OS Setup

Ubuntu 20.04.4

### Other Setup

I followed the chipyard documentation to setup the repo

### Current Behavior

When I try to execute `make SUB_PROJECT=arty100t bitstream` the following error is shown and the make exits.

~~~
****** Vivado v2022.1.1 (64-bit)
  **** SW Build 3557992 on Fri Jun  3 09:56:20 MDT 2022
  **** IP Build 3556342 on Fri Jun  3 19:31:36 MDT 2022
    ** Copyright 1986-2022 Xilinx, Inc. All Rights Reserved.

source /home/majo/chipyard-arty/chipyard/fpga/fpga-shells/xilinx/common/tcl/vivado.tcl
# set scriptdir [file dirname [info script]]
# source [file join $scriptdir "prologue.tcl"]
## set ip_vivado_tcls {}
## while {[llength $argv]} {
##   set argv [lassign $argv[set argv {}] flag]
##   switch -glob $flag {
##     -top-module {
##       set argv [lassign $argv[set argv {}] top]
##     }
##     -F {
##       # This should be a simple file format with one filepath per line
##       set argv [lassign $argv[set argv {}] vsrc_manifest]
##     }
##     -board {
##       set argv [lassign $argv[set argv {}] board]
##     }
##     -ip-vivado-tcls {
##       set argv [lassign $argv[set argv {}] ip_vivado_tcls]
##     }
##     -pre-impl-debug-tcl {
##       set argv [lassign $argv[set argv {}] pre_impl_debug_tcl]
##     }
##     -post-impl-debug-tcl {
##       set argv [lassign $argv[set argv {}] post_impl_debug_tcl]
##     }
##     -env-var-srcs {
##       set argv [lassign $argv[set argv {}] env_var_srcs]
##     }
##     default {
##       return -code error [list {unknown option} $flag]
##     }
##   }
## }
## if {![info exists top]} {
##   return -code error [list {--top-module option is required}]
## }
## if {![info exists vsrc_manifest]} {
##   return -code error [list {-F option is required}]
## }
## if {![info exists board]} {
##   return -code error [list {--board option is required}]
## }
## set commondir [file dirname $scriptdir]
## set boarddir [file join [file dirname $commondir] $board]
## source [file join $boarddir tcl board.tcl]
### set name {arty-a7-100}
### set part_fpga {xc7a100ticsg324-1L}
### set part_board {digilentinc.com:arty-a7-100:part0:1.0}
### set bootrom_inst {rom}
## set constraintsdir [file join $boarddir constraints]
## set srcdir [file join $commondir vsrc]
## set wrkdir [file join [pwd] obj]
## set ipdir [file join $wrkdir ip]
## create_project -part $part_fpga -force $top
## set_param messaging.defaultLimit 1000000
## set_property -dict [list \
## 	BOARD_PART $part_board \
## 	TARGET_LANGUAGE {Verilog} \
## 	DEFAULT_LIB {xil_defaultlib} \
## 	IP_REPO_PATHS $ipdir \
## 	] [current_project]
ERROR: [Board 49-71] The board_part definition was not found for digilentinc.com:arty-a7-100:part0:1.0. The project's board_part property was not set, but the project's part property was set to xc7a100ticsg324-1L. Valid board_part values can be retrieved with the 'get_board_parts' Tcl command. Check if board.repoPaths parameter is set and the board_part is installed from the tcl app store.
INFO: [Common 17-17] undo 'set_property'

    while executing
"rdi::add_properties -dict {BOARD_PART digilentinc.com:arty-a7-100:part0:1.0 TARGET_LANGUAGE Verilog DEFAULT_LIB xil_defaultlib IP_REPO_PATHS /home/maj..."
    invoked from within
"set_property -dict [list \
	BOARD_PART $part_board \
	TARGET_LANGUAGE {Verilog} \
	DEFAULT_LIB {xil_defaultlib} \
	IP_REPO_PATHS $ipdir \
	] [current_..."
    (file "/home/majo/chipyard-arty/chipyard/fpga/fpga-shells/xilinx/common/tcl/prologue.tcl" line 78)

    while executing
"source [file join $scriptdir "prologue.tcl"]"
    (file "/home/majo/chipyard-arty/chipyard/fpga/fpga-shells/xilinx/common/tcl/vivado.tcl" line 7)
INFO: [Common 17-206] Exiting Vivado at Mon Jul 18 10:53:44 2022...
make: *** [Makefile:162: /home/majo/chipyard-arty/chipyard/fpga/generated-src/chipyard.fpga.arty100t.Arty100TFPGATestHarness.RocketArty100TConfig/obj/Arty100TFPGATestHarness.bit] Error 1
~~~

### Expected Behavior

I want the bitstream to be generated correctly.

Yuxin-Yu, 2022-07-18:
hi，I got the same error ,have you know how to solve this

Yuxin-Yu, 2022-07-18:
hi，I got the same error ,have you solve it

mbelda, 2022-07-18:
not yet

mbelda, 2022-07-21:
I have an idea of how to solve it now.
First, check that your vivado version is old enough to have a folder named board_parts (2018.3 or 2019.2 work)
Then, install the files of your board in that folder.

closed 2022-07-21 by mbelda